esphome/issues #3668: esp32c3 bluetooth_proxy can't connet wifi

liwei19920307, 2022-10-08:
### The problem

esp32c3 bluetooth_proxy can't connet wifi, only stop scan on boot can fix but not everytime work

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.10.1

### What platform are you using?

ESP32

### Board

esp_c3_13u

### Component causing the issue

esp32_ble_tracker,bluetooth_proxy

### Example YAML snippet

```yaml
substitutions:
  device_name: esp485

esphome:
  name: ${device_name}
  platformio_options:
    platform: https://github.com/tasmota/platform-espressif32.git#Tasmota/203
    platform_packages:
      - framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.3
    board_build.flash_mode: dio
  # on_boot:
  #   - priority: 600
  #     then:
  #       - esp32_ble_tracker.stop_scan:
  #   - priority: 200
  #     then:
  #       - esp32_ble_tracker.start_scan:

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

logger:

api:
  password: !secret api_password

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on

web_server:
  port: 80

esp32_ble_tracker:
  scan_parameters:
    active: true
    continuous: true

bluetooth_proxy:
  active: true
```

### Anything in the logs that might be useful for us?

```txt
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:22:49]Build:Feb  7 2021
[19:22:49]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[19:22:49]SPIWP:0xee
[19:22:49]mode:DIO, clock div:1
[19:22:49]load:0x3fcd6100,len:0x38c
[19:22:49]load:0x403ce000,len:0x6a8
[19:22:49]load:0x403d0000,len:0x2358
[19:22:49]SHA-256 comparison failed:
[19:22:49]Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
[19:22:49]Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
[19:22:49]Attempting to boot anyway...
[19:22:49]entry 0x403ce000
[19:22:49][I][logger:243]: Log initialized
[19:22:49][C][ota:465]: There have been 0 suspected unsuccessful boot attempts.
[19:22:49][D][esp32.preferences:113]: Saving 1 preferences to flash...
[19:22:49][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:22:49][I][app:029]: Running through setup()...
[19:22:49][C][uart.arduino_esp32:077]: Setting up UART...
[19:22:49][   354][E][HardwareSerial.cpp:494] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.
[19:22:49]
[19:22:50][D][esp32_ble_tracker:264]: Starting scan...
[19:22:50][C][wifi:037]: Setting up WiFi...
[19:22:50][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:23:20][W][wifi:541]: Timeout while connecting to WiFi.
[19:23:20][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:23:20][W][esp32_ble_tracker:112]: Too many BLE events to process. Some devices may not show up.
[19:23:50][W][wifi:541]: Timeout while connecting to WiFi.
[19:24:25][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:8C reason='Association Leave'
[19:24:55][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:8C reason='Association Leave'
[19:25:30][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:7D reason='4-Way Handshake Timeout'
[19:25:30][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:8C reason='Association Leave'
[19:26:00][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:7D reason='4-Way Handshake Timeout'
[19:26:00][W][esp32_ble_tracker:112]: Too many BLE events to process. Some devices may not show up.
[19:26:05][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:26:35][W][wifi:541]: Timeout while connecting to WiFi.
[19:26:35][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:8C reason='Association Leave'
[19:26:35][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:7D reason='4-Way Handshake Timeout'
[19:26:35][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:26:35][W][esp32_ble_tracker:112]: Too many BLE events to process. Some devices may not show up.
[19:27:05][W][wifi:541]: Timeout while connecting to WiFi.
[19:27:05][W][esp32_ble_tracker:112]: Too many BLE events to process. Some devices may not show up.
[19:27:10][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:27:40][W][wifi:541]: Timeout while connecting to WiFi.
[19:27:40][290965][E][WiFiGeneric.cpp:112] set_esp_interface_ip(): Netif Set IP Failed! 0x5007
[19:27:40][290966][E][WiFiSTA.cpp:315] begin(): set ip failed!
[19:28:10][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:7D reason='Auth Expired'
[19:28:10][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:8C reason='Beacon Timeout'
[19:28:10][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:8C reason='Beacon Timeout'
[19:28:10][D][esp32_ble_tracker:264]: Starting scan...
[19:28:10][W][esp32_ble_tracker:112]: Too many BLE events to process. Some devices may not show up.
[19:28:15][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:28:45][W][wifi:541]: Timeout while connecting to WiFi.
[19:28:45][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:7D reason='Beacon Timeout'
[19:28:45][W][wifi:583]: Restarting WiFi adapter...
[19:28:45][D][esp-idf:000]: E (356434) wifi:
[19:28:45][D][esp-idf:000]: timeout when WiFi un-init, type=4
[19:28:45][D][esp-idf:000]: 
[19:28:45]
[19:28:50][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:29:20][W][wifi:541]: Timeout while connecting to WiFi.
[19:29:20][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:7D reason='Beacon Timeout'
[19:29:20][W][wifi:583]: Restarting WiFi adapter...
[19:29:20][D][esp-idf:000]: E (391569) wifi:
[19:29:20][D][esp-idf:000]: timeout when WiFi un-init, type=4
[19:29:20][D][esp-idf:000]: 
[19:29:20]
[19:29:25][I][wifi:255]: WiFi Connecting to 'Miya'...
[19:29:55][W][wifi:541]: Timeout while connecting to WiFi.
[19:29:55][W][wifi_esp32:495]: Event: Disconnected ssid='Miya' bssid=D4:35:38:77:AB:7D reason='Beacon Timeout'
[19:29:55][W][wifi:583]: Restarting WiFi adapter...
[19:29:55][D][esp-idf:000]: E (426704) wifi:
[19:29:55][D][esp-idf:000]: timeout when WiFi un-init, type=4
[19:29:55][D][esp-idf:000]:
```

### Additional information

# add this conf to be able connet wifi
 on_boot:
    - priority: 600
      then:
        - esp32_ble_tracker.stop_scan:
    - priority: 200
      then:
        - esp32_ble_tracker.start_scan:

stumpylog, 2022-10-08:
I'm also seeing this, with an Adafruit Qt Py (which is a -C3).  It doesn't look like stopping the scan at boot is working for me, the wifi connection is flaky still.  It might connect, but many minutes after powerup and then disconnect after a random amount of time.

If I remove the Bluetooth related configuration, it will connect and work normally.  Except that isn't useful for its purpose.

jerobins, 2022-10-22:
Seeing the same thing with the M5Stamp C3. Mine will generally stay connected, but getting connected can be instant or take 10 mins. I tested the boot priority scripts above, including adding a delay before the start_scan, but had no affect for me.

liwei19920307, 2022-12-05:
new version ok

closed 2022-12-05 by liwei19920307

SeanCline, 2023-04-02:
> Using stop_scan does stop continuous scanning. One would have to do continuous scanning via lambdas or such...

@jerobins' snippet solved my esp32-c3 bluetooth_proxy woes, but then I ran into the aforementioned continuous scanning issue. My pvvx_mithermometers would only receive advertises for the first couple minutes after boot. My workaround to that was simple. Instead of periodic scanning using lambdas, just set `continuous: true` for `start_scan`.

```
+  on_boot:
+    - priority: 500
+      then:
+        - esp32_ble_tracker.stop_scan:
+    - priority: 100
+      then:
+        - delay: 5min
+        - esp32_ble_tracker.start_scan:
+            continuous: true
```

Hopefully this issue is still on someone's radar as some devices can be a real pain to get UART set up to unbrick.

ssieb, 2023-12-23:
For anyone finding this later, if you reboot the device 10 times, it will go to safe mode and you can OTA update it.