esphome/issues #3669: ESP32: Multiple UART not working (works on ESP8266)

jjansen85, 2022-10-08:
### The problem

I configure 3 UART instances on a ESP32 (see YAML below).
One for P1 DSMR (utility meter; electricity)
Two for Multical (second utility meter; district heating)

Logger is disabled (baudrate: 0).

Depending on the order of the UART instances, only one of my setups is working.
First 2 Multical UART, then 1 DSMR UART configuration --> result: Multical is working
First 1 DSMR UART, then 2 Multical UART configuration --> result: DSMR is working

How can I ensure both setups are working?

Same software runs perfectly on a ESP8266 (software UARTs)

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.10

### What platform are you using?

ESP32

### Board

mhetesp32minikit

### Component causing the issue

uart

### Example YAML snippet

```yaml
substitutions:
  device_name: "meterkast-esp32"
  friendly_name: Meterkast 

esp32:
  board: mhetesp32minikit
  framework:
    type: arduino

esphome:
  name: ${device_name}
  includes:
  - includes/multical401.h  

external_components:
- source: github://oxan/esphome-stream-server  

# Disable logging
logger:
  # level: DEBUG
  baud_rate: 0

wifi: 
  manual_ip:
    static_ip: !secret ip_meterkast_esp32

packages:
[... 322 lines cut ...]

  }
 private:
  bool parity_check(unsigned input) {
    bool inputparity = input & 128;
    int x = input & 127;
 
    int parity = 0;
    while(x != 0) {
        parity ^= x;
        x >>= 1;
    }

    if ( (parity & 0x1) != inputparity )
      return(1);
    else
      return(0);
  }
//  Sensor *temperature_sensor = new Sensor();
  UARTDevice *_tx;
  UARTDevice *_rx;  

};

```

JeroenVanOort, 2022-10-13:
It should work...

Can you be more specific about UART's not functioning? It would be helpful to know if it is RX and or TX not working in all three cases. Maybe you could do some measurements with a USB-serial-converter/logic analyser/scope.

ssieb, 2022-10-13:
In order to use 3 uarts, one of them has to use the default pins for UART0.

jjansen85, 2022-10-13:
> In order to use 3 uarts, one of them has to use the default pins for UART0.

It looks like you are correct here. Previously the YAML below was not working (last UART config; TX to my Multical was not functioning).
When I change the pin of my first UART instance from GPIO23 to GPIO3, my Multical setup is (also) working :-)

```yaml
uart:    
#Uart for DSMR P1-meter    
- rx_pin: GPIO3
 #rx_pin: GPIO23
  baud_rate: 115200
  id: uart_dsmr
  rx_buffer_size: 1700

#Multical UART, 2 instances due to different baud-rates.
- rx_pin: GPIO22
  baud_rate: 1200
  data_bits: 8
  parity: NONE
  stop_bits: 1
  id: uart_rx
- tx_pin: GPIO21
  baud_rate: 300
  data_bits: 8
  parity: NONE
  stop_bits: 1
  id: uart_tx   
```

[Documentation](https://esphome.io/components/uart.html) seems to be be wrong in this case:
`The ESP32 has three UARTs. Any pair of GPIO pins can be used, as long as they support the proper output/input modes.`

synegic, 2022-11-02:
Just wondering if there is a way to solve this without changing the pins to the default for UART0? I am attempting to create an ESPHome set-up with off the shelve devices that do not follow the original pinout and where I cannot rearrange them. Same UART configuration utilising all three UART's with the custom pins work well in Arduino-based project, but not in ESPHome. Would be great if we could make this possible.

jeroen85, 2023-03-15:
Not solved.