esphome/issues #3671: ESPHOME_PROJECT_NAME and ESPHOME_PROJECT_VERSION not working

Links2004, 2022-10-09:
### The problem

the config for project with name and version is not working, but the defines are there in code.

### Which version of ESPHome has the issue?

git dev 2022.10.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

project

### Example YAML snippet

```yaml
esphome:
  name: esphome-test

project:
  name: mqtt-discovery-test
  version: 1.0.0
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration ../data/test.yaml...
Failed config

project: [source ../data/test.yaml:5]
  
  Component not found: project.
  name: mqtt-discovery-test
  version: 1.0.0
```

### Additional information

https://github.com/esphome/esphome/blob/fd57b21affab08a48fd03e7b1e61c522fd524f0b/esphome/core/config.py#L370-L372

```cpp
    if CONF_PROJECT in config:
        cg.add_define("ESPHOME_PROJECT_NAME", config[CONF_PROJECT][CONF_NAME])
        cg.add_define("ESPHOME_PROJECT_VERSION", config[CONF_PROJECT][CONF_VERSION])
```

bkaufx, 2022-10-10:
You need to tab in the project section, see example here: https://esphome.io/guides/creators.html

Links2004, 2022-10-10:
my bad, miss read the code.
still warping my head around all this for my change here:
https://github.com/esphome/esphome/pull/3887

closed 2022-10-10 by Links2004