esphome/issues #3680: esp-idf

marcinkowalczyk, 2022-10-11:
### The problem

Device: Shelly Plus1 (esp32 based) 

```
Linking /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14button_commandERKNS0_20ButtonCommandRequestE+0x0): undefined reference to `esphome::button::Button::press()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection16send_button_infoEPNS_6button6ButtonE+0x4): undefined reference to `esphome::button::Button::get_device_class[abi:cxx11]()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::button_command(esphome::api::ButtonCommandRequest const&)':
/config/esphome/.esphome/build/shelly-plus1-furtka/src/esphome/components/api/api_connection.cpp:692: undefined reference to `esphome::button::Button::press()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::send_button_info(esphome::button::Button*)':
/config/esphome/.esphome/build/shelly-plus1-furtka/src/esphome/components/api/api_connection.cpp:683: undefined reference to `esphome::button::Button::get_device_class[abi:cxx11]()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/main.o:(.literal._Z5setupv+0xe4): undefined reference to `vtable for esphome::output::OutputButton'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/main.o:(.literal._Z5setupv+0xe8): undefined reference to `vtable for esphome::output::OutputButton'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/main.o:(.literal._Z5setupv+0x1e8): undefined reference to `esphome::button::Button::Button()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/shelly-plus1-furtka/.pioenvs/shelly-plus1-furtka/src/main.o: in function `setup()':
/config/esphome/.esphome/build/shelly-plus1-furtka/src/main.cpp:290: undefined reference to `esphome::button::Button::Button()'
collect2: error: ld returned 1 exit status
```

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2022.10.2

### What platform are you using?

ESP32-IDF

### Board

esp32doit-devkit-v1

### Component causing the issue

button

### Example YAML snippet

```yaml
output:
  - platform: gpio
    id: "relay_output"
    pin: GPIO26

button:
  - platform: output
    name: "Furtka"
    icon: "mdi:gate"
    output: relay_output
    duration: 1500ms
```

ssieb, 2022-10-12:
Try cleaning the build directory.

marcinkowalczyk, 2022-10-12:
Hi,

Seems it worked. Thank!

closed 2022-10-12 by marcinkowalczyk