esphome/issues #3682: GL-S10 OTA Failure, even with safe boot

tony-fav, 2022-10-12:


tony-fav, 2022-10-12:
Thanks for the confirmation and move!

DJBenson, 2022-10-13:
Following on from https://github.com/esphome/bluetooth-proxies/issues/39, even if safe boot is used, I cannot update the GL-S10 using OTA.

I have a revision 2.1 board. I have flashed the initial firmware using the webflash tool and now trying to adopt the device into ESPHome and do the initial OTA. The process will not get beyond single-digit percentage without timing out.

Steps to reproduce:
Flash with intitial binary
Attempt to adopt into ESPHome
OTA times out
Repeat using the safe boot button in HA
OTA times out

INFO Uploading /data/gl-s10-bt-proxy-9ad9b0/.pioenvs/gl-s10-bt-proxy-9ad9b0/firmware.bin (1493552 bytes)
Uploading: [=== ] 5%
ERROR Error sending data: timed out

I've seen OTA fail on other ESPHome projects due to the flash size being too high to allow the OTA binary - do these stats look ok in that respect;

RAM: [== ] 16.7% (used 54592 bytes from 327680 bytes)
Flash: [======== ] 81.4% (used 1493450 bytes from 1835008 bytes)

trvrnrth, 2022-10-13:
To add another data point, OTA is incredibly flakey for me as well, but will eventually succeed after numerous retries. The device won't hold a stable API connection to HA though with frequent drops.

nagyrobi, 2022-10-14:
https://github.com/esphome/issues/issues/3690

pvizeli, 2022-10-18:
If you would ping the device, you could see the issue. I very sure that the new IDF/Arduino framework version would fix it

tony-fav, 2022-10-19:
Because I'm incredibly lazy and don't want to get the device back out for a reflash, I've just been pounding the OTA with a loop until it seems to work:
```python
from esphome import espota2
while True:
  out = espota2.run_ota('192.168.4.111', 3232, '', 'gl-s10-bt-proxy-111111.bin')
  if out == 1:
    continue
  else:
    break
```
Hope this helps someone else. Latest versions of ESPHome seem to be more stable so the upgrade is worth it, but I just didn't want to mess with the hardware again.

Go0oSer, 2022-10-27:
Interestingly enough manually setting my switch port that was connected to the device to 10Mb rather than 100Mb autoneg. allowed me to OTA with no issue.

lupac, 2022-10-28:
> Interestingly enough manually setting my switch port that was connected to the device to 10Mb rather than 100Mb autoneg. allowed me to OTA with no issue.
> 
> The device is also causing CRC errors on the switch. Ive tried different cables and it does not change anything.

Same, after setting 10FDX manually, OTA worked on first attempt.

Gunsmithy, 2022-10-31:
> The device is also causing CRC errors on the switch. Ive tried different cables and it does not change anything.

Yeah I had to actually disable RSTP on my UniFi switch port to which my GL-S10 was connected or it would block the port thinking it was looping.

I also cannot OTA, and it seems to reboot itself every 15 seconds or so. Not sure if Bluetooth or Ethernet related, but I do have a lot of advertisements nearby so the LED flashes very rapidly.

Hardware revision v2.1

trvrnrth, 2022-10-31:
I had a bit of mess about a couple of weeks ago and have had my 2.1 revision device running fine since (OTA and availability) by:
- using the latest ESP-IDF framework (compile from https://github.com/trvrnrth/esphome/tree/gls10, set framework to `esp-idf`)
- disabling the LED blink on advertisement (comment out `on_ble_advertise` automation)
- using the default `esp32_ble_tracker` scan parameters (comment out `interval` and `window`)

I've not spent any time since digging into exactly what impact each of those changes has individually but hopefully this information can help the overall investigation/resolution.

gerard33, 2022-11-04:
Thanks for the update @trvrnrth.
Can you share a bit information how we can compile from your Github? Can we use [external_components](https://esphome.io/components/external_components.html?highlight=external_components) for that?

efaden, 2022-11-09:
Having similar trouble.

trvrnrth, 2022-11-09:
@gerard33 I'm guessing you're using the home assistant add-on. If you install the dev version you should be able to specify an `esphome_fork` of `trvrnrth:gls10` but I've never actually tried as I just compile locally for dev testing.

@efaden The only changes are as I describe but [here you go](https://github.com/esphome/issues/files/9974661/gl-s10-bt-proxy.yaml.txt).

efaden, 2022-11-09:
> @gerard33 I'm guessing you're using the home assistant add-on. If you install the dev version you should be able to specify an `esphome_fork` of `trvrnrth:gls10` but I've never actually tried as I [...]

Thanks.  Is there a way to specify the fork on only the few devices?  Can you change the fork in the yaml?  Can't seem to find a way.

efaden, 2022-11-09:
> @gerard33 I'm guessing you're using the home assistant add-on. If you install the dev version you should be able to specify an `esphome_fork` of `trvrnrth:gls10` but I've never actually tried as I [...]

What version of the board you have .... 
Getting an error "  
  Unknown value 'IP101', valid options are 'LAN8720', 'TLK110'."

gerard33, 2022-11-10:
@trvrnrth I indeed use the HA add-on. I have installed the dev version and specified your fork. I got an error during compiling, so I forked the latest dev branch from esphome github and made your changes [there](https://github.com/gerard33/esphome). Then compiling went okay.

When changing the framework to `esp-idf` in the yaml code however I get an error that the ethernet feature is only available for Arduino. How did you handle this?

[image]

efaden, 2022-11-11:
@trvrnrth Upgraded to what you suggested.... seems to hold connection with HA better, but still getting tons of loss if I ping it.

gerard33, 2022-11-12:
@efaden, how did you bypass the error for `ethernet` in combination with `esp-idf` framework as I see in my post above yours. The code I use for this part is the same as in the example from trvrnrth.

trvrnrth, 2022-11-12:
@gerard33 The dev branch in your fork does not appear to include all the changes in the branch I tested with. At a glance you seem to have only pulled in a single commit to bump versions.

gerard33, 2022-11-12:
Thanks, I will have a look at your other commits and update my branch accordingly.

efaden, 2022-11-20:
Mine is still barely reliable.  Getting the bluetooth data, but can't get OTA or ping stability or anything.

verstrepeng, 2022-11-23:
Was having exactly the same proble, ping packet loss about 20-25% and unable to flash OTA.
I followed a tip in this thread and set the switchport fixed to 10/full.
This extremely reduces packet loss (now 1-2%) and I able to do an OTA flash. So I would guess there is something with the ethernet driver ?

oywino, 2022-12-06:
This device is referred to as "The Best Home Assistant Bluetooth Proxy" !
How can that be, when everyone seems to have nothing but trouble to get it working?

Isn't there a single success story out there?

efaden, 2022-12-06:
I'm with you there.... I have mine out of service until there is a fix

jsimonetti, 2022-12-07:
Just to shift the odds. Mine has been working like a charm ever since i set speed and duplex to fixed as suggested earlier on in the comments. 

Looking for success stories on an issue tracker sounds very counterproductive.

DJBenson, 2022-12-07:
> Just to shift the odds. Mine has been working like a charm ever since i set speed and duplex to fixed as suggested earlier on in the comments.
> 
> Looking for success stories on an issue tracker sounds very counterproductive.

Likewise, I've set the port on my UniFi switch to 10 meg and it's been solid since.

efaden, 2022-12-07:
Powering with Poe or USB?

DJBenson, 2022-12-07:
PoE

DJBenson, 2022-12-07:
[image]

DJBenson, 2022-12-07:
Prior to this, I could not flash the device OTA - now it succeds every time.

oywino, 2022-12-08:
> Please try 2022.12(Beta), Ethernet driver has been rewritten.

Link please ?

phaeton82, 2022-12-08:
I installed the ESP home (Beta) add-on. Changed the YAML to the YAML below and flashed the device with serial connecton. Flash went fine, device is accessible. After this I tried OTA updates. Pressed the 'safe boot'  button for the device and tried OTA multiple times. This unfortionately results in a time out. Hope this input might help:

`# Instructions on opening and wiring for flashing on https://blakadder.com/gl-s10!
esphome:
  name: gl-s10-bt-testunit
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"

# turn on Power LED when esphome boots!
  on_boot:
    then:
      - output.turn_on: power_led

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

# # Configuration fo V2.3 hardware revision
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO5

# Comment the above and use this instead for V1.0 revision of the hardware
#ethernet:
#  type: LAN8720
#  mdc_pin: GPIO23
#  mdio_pin: GPIO18
#  clk_mode: GPIO17_OUT
#  phy_addr: 1

api:
logger:
ota:

# dashboard_import:
#   package_import_url: github://esphome/bluetooth-proxies/gl-s10.yaml@main

esp32_ble_tracker:

# Bluetooth LED blinks when receiving Bluetooth advertising
  # on_ble_advertise:
  #   then:
  #     - output.turn_on: bluetooth_led
  #     - delay: 0.5s
  #     - output.turn_off: bluetooth_led

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

sensor:
  - platform: uptime
    name: Uptime Sensor

## DEVICE SPECIFIC CONFIGURATION
# network LED (white one) configured as status led

# button on the side labeled RESET
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO33
      inverted: true
    name: "Reset Button"

# output settings for LED's marked Power and Bluetooth
# power LED use: see code line 12
# bluetooth LED use: see code line 41
output:
  - platform: gpio
    pin: GPIO14
    inverted: true
    id: power_led
  - platform: gpio
    pin: GPIO12
    inverted: true
    id: bluetooth_led

# # since these pins are broken out inside and labeled as I2C pins they're configured here
# i2c:
#   sda: 15
#   scl: 13
#   scan: true
`
logs:
`INFO Reading configuration /config/esphome/gl-s10-bt-proxy-d628bc.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing gl-s10-bt-testunit (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
RAM:   [==        ]  15.8% (used 51888 bytes from 327680 bytes)
Flash: [======    ]  56.0% (used 1028137 bytes from 1835008 bytes)
Building /data/gl-s10-bt-testunit/.pioenvs/gl-s10-bt-testunit/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/gl-s10-bt-testunit/.pioenvs/gl-s10-bt-testunit/firmware.bin"], ["/data/gl-s10-bt-testunit/.pioenvs/gl-s10-bt-testunit/firmware.elf"])
Wrote 0x10c6b0 bytes to file /data/gl-s10-bt-testunit/.pioenvs/gl-s10-bt-testunit/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 24.59 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of gl-s10-bt-testunit.local
INFO  -> 192.168.1.41
INFO Uploading /data/gl-s10-bt-testunit/.pioenvs/gl-s10-bt-testunit/firmware.bin (1033904 bytes)
Uploading: [=                                                           ] 2% 
ERROR Error sending data: timed out
`

nagyrobi, 2022-12-08:
> > Please try 2022.12(Beta), Ethernet driver has been rewritten.
> 
> Link please ?

RTFM: https://esphome.io/guides/faq.html#how-do-i-update-to-the-latest-beta-release

oywino, 2022-12-09:
After a while, the GL-S10 began behaving very strange. No matter what I do in ESPHome, I always get only this in the log:

```
Failed config

ethernet: [source /config/.esphome/packages/f8744889/gl-s10.yaml:24]
  
  This feature is only available with frameworks ['arduino'].
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO5
```
I tried to reboot (power cycle) and I tried SafeBoot, Nothing seems to help. What could have happened?
How do I solve this problem?

phaeton82, 2022-12-09:
Did you use the beta addon? You should for now when using idf instead of Arduino. Some more info like the YAML would help

oywino, 2022-12-09:
> Did you use the beta addon? You should for now when using idf instead of Arduino. Some more info like the YAML would help

If that's an answer/question to me @phaeton82 - I appreciate your support, but I do not understand what you mean?
The beta that @nagyrobi linked to seems to me to be a beta of the ESPHome docker container (Web GUI), while I was looking for a beta firmware for the GL-S10 
My problem is that I can no longer access the device at all through the Web GUI

nagyrobi, 2022-12-09:
Please read the documentation carefully. 
With the Beta add-on you create beta firmware. With the Dev addon, you create dev firmware.

nagyrobi, 2022-12-09:
> no longer access the device at all through the Web GUI

You shouldn't use `bluetooth-proxy` and `webserver` on the same node, they are too much together for an ESP32 under `arduino` platform. `webserver` is not available (yet) under `esp-idf` platform.

oywino, 2022-12-09:
Sorry for my ignorance, but you can see that I need guidance. 
What exactly do you mean by "add-on" ?  I didn't "add" anything "on" anywhere.  All I did was to install a self conitained docker container after successfully flashing the device. and everything seemed ok - until this morning when nothing seems ok.

nagyrobi, 2022-12-09:
I suggest to start reading from the beginning:
https://esphome.io/guides/getting_started_hassio.html

Using ESPHome the most straightforward way is to have it installed as a Home Assistant Add-on.
There are 3 possible ESPHome Add-ons you can choose from: normal, beta, and dev. You can install beta and dev if you set yourself as an advanced user in Home Assistant.

oywino, 2022-12-09:
I cannot install anything as a Home Assistant "add-on" as I am not running Home Assistant Supervised (or Hassio as used to be known as). I think my problem is that everybody automatically assumes that averyong - including me - is running Home Assistant Supervised.  So, that's why I installed ESPHome as a self-contained Docker image.  It has no connection at all to Home Assistant.

nagyrobi, 2022-12-09:
> everybody automatically assumes that everyone - including me - is running Home Assistant Supervised

Yes because that's the _supported_ way to use Home Assistant, and that's how the big majority uses it. Of course it can be deployed by other means, but that's another story.

You should research then how to install ESPHome Beta as a self-contained Docker image, with that, you'd be able to compile the beta firmware releases.

oywino, 2022-12-09:
Exactly, and that's the link I was hoping someone could pass me. Caus all the DOC's and everything else is pretty useless to me as I'm not running Supervised.  Where can I find a container image of ESPHome beta?

nagyrobi, 2022-12-09:
It has been passed to you [just above](https://github.com/esphome/issues/issues/3682#issuecomment-1342943451).

If you can't find your way based on that, for more adequate questions please join to the Discord support channel.

oywino, 2022-12-09:
Sorry for the inconvenience. I did manage to install v2022.12.0b2 by creating a docker compose file as follows:

```
version: '3'

services:
  homeass:
    image: esphome/esphome:beta
    restart: "no"
    network_mode: host

    environment:
      LANG: nb_NO.UTF-8
      TZ: Europe/Oslo

    volumes:
      - /share/Public/ESPHome/beta:/config
```

Go0oSer, 2022-12-09:
I flashed the new beta version using esp-idf instead of arduino.  The unit seems more stable at 100Mb but Im unable to OTA any longer either on 10Mb or 100Mb.  Seems some good progress was made, but we are not all there quite yet.

mehuman, 2022-12-16:
I'm having the same OTA issue using ethernet. Setting the port to 10mbps fails immediately, setting the port to 100mbps allows a partial upload before timing out. Running 2022.12.1

raoulbhatia, 2022-12-17:
@mehuman if you configued your device BEFORE `2022.12.x`,

- did you change `esp32:framework:type` FROM `arduino` TO `esp-idf`, AND
- did you do a serial flash?

I was able to perform my first OTA from `2022.12.0`to `2022.12.1` successfully after this procedure.

mehuman, 2022-12-19:
@raoulbhatia I did both. Although thinking of it I may not have serially flashed with the framework change. I'll do another serial flash with the framework updated and see if that fixes the issue. Thanks!

oywino, 2022-12-20:
I'm currently on 2022.12.0b2 and OTA was 100% stable after changing to 10MF on all switch ports.
This was 2 weeks ago. Today, attempts to OTA to 12.1 fails 100%  every time.
So, does this mean that the update package itself has changes that can influence the device's ability to respond and stay stable, even before being installed?

efaden, 2022-12-21:
I'm on 2022.12.2 .... Still totally unstable.

oywino, 2022-12-21:
Every attempt to ping the GL-S10 ends with > 25% packet loss. sometimes 100%
Before upgrading to 2022.12.0b2 I managed to get it 100% stable by switching to 10MF.
After 2022.12.0b2 it is permanently unstable again.
I can try to do another serial flash, but which FW version should I choose? Link please?

mehuman, 2022-12-21:
I reflashed via serial and am still getting OTA failures. My board is labeled version 2.1. Including my code and the outcome of setting the port to auto-negotiate, 100FDX, and 10FDX via the Unifi switch below. I tried the same tests using the firmware [here](https://esphome.github.io/bluetooth-proxies/?diy)

Config
```
substitutions:
  device_name: gl-s10-ble
  name: gl-s10-bt-proxy
  static_ip: REDACTED

esphome:
  name: ${name}
  name_add_mac_suffix: true
  on_boot:
    then:
      - output.turn_on: power_led

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

# Configuration fo V2.3 hardware revision
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO5
  domain: !secret domain
  manual_ip:
    static_ip: ${static_ip}
    subnet: !secret subnet
    gateway: !secret gateway

api:
  password: !secret ota_password
logger:
ota:
  password: !secret ota_password

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
# Bluetooth LED blinks when receiving Bluetooth advertising
  on_ble_advertise:
    then:
      - output.turn_on: bluetooth_led
      - delay: 0.5s
      - output.turn_off: bluetooth_led

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

## DEVICE SPECIFIC CONFIGURATION
# network LED (white one) configured as status led
status_led:
  pin:
    number: GPIO32
    inverted: true

# button on the side labeled RESET
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO33
      inverted: true
    name: "Reset Button"

# output settings for LED's marked Power and Bluetooth
# power LED use: see code line 12
# bluetooth LED use: see code line 41
output:
  - platform: gpio
    pin: GPIO14
    inverted: true
    id: power_led
  - platform: gpio
    pin: GPIO12
    inverted: true
    id: bluetooth_led

# since these pins are broken out inside and labeled as I2C pins they're configured here
i2c:
  sda: 15
  scl: 13
  scan: true
```
[auto-negotiation.txt](https://github.com/esphome/issues/files/10279740/auto-negotiation.txt)
[100FDX.txt](https://github.com/esphome/issues/files/10279741/100FDX.txt)
[10FDX.txt](https://github.com/esphome/issues/files/10279742/10FDX.txt)

oywino, 2022-12-21:
Whoever wrote the network driver for ESPHome need to get their act together. As it is now, this thing is completely useless!

ACrazyConcept, 2022-12-23:
Not only ota is unstable. I also get a handful of failures a day in HA.

```
2022-12-21 17:51:45.245 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esphome-gl-s10-bt-proxy @ 10.0.0.241: Hello timed out
2022-12-21 17:54:05.808 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.0.241: Timeout waiting for response for <class 'api_pb2.DeviceInfoRequest'> after 10.0s
2022-12-21 18:19:13.105 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.0.241: Timeout waiting for response for <class 'api_pb2.DeviceInfoRequest'> after 10.0s
2022-12-21 20:36:06.528 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esphome-gl-s10-bt-proxy @ 10.0.0.241: Timeout while connecting to ('10.0.0.241', 6053)
2022-12-21 21:32:17.617 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esphome-gl-s10-bt-proxy @ 10.0.0.241: Timeout waiting for response for <class 'api_pb2.ConnectRequest'> after 10.0s
2022-12-22 05:28:29.718 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esphome-gl-s10-bt-proxy @ 10.0.0.241: Timeout while connecting to ('10.0.0.241', 6053)
2022-12-22 20:34:58.973 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.0.241: Timeout waiting for response for <class 'api_pb2.DeviceInfoRequest'> after 10.0s
2022-12-22 23:24:06.710 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esphome-gl-s10-bt-proxy @ 10.0.0.241: Hello timed out
2022-12-23 06:30:22.106 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.0.241: Timeout waiting for response for <class 'api_pb2.DeviceInfoRequest'> after 10.0s
```

saschaludwig, 2023-01-26:
With 2022.12.7 it is still impossible to do OTA updates. Packet loss is around 25% if using ethernet.

oywino, 2023-01-27:
ESPHome is owned by NabuCasa, as far as I understand. Can we turn to them to request assistance to fix this?
It doesn't seem like anyone from "codeowner" follows this thread.....

liudger, 2023-01-28:
I have the same issue. Tried a lot to get this bluetooth proxy stable. I really hope the developers spend some time to this issue as this is one of the promoted bluetooth poe proxies on [this](https://esphome.github.io/bluetooth-proxies/?diy) page. It's hard of course if this is an issue with certain hardware. I have also a unifi poe switch [link](https://store.ui.com/collections/unifi-network-switching/products/usw-lite-16-poe). It looks like many people have the same setup? Would maybe a good idea if one of the developers get the same setup?

oywino, 2023-01-28:
I don't think the GL-S10 is a PoE capable device.

DJBenson, 2023-01-28:
> I don't think the GL-S10 is a PoE capable device.

Huh? It absolutely is PoE enabled, I power mine by PoE from my UniFi switch.

[16 more replies]