esphome/issues #3687: Active Bluetooth connections not working with August Smart Lock Gen 3's

liquidbear99, 2022-10-14:


hgregorian, 2022-10-14:
As others have pointed out, I too am experiencing issues with an active bluetooth proxy and a Level lock.  Removing BT proxy or disabling "active" allows the Lock to reestablish connectivity with a USB BT adapter (albeit with poor signal, hence the desire for a working BT proxy).  Attaching logs (VERY_VERBOSE) from ESPHome BT proxy where lock/unlock/identify actions are performed from HA but result in no lock actions.  Versions are as follow; HA: 2022.10.3 and ESPHome: 2022.9.4.  Relevant ESPHome yaml:

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

[esp32-bt-proxy-level-lock.log](https://github.com/esphome/issues/files/9788316/esp32-bt-proxy-level-lock.log)

amwalters, 2022-10-16:
Not working on 2022.11.0-dev either.

> Could this PR be related? [esphome/esphome#3425](https://github.com/esphome/esphome/pull/3425)

Anyone smart enough to capture a handshake in wireshark?

lokanx, 2022-10-16:
Any progress solving this issue? Is there any workaround that can be applied for now?

hgregorian, 2022-10-18:
Tried esphome 2022.10.0b2 on my BT proxy (attempting to get Level Lock to function for clarity).  Issue persists.  I did notice a couple of interesting behaviors which may be helpful (I hope!).  Shortly after HK controller initializes the lock (within ~30 seconds) after HA restart, actions performed against it function and ESPHome logs, along with improved RSSI, appear to indicate that the BT proxy is behaving as expected.  However, less than 30 seconds later (after performing no lock action), actions no longer function and HA eventually logs timeout errors.  Curiously, if an action is performed (identify for example) inside that 30 second window and every <30 seconds, the lock continues to accept actions.  I've reliably reproduced this at least a dozen times today.

```
Logger: homeassistant.core
Source: components/esphome/bluetooth/client.py:66
First occurred: 1:03:17 PM (6 occurrences)
Last logged: 1:24:32 PM

Error executing service: <ServiceCall button.press (c:01GFPB0Q5QZRW8BES7E0NSMS3E): entity_id=['button.patio_door_lock_identify']>
Error executing service: <ServiceCall button.press (c:01GFPB0WTP1QJDJWA0AXSJ82A7): entity_id=['button.patio_door_lock_identify']>
Error executing service: <ServiceCall button.press (c:01GFPB0YPCF7JHFD98RY2EA04K): entity_id=['button.patio_door_lock_identify']>
Error executing service: <ServiceCall button.press (c:01GFPC2TKABRQHATG11W55JKZR): entity_id=['button.patio_door_lock_identify']>
Error executing service: <ServiceCall lock.unlock (c:01GFPC7DA1K4XQFF9JESQ09XE5): entity_id=['lock.patio_door']>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 445, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 444, in send_message_await_response_complex
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 64, in _async_wrap_bluetooth_operation
    return await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 317, in read_gatt_char
    return await self._client.bluetooth_gatt_read(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 538, in bluetooth_gatt_read
    resp = await self._connection.send_message_await_response_complex(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 447, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.BluetoothGATTReadRequest'>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1756, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 116, in _async_press_action
    await self.async_press()
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/button.py", line 118, in async_press
    await self.async_put_characteristics({key: val})
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/entity.py", line 91, in async_put_characteristics
    return await self._accessory.put_characteristics(payload)
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/connection.py", line 616, in put_characteristics
    results = await self.pairing.put_characteristics(characteristics)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/pairing.py", line 110, in _async_wrap
    return await func(self, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 76, in _async_wrap
    return await func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/pairing.py", line 798, in put_characteristics
    await self._async_request_under_lock(
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/pairing.py", line 250, in _async_request_under_lock
    pdu_status, result_data = await ble_request(
[... cut ...]
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 52, in _async_wrap_bluetooth_connected_operation
    return await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 66, in _async_wrap_bluetooth_operation
    raise asyncio.TimeoutError(str(err)) from err
asyncio.exceptions.TimeoutError: Timeout waiting for response for <class 'api_pb2.BluetoothGATTReadRequest'>
```

austwhite, 2022-10-19:
This seems to happen with all bluetooth locks trying to use the ESPHome Active Bluetooth Proxy.  Title of this issue may need updating to reflect this

Tekuchb, 2022-10-19:
Hey there!
I`ve got a same trouble with ESPHome Bluetooth Proxy and Level Bolt lock.
Any ideas how to make it work?

heytcass, 2022-10-19:
Looks very similar to the memory issue I am having with smoke detectors (https://github.com/home-assistant/core/issues/77926) causing the ESP to run out of memory. esphome/issues#3653

However, in that thread there was supposed to be a memory leak mitigation in this morning's version of ESPHome that went out, but I am still having the issue. Hopefully having other platforms with the same issues will lead to an overall solution, though.

nagyrobi, 2022-10-20:
Note that Bluetooth Proxy currently only supports active connection with one single Bluetooth device.

danlo315, 2022-10-20:
I have a dedicated esp32 for this purpose, not connected with anything else except trying with my Assure Lock.

hgregorian, 2022-10-20:
@nagyrobi 

That's definitely interesting news!  Currently, I've got a BT proxy right next to a door with a Level Lock (Bolt).  I've got two other Level Locks in the house that I was confident were not connected to the BT proxy, but rather back to a BT adapter attached to my HA hardware.  To test this assertion, I removed the batteries from the other two Level Locks.  The sole Level Lock available, which is right next to the BT proxy continues to not accept any actions.

Also, if, for example, there were other devices in someone's dwelling that were also seeking active BT connections, your statement might imply that control of the intended device (my lock in this case) would not function?  Please clarify.

To the best of my knowledge this single lock is now the only device in my house seeking an active BT connection (the other 2 locks are now without power).  Perhaps some steps could be provided to determine what other devices may be seeking active BT connections?

> Note that Bluetooth Proxy currently only supports active connection with one single Bluetooth device.

austwhite, 2022-10-20:
@nagyrobi 
The issue with the locks is not due to single active connection.
Even if the lock is the only Bluetooth device, it still fails, even with the latest version of ESPHome.

amwalters, 2022-10-20:
> @nagyrobi The issue with the locks is not due to single active connection. Even if the lock is the only Bluetooth device, it still fails, even with the latest version of ESPHome.

Confirming.  Like @danlo315, the lock is the only device I am connected to.

hgregorian, 2022-10-20:
Issue persists with newly released HA 2022.10.5 (and ESPHome 2022.10.0)

duckmavis, 2022-10-21:
I can confirm the same issue with august g4 locks (I have 2). I am updated to the latest versions of both HA and ESPHome.

marc25v, 2022-10-21:
Also causes same issue with Kwikset/Weiser Premis locks. 

Seems to also affect my MELNOR Bluetooth water valve. Perhaps the issue stems from devices that are protected by a passcode or security hash of sorts.

amwalters, 2022-10-21:
Not clear if related, but it looks like [aioesphomeapi](https://github.com/esphome/aioesphomeapi) does not recognise all message types available in the [esphome api_bp2 service](https://esphome.io/api/api__pb2__service_8cpp_source.html) where we are seeing errors.  Specifically I am getting notification in my HA logs: 
[aioesphomeapi.connection] atom-btproxy @ 192.168.88.103: Skipping message type XX
when trying to send commands.  XX corresponds to 82, 83, or 84.

These map to: 
- 82 - [BluetoothGATTErrorResponse](https://esphome.io/api/classesphome_1_1api_1_1_bluetooth_g_a_t_t_error_response.html)
- 83 - [BluetoothGATTWriteResponse](https://esphome.io/api/classesphome_1_1api_1_1_bluetooth_g_a_t_t_write_response.html)
- 84 - [BluetoothGATTNotifyResponse](https://esphome.io/api/classesphome_1_1api_1_1_bluetooth_g_a_t_t_notify_response.html)

Could be nothing as we are waiting for a response to BluetoothGATTReadRequest, but maybe one of these is a valid response that is being dropped causing the timeout?

nagyrobi, 2022-10-21:
@jesserockz

abishur, 2022-10-22:
I'm also having these issues with my level lock and D1 Mini NodeMCU ESP32 ESP-WROOM-32.  Out of curiosity, is anyone here using an ESP with a ethernet connection or are we all using it wirelessly?

liquidbear99, 2022-10-22:
Looks like still no progress on this front even when updating to 2022.10.5 and ESPHome 2022.10.0. I updated the title to reflect that multiple smart locks are experiencing a problem. I have success connecting only with the pi's onboard bluetooth, despite abysmal range (the pi needs to be within a few feet from the lock to work).

Also I believe overcoming any passcode or security hash can be accomplished. As of the October release of home assistant, I am able to connect bluetooth only through the Yale access bluetooth integration; however, the dev's of that integration did need to update it to get it to work. It may be worth contacting the people who worked on that integration and asking what they fixed from the September release of yale access bluetooth to the October release. (in October, the Pi's onboard bluetooth was not working for that lock)

hgregorian, 2022-10-22:
@liquidbear99 I’m curious…could you see if you’re able to interact with the lock shortly after HA is restarted while the BT proxy is within the lock’s range?  In my case, I’m able to perform actions against my lock shortly after an HA restart (and can confirm it’s happening via BT proxy), but if I try to perform actions more than ~30 seconds after HA restart, the lock will not behave.

bdraco, 2022-10-24:
It looks like we might have a race in allocating connections.

AFAICT the connection slot only gets marked as used when the connection is made, not when the connection is in progress

hgregorian, 2022-10-26:
ESPHome 2022.10.1 seems to have moved things in the right direction.  Now, instead of timeouts, lock actions take effect, although sometimes with significant delay (most I counted was ~20 seconds).  The latency does not appear to be caused by the BT proxy's Wi-Fi connection as it's RSSI is currently -54 dBm.  FWIW, the locks's BT RSSI is somewhere around -70 and -60 dBm.

austwhite, 2022-10-26:
@hgregorian 
I find even a direct Bluetooth, not through a proxy, sometimes the Yale Bluetooth integration will take 30 seconds with my August locks  
This may be the integration?

hgregorian, 2022-10-26:
@austwhite the two locks that are closest to the USB BT adapters are extremely responsive (under 5 second), so I don't think it's the integration (HomeKit Controller in my case).

> @hgregorian 
> 
> I find even a direct Bluetooth, not through a proxy, sometimes the Yale Bluetooth integration will sometimes take 30 seconds with my August locks  
> 
> This may be the integration?

austwhite, 2022-10-26:
@hgregorian 
See how it goes, but if your ESP device is WiFi it could be a delay due to ESP32's time sharing Bluetooth and Wifi on the one antenna,
I'll do some additional tests with my system when I get home also :)

nagyrobi, 2022-10-26:
> once BLE proxy was enabled on an ESPHome device, it would disconnect from the HA API from time to time if the ESP device was on WiFi

Yes that's exactly the reason why Ethernet devices are recommended for using `bluetooth-proxy`. However, you could also try tweaking `interval` and `window` parameters of your config. The optimal values depend on many factors, like stuff you use on your nodes, advertising frequency of you Bluetooth devices, WiFi signal interference and thus packet retransmission rate etc. You might try to reduce these values even more or reduce `window` to lower than `interval`.

Tekuchb, 2022-10-26:
> @hgregorian See how it goes, but if your ESP device is WiFi it could be a delay due to ESP32's time sharing Bluetooth and Wifi on the one antenna, I found once BLE proxy was enabled on an ESPHome device, it would disconnect from the HA API from time to time if the ESP device was on WiFi. That dropout did not happen on Ethernet connected ESP devices. I'll do some additional tests with my system when I get home also :)

I was got a same problem with disconnections time to time from HA.
Solved by deleting device from "Settings - Devices - ESPHome - %devicename%"
Anytime when i chande code caused bluetooth features by ESPHome - it make problems with disconnections from API and this solution helps great)

nagyrobi, 2022-10-26:
Setting static IP addres and adding the node by IP improves connection stability _a lot_.

hgregorian, 2022-10-26:
AFAICT, the BT proxy is not suffering from Wifi/API disconnects.  I've pointed it out before, but here it goes again.  Shortly after HA is started, the lock is very responsive with the BT proxy (<1 sec response time).  If I continue to perform actions against it (<30 seconds apart from each other), such as lock/unlock/identify, response time remains great.  It's only after it's been idle for >30 seconds that long delays occur (previous to 2022.10.1, unresponsive).  When the lock eventually responds during one of the long delays, continuing to perform actions against it (<30 seconds apart from each other) results in very quick response time.  I've seen several recent pulls on home-assistant/core relating to BT connectivity (some related to BT proxy, and some related to the integration I'm using specifically - HK controller ) by @bdraco.  They definitely seem related.

https://github.com/home-assistant/core/pull/80887
https://github.com/home-assistant/core/pull/80882
https://github.com/home-assistant/core/pull/80891
https://github.com/home-assistant/core/pull/80899
https://github.com/home-assistant/core/pull/80885
https://github.com/home-assistant/core/pull/80908

bdraco, 2022-10-26:
While its going to be a lot better with 2022.11.x, until we fix the problem with [slot allocation](https://github.com/esphome/issues/issues/3687#issuecomment-1289260848) or add support for multiple connections at the same time, I expect its still going to have issue

Probably good to watch the outcome of https://github.com/esphome/aioesphomeapi/pull/293 as once that merges we will likely have fixed the race where the connection slot gets used by multiple integrations at the same time and things don't go so well

abishur, 2022-10-26:
To avoid the proxy from trying to connect to multiple devices at once is there a way to specify which device to allow connections to?  Or where should I look for more documentation on the bluetooth proxy to make sure I'm actually using best practices while testing this?

bdraco, 2022-10-26:
> To avoid the proxy from trying to connect to multiple devices at once is there a way to specify which device to allow connections to?

No, its a bug that needs to be fixed so there aren't workarounds for it

hgregorian, 2022-10-26:
I definitely appreciate your hard work on this issue!  I'll keep an eye out for HA 2022.11.0 beta.

> While its going to be a lot better with 2022.11.x, until we fix the problem with [slot allocation](https://github.com/esphome/issues/issues/3687#issuecomment-1289260848) or add support for multiple connections at the same time, I expect its still going to have issues
> 
> Probably good to watch the outcome of [esphome/aioesphomeapi#293](https://github.com/esphome/aioesphomeapi/pull/293) as once that merges we will likely have fixed the race where the connection slot gets used by multiple integrations at the same time and things don't go so well

austwhite, 2022-11-02:
Just putting a comment to follow this issue also really.
Still present in 2022.10.2 with HA 2022.10.5 :)    Guessing it was known it would still be present, but I want to track the issue :)

bdraco, 2022-11-02:
## 🤚 I have the same issue! 👇 Read below

Hi there! Sorry to read you are experiencing the same issue as other people here. Nobody likes having issues 😞

Finding an issue on a GitHub issue tracker that matches your problem is kinda nice: At least you know you are not alone. So, let's leave a comment with: "Yeah, I have the same issue", or "+1!!!". 🚫 **No!**

### Please do not create "I have the same issue" comments!

Not just this issue or this project, but anywhere on GitHub or any issue tracker on the internet even. It doesn't add to triaging the issue. It only generates noise when reading / triaging the case. 

### What to do instead?

Instead, go to the top of the issue, and add a 👍 emoji reaction. That way, we can still see that you (and how many others) experience this issue while keeping the issue threads nice and clean.

bdraco, 2022-11-02:
This issue will get closed when the problem is resolved.

You'll at least need HA 2022.11.0 and a device flashed with https://github.com/esphome/esphome/pull/3971

amwalters, 2022-11-03:
Confirmed working for level locks. Thanks!

Tekuchb, 2022-11-03:
> Confirmed working for level locks. Thanks!

Is it works stable?
Can you post your yaml code please?

Tekuchb, 2022-11-03:
> Confirmed working for level locks. Thanks!

At the moment, adding to HA has passed from 5 times but finaly - it works!
But the opening / closing of the lock is delayed from 2 to 13 seconds from the moment the button is pressed.
Nothing else is connected to the ESP Gateway - only the Level Bolt lock.

bdraco, 2022-11-03:
> But the opening / closing of the lock is delayed from 2 to 13 seconds from the moment the button is pressed.

That's normal as bluetooth connections take a few seconds to establish and the ESP32's bluetooth isn't super fast.

Tekuchb, 2022-11-04:
Maybe it is possible to get some more speed by restricting bluetooth connections?
Is it possible to proxy only fixed MAC addresses? 
My esp shows ~20 different MAC addresses (TV`s, mobile phones, BLE sensors etc.).
Maybe there is a possibility to filter data by 1 or 2 mac addresses and it will get more performance?

bdraco, 2022-11-04:
https://github.com/esphome/esphome/pull/3868/files/21908154b952bba8f5956ddc7cabaffce601a0f0..c4fa46b420d1d7f23b1b87504e4835b949f3c082

abishur, 2022-11-04:
What is the correct yaml code to get the level lock to connect?  I thought I was using it before, but I unplugged my bluetooth dongle to make sure it was connecting to the esp32 and now it will not pair.  Home Assistant detects i and when I go to configure it it asks for the pairing code and gets stuck asking for that code over and over again.  I've reset the lock and confirmed I'm entering the right pairing code.

I have the D1 Mini NodeMCU ESP32 ESP-WROOM-32 and my current YAML I'm uploading to the EPS32 (Generic) is this: 
```
substitutions:
  name: esp32-bluetooth-proxy-25728c
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

doguitar, 2022-11-04:
The solution for me was to just keep trying. Eventually it went through and now it works. It took me quite a while, I just left the home assistant window up on a different monitor and redid it repeatedly until it worked.

For what it's worth, I had the lock in factory reset state and then paired the app after it was paired in home assistant.

Tekuchb, 2022-11-04:
As I wrote earlier - my level lock only connected on the 5th try ... I just enter the code every time.
Also I disable ble_tracker in ESP code.
A little earlier @bdraco posted a link to a pull-request "add basic filtering to ble_proxy".
As far as I understand it is just rssi filtering at the moment, but if the developer could add mac address filtering, I think he will get more performance with the addition and use of bluetooth devices. Hope so =)

abishur, 2022-11-04:
Okay, patience and persistence did the trick.  I also had the lock in a reset state.  I'll disable the ble_tracker and see if that improves response time some, but I suspect the overarching issue is that the ESP32 doesn't handle Bluetooth and WiFi at the same time.  It would be neat if I could tie two of them together (since I have plenty lying around) and let one handle bluetooth and the other handle wifi, but a more logical solution would be to use a wired variant.  I'd be VERY interested in seeing response times from an individual using the wired ESP32 leaving Wifi turned off.  Until then, I now have a stable connection to my level lock through home assistant so this is a major win

Tekuchb, 2022-11-04:
Tomorrow i can take lock from my office and try it with wt32-s1 by ethernet at home and post result

bdraco, 2022-11-04:
I wouldn’t consider using the wifi ones for active connections as the latency is going to be unbearable.  The ethernet ones should be a much better experience.

steve28, 2022-11-04:
So if have ESPHome 2022.11.0-dev and HA 2022.11.1, the proxy should work to join locks - specifically level locks?

Tekuchb, 2022-11-05:
Already tested Level Bolt lock with wt32-s1 on wired connection. Couple Xiaomi temp\humidity sensors attached to this ESP.
- Lock added from first try without troubles. Perhaps because I have fewer bluetooth devices around at home?
- Delay betwen butoon press and lock reaction 8 - 10 seconds
Still too much delay. Natively on android app delay is ~3 seconds.

sionide21, 2022-11-05:
Disabling `esp32_ble_tracker` did the trick for me. After doing that I was able to pair to my Level lock on the first attempt. 

I'm still getting this error a lot in the logs, but it doesn't seem to stop it from actually operating the lock:

```
Error writing char/descriptor at handle 0x5C, status=133
```

I'm running `esphome @ git+https://github.com/esphome/esphome@1554c5700e032d390b04a54434da7f2c0081c7dd` and HA 2022.11.1

steve28, 2022-11-05:
> Disabling `esp32_ble_tracker` did the trick for me. After doing that I was able to pair to my Level lock on the first attempt.

@sionide21 forgive the perhaps newb question?  How you disable esp32_ble_tracker?  My whole config is:

```
substitutions:
  name: esp32-bluetooth-proxy-3c80c0
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name_add_mac_suffix: false

logger:
  level: VERBOSE

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

sionide21, 2022-11-05:
Im not sure how to do it with the packaged version, but it looks like that is using [this yaml config](https://github.com/esphome/bluetooth-proxies/blob/main/esp32-generic.yaml) under the covers. 

There's probably a way to pull that in to make customizations.

If you remove lines [30-34](https://github.com/esphome/bluetooth-proxies/blob/main/esp32-generic.yaml#L30-L34) it will disable the tracker.

mrrobotnic, 2022-11-05:
I am wondering the same thing about disabling the esp32_ble_tracker and how to implement that fix.  I tried pulling in the code from https://github.com/esphome/bluetooth-proxies/blob/main/esp32-generic.yaml and just removed the lines associated with the esp32_ble_tracker but seems like that is not doing the trick since my August Pro 3 just gets stuck in an initializing loop through the Yale Access Bluetooth integration.

mrrobotnic, 2022-11-12:
> while you wait, if your homeassistant has built in BLE and able to move to have it in range of the lock you might be able to proceed; i tried this with two ASL-06 Yale Linus locks today, works with onboard BLE in my Pi but not Bluetooth proxy through ESPHome.

@beikeland Was it just the initial setup that needed to happen with onboard BLE and then able to switch over to the bluetooth proxy or does it always have to be through the onboard BLE?  Thanks a bunch!

beikeland, 2022-11-14:
@mrrobotnic unfortunately it doesn't seem to be working with the proxy after the initial setup; i relocated my pi temporarily while issue 3761 gets resolved.