esphome/issues #3690: GL-S10 issues

pvizeli, 2022-10-14:


nagyrobi, 2022-12-08:
Please try 2022.12(Beta), Ethernet driver has been rewritten.

unmesh59, 2023-01-26:
@nagyrobi 

I flashed 2022.12.5 over serial before I saw this. Does it have the rewritten driver?

nagyrobi, 2023-01-26:
It should, yes

unmesh59, 2023-01-26:
Still getting timeouts doig OTA updates :-(

Lampyridae1984, 2023-01-26:
> Still getting timeouts doing OTA updates :-(

I stopped doing OTA on the GL-S10. So far it's not working properly for me either (hardware version 1.0).

saschaludwig, 2023-02-02:
With 2022.12.8 OTA is still impossible due to massive packet loss when using the ethernet driver

liudger, 2023-02-14:
Still unstable with 2023.2.0b5

Go0oSer, 2023-03-26:
https://github.com/esphome/bluetooth-proxies/pull/78/commits/d3185745b7d6a5a1658a3c473441d50e8fd44fc6

For users who have v2.1 of the hardware.