esphome/issues #3691: Tuya MCU regression on Dimmer

rpavlik, 2022-10-14:
### The problem

I have a "Feit Electric" smart dimmer, that uses tuya-mcu. I've manually flashed esphome on it and it worked well for quite a while. However, updating in the last few months broke it, it will error with `Tuya:  Initialization failed. Current init_state: 0`  (or sometimes init state of 3 after a reset).

I can still talk to it fine via the CLI, but because it can't talk to the MCU it won't show up in Home Assistant, etc.

2022.1.2 works fine with this abbreviated log:

```
[11:26:41][C][logger:233]: Logger:
[11:26:41][C][logger:234]:   Level: DEBUG
[11:26:41][C][logger:235]:   Log Baud Rate: 0
[11:26:41][C][logger:236]:   Hardware UART: UART0
[11:26:41][C][uart.arduino_esp8266:100]: UART Bus:
[11:26:41][C][uart.arduino_esp8266:101]:   TX Pin: GPIO1
[11:26:41][C][uart.arduino_esp8266:102]:   RX Pin: GPIO3
[11:26:41][C][uart.arduino_esp8266:104]:   RX Buffer Size: 256
[11:26:41][C][uart.arduino_esp8266:106]:   Baud Rate: 9600 baud
[11:26:41][C][uart.arduino_esp8266:107]:   Data Bits: 8
[11:26:41][C][uart.arduino_esp8266:108]:   Parity: NONE
[11:26:41][C][uart.arduino_esp8266:109]:   Stop bits: 1
[11:26:41][C][uart.arduino_esp8266:111]:   Using hardware serial interface.
[11:26:41][C][light:098]: Light 'dining'
[11:26:41][C][light:100]:   Default Transition Length: 0.0s
[11:26:41][C][light:101]:   Gamma Correct: 1.00
[11:26:41][C][tuya.light:069]: Tuya Dimmer:
[11:26:41][C][tuya.light:071]:    Dimmer has datapoint ID 2
[11:26:41][C][tuya.light:073]:    Switch has datapoint ID 1
...
[11:26:42][C][tuya:028]: Tuya:
[11:26:42][C][tuya:039]:   Datapoint 1: switch (value: OFF)
[11:26:42][C][tuya:041]:   Datapoint 2: int value (value: 0)
[11:26:42][C][tuya:055]:   Product: '{"p":"tebc75erlgslgpn5","v":"1.0.8","m":1}'
```

while at least 2022.9.4 (but I think a few versions before), don't work

### Which version of ESPHome has the issue?

2022.9.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

configured as esp01_1m but actually tywe2s

### Component causing the issue

tuya

### Example YAML snippet

```yaml
# disable logging and serial connection. baud_rate/serial connection is used by tuyamcu
logger:
  baud_rate: 0
  # level: NONE

# My dimmer used this hardware serial port
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

# Enable light and dimmer
light:
  - platform: "tuya"
    name: "dining"
    dimmer_datapoint: 2
    switch_datapoint: 1
    # min_value: 10
    min_value: 100
    max_value: 1000
    # state_topic: "dining_dimmer/light/state"
    # command_topic: "dining_dimmer/light/command"
    retain: true
```

### Anything in the logs that might be useful for us?

```txt
[10:59:41][I][app:102]: ESPHome version 2022.9.4 compiled on Oct 14 2022, 10:58:48
[10:59:41][C][wifi:502]: WiFi:
[10:59:41][C][logger:275]: Logger:
[10:59:41][C][logger:276]:   Level: DEBUG
[10:59:41][C][logger:277]:   Log Baud Rate: 0
[10:59:41][C][logger:278]:   Hardware UART: UART0
[10:59:41][C][uart.arduino_esp8266:102]: UART Bus:
[10:59:41][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[10:59:41][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[10:59:41][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[10:59:41][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[10:59:41][C][uart.arduino_esp8266:109]:   Data Bits: 8
[10:59:41][C][uart.arduino_esp8266:110]:   Parity: NONE
[10:59:41][C][uart.arduino_esp8266:111]:   Stop bits: 1
[10:59:41][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[10:59:41][C][light:104]: Light 'dining'
[10:59:41][C][light:106]:   Default Transition Length: 0.0s
[10:59:41][C][light:107]:   Gamma Correct: 1.00
[10:59:41][C][tuya.light:094]: Tuya Dimmer:
[10:59:41][C][tuya.light:096]:    Dimmer has datapoint ID 2
[10:59:41][C][tuya.light:098]:    Switch has datapoint ID 1
[10:59:41][C][status:034]: Status Binary Sensor 'Dining Room Dimmer Node Status'
[10:59:41][C][status:034]:   Device Class: 'connectivity'
[10:59:41][C][version.text_sensor:021]: Version Text Sensor 'Dining Room Dimmer ESPHome Version'
[10:59:41][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[10:59:41][C][captive_portal:088]: Captive Portal:
[10:59:41][C][mdns:100]: mDNS:
[10:59:41][C][mdns:101]:   Hostname: diningdimmer
[10:59:42][C][ota:089]: Over-The-Air Updates:
[10:59:42][C][ota:090]:   Address: diningdimmer.lan2:8266
[10:59:42][C][api:138]: API Server:
[10:59:42][C][api:139]:   Address: diningdimmer.lan2:6053
[10:59:42][C][api:143]:   Using noise encryption: NO
[10:59:42][C][mqtt:065]: MQTT:
[10:59:42][C][mqtt:069]:   Client ID: 'diningdimmer-50029113d877'
[10:59:42][C][mqtt:071]:   Discovery prefix: 'homeassistant'
[10:59:42][C][mqtt:072]:   Discovery retain: YES
[10:59:42][C][mqtt:074]:   Topic Prefix: 'diningdimmer'
[10:59:42][C][mqtt:076]:   Log Topic: 'diningdimmer/debug'
[10:59:42][C][mqtt:079]:   Availability: 'diningdimmer/status'
[10:59:42][C][mqtt.light:077]: MQTT Light 'dining':
[10:59:42][C][mqtt.light:078]:   State Topic: 'diningdimmer/light/dining/state'
[10:59:42][C][mqtt.light:078]:   Command Topic: 'diningdimmer/light/dining/command'
[10:59:42][C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Dining Room Dimmer Node Status':
[10:59:42][C][mqtt.binary_sensor:023]:   State Topic: 'diningdimmer/status'
[10:59:42][C][mqtt.text_sensor:023]: MQTT Text Sensor 'Dining Room Dimmer ESPHome Version':
[10:59:42][C][mqtt.text_sensor:024]:   State Topic: 'diningdimmer/sensor/dining_room_dimmer_esphome_version/state'
[10:59:42][C][tuya:033]: Tuya:
[10:59:42][C][tuya:036]:   Initialization failed. Current init_state: 0
[10:59:42][C][tuya:041]:   If no further output is received, confirm that this is a supported Tuya device.
[10:59:42][E][tuya:409]: Initialization failed at init_state 0

or later

[11:03:17][C][tuya:033]: Tuya:
[11:03:17][C][tuya:038]:   Configuration will be reported when setup is complete. Current init_state: 3
[11:03:17][C][tuya:041]:   If no further output is received, confirm that this is a supported Tuya device.
```

wtsoper, 2022-10-29:
Hi Ryan, did you get any resolution? my Feit dimmers won't work and have the same issue. How can you downgraded esphome to  install an older version to get it working again?

rpavlik, 2022-10-31:
I got no information, but I installed esphome=2022.6.3 into a new virtual env and used that to flash it back to that version. (You may need to delete your platformio cache files in your home directory if you get an error) I then made a note in the config file to not update past that version and linked to this issue so I don't update it in the future, at least until this is fixed

zarko-a, 2023-03-09:
@rpavlik Did you have to do anything special to get the old esphome working? 

I'm encountering the same issue with `Initialization failed. Current init_state: 0` but when I installed the old version `pip install esphome==2022.6.3` and tried to compile esphome it complained about PlatformIO version. 

```
INFO Generating C++ source...
INFO Compiling app...
********************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
********************************************************************************************************************************************************************************************

Processing loft-light (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif8266 @ 3.2.0
Error: Please upgrade to the PlatformIO Core 6
```
I can't find any workarounds for this.

rpavlik, 2023-03-09:
I had to remove all platform io related files from my system, and I installed the old version in a fresh venv.