esphome/issues #3692: the sprinkler.next_valve and sprinkler.previous_valve actions don't honor valve enable/disable status

brembygit, 2022-10-14:
### The problem

the sprinkler.next_valve and sprinkler.previous_valve actions don't honor valve enable/disable status
when you call NEXT and PREVIOUS the next or the previous valve is turned on even if the valve status is disable, the expected behavior should skip the disabled valve and move to the first [next or previous] enable valve

```
sprinkler.next_valve action
Advances to the next valve (numerically). If manual_selection_delay is configured, the controller will wait before activating the selected valve. If no valve is active, the first valve (as they appear in the controller’s configuration) will be started.

on_...:
  then:
    - sprinkler.next_valve: sprinkler_ctrlr
```

```
sprinkler.previous_valve action
Advances to the previous valve (numerically). If manual_selection_delay is configured, the controller will wait before activating the selected valve. If no valve is active, the last valve (as they appear in the controller’s configuration) will be started.

on_...:
  then:
    - sprinkler.previous_valve: sprinkler_ctrlr
```

### Which version of ESPHome has the issue?

v2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

brembygit, 2022-10-14:
This behavior is documented, should we consider to switch to a change request to change this behaviors?

`The next_valve, previous_valve and start_single_valve actions ignore whether a valve is enabled via its enable switch.`

schinckel, 2022-12-07:
Whilst it is documented, I found it surprising (and not easy to work around). I'd support it being the default that it does not include disabled valves.

closed 2023-02-22 by kbx81