esphome/issues #3693: ESPHome crashes on OTA update

bkbartk, 2022-10-14:
### The problem

Hello,

I have now 5 ESPHome devices on an esp8285 chip. But a lot of the time(between 40 and 60%) they crash during OTA install.

What happens is.
I OTA install, then at some level the connection times out.

And all my entities are gone untill repowering, See screenshot
[image]
I tried disabling safe mode but that doesn't help.
and a lot of other settings.

When I log very verbose
```
19:52:11	[D]	[ota:312]	OTA in progress: 70.5%
19:52:26	[D]	[ota:312]	OTA in progress: 70.8%
19:53:19	[D]	[ota:312]	OTA in progress: 71.6%
19:56:55	[V]	[json:031]	Attempting to allocate 512 bytes for JSON serialization
19:56:59	[V]	[json:051]	Size after shrink 64 bytes
19:57:20	[V]	[json:031]	Attempting to allocate 512 bytes for JSON serialization
19:57:21	[V]	[json:051]	Size after shrink 64 bytes
19:57:57	[V]	[json:031]	Attempting to allocate 512 bytes for JSON serialization
19:58:01	[V]	[json:051]	Size after shrink 64 bytes
```
you can see 2 lines are added in the log whenever a new browser opens the webpage, but the page still be empty. 
Also in HA the entities are gone.

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.3

### What platform are you using?

ESP8266

### Board

sonoff basic and sonoff tx t0 eu 1c

### Component causing the issue

ota

### Example YAML snippet

```yaml
substitutions:
  device_name: bijkeuken-light
  switch_name: Lamp Bijkeuken

esphome:
  name: ${device_name}

wifi:
  id: wifiId
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: HIGH
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

# Enable logging
logger:
  # level: VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  safe_mode: false
  password: !secret ota_password
  # on_begin:
  #   - lambda: id(wifiId).set_power_save_mode(esphome::wifi::WIFI_POWER_SAVE_NONE);
  # on_end:
  #   - lambda: id(wifiId).set_power_save_mode(esphome::wifi::WIFI_POWER_SAVE_HIGH);
  # on_error:
  #   - lambda: id(wifiId).set_power_save_mode(esphome::wifi::WIFI_POWER_SAVE_HIGH);

web_server:
  port: 80
  auth:
    username: admin
    password: !secret web_server_password

button:
  - platform: restart
    name: "${device_name} switch Restart"
    icon: "mdi:restart"

esp8266:
  board: esp01_1m
  restore_from_flash: true
  early_pin_init: false

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode:
        input: true
        pullup: true
      inverted: true
    id: button_1
    on_press:
      then:
        - switch.toggle: light_1

switch:
  - platform: gpio
    pin: GPIO12
    name: ${switch_name}
    id: light_1
    icon: "mdi:lightbulb"
    on_turn_on:
    - light.turn_on: stat_led
    on_turn_off:
    - light.turn_off: stat_led
light:
  - platform: status_led
    id: stat_led
    pin:
      number: GPIO13
      inverted: yes
```

### Anything in the logs that might be useful for us?

```txt
19:53:19	[D]	[ota:312]	OTA in progress: 71.6%
19:56:55	[V]	[json:031]	Attempting to allocate 512 bytes for JSON serialization
19:56:59	[V]	[json:051]	Size after shrink 64 bytes
19:57:20	[V]	[json:031]	Attempting to allocate 512 bytes for JSON serialization
19:57:21	[V]	[json:051]	Size after shrink 64 bytes
19:57:57	[V]	[json:031]	Attempting to allocate 512 bytes for JSON serialization
19:58:01	[V]	[json:051]	Size after shrink 64 bytes
```

### Additional information

I tried a lot of different configs on 5 different devices formerly flashed with tasmota.
And with Tasmota I didn't have those issues.

zigomatichub, 2022-10-20:
I'm a user like you :) I don't have knowledge but 95% of OTA is working fine. 

Try to implement this:

button: 
  - platform: restart
    name: $device_name Restart
    device_class: restart
    entity_category: diagnostic

  - platform: safe_mode
    name: $device_name Safe Mode Boot
    entity_category: diagnostic

then before OTA, go to HA is start in Safe Mode

bkbartk, 2022-10-20:
I tried safe mode but it doesn't make any difference.
The issue with additional buttons or attributes is that they become unavailable when all crashes.
both in the web interface as in HA so I can't see or use them.
the only input element left on the web interface is the upload button but when I use that one after an issue it always says "not enough space"

I know 95% goes well, there is some issue with esp and my wifi, but whenever it fails it shouldn't crash.
only esp8285 devices crash, esp8266 devices recover nicely. And ESP32 doesn't seem to have an issue at all,

Sonoff somehow has there default input button on GPIO0, so that should be ok.

bkbartk, 2022-10-20:
I found 2 new things
whenever this issue occurs on an ESP8266 device the logs keeps spitting out
```
18:44:06	[W]	[ota:139]	Socket could not enable tcp nodelay, errno: 104
18:44:06	[W]	[ota:139]	Socket could not enable tcp nodelay, errno: 104
```

and when the device becomes unavailable, except from an empty web interface, it will be available again after restarting the AP.
So probably disconnecting form wifi will do.

Still this is an issue

closed 2023-03-11 by github-actions[bot]