esphome/issues #3702: PMSX003 update_interval errors

grucha532, 2022-10-18:
### The problem

The update_interval does not work with PMS using the same syntax as with DHT22. I've tried a lot: with 'ms','s', without anything but number.

Is there a way to set up an interval?

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.7

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

PMSX003

### Example YAML snippet

```yaml
# PMS7003
uart:
  rx_pin: 5
  tx_pin: 4
  baud_rate: 9600

sensor:

# DHT22
  - platform: dht
    pin: 13
    temperature:
      name: "Hall Temperature"
    humidity:
      name: "Hall Humidity"
    update_interval: 60s

# PMS7003
  - platform: pmsx003
    type: PMSX003
    pm_1_0:
      name: "Particulate Matter <1.0µm Concentration"
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
    pm_10_0:
      name: "Particulate Matter <10.0µm Concentration"
    update_interval: 60000ms
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp-smogomierz.yaml...
Failed config

sensor.pmsx003: [source /config/esphome/esp-smogomierz.yaml:45]
  platform: pmsx003
  type: PMSX003
  pm_1_0: 
    name: Particulate Matter <1.0µm Concentration
  pm_2_5: 
    name: Particulate Matter <2.5µm Concentration
  pm_10_0: 
    name: Particulate Matter <10.0µm Concentration
  
  [update_interval] is an invalid option for [sensor.pmsx003]. Please check the indentation.
  update_interval: 60000ms [source /config/esphome/esp-smogomierz.yaml:53]
```

Warter21, 2022-10-18:
it is supported from 2022.5.0.

armonicfpv, 2022-10-28:
I'm on version 2022.10.5 and I'm also receiving the same error trying to use update_interval 
Any way to fix this?

Warter21, 2022-10-29:
> I'm on version 2022.10.5 and I'm also receiving the same error trying to use update_interval Any way to fix this?

Are you sure? 2022.10.1 is the latest version.

armonicfpv, 2022-10-29:
> > I'm on version 2022.10.5 and I'm also receiving the same error trying to use update_interval Any way to fix this?
> 
> Are you sure? 2022.10.1 is the latest version.

Yes I'm sure the current version is 2022.10.5

Warter21, 2022-10-29:
> > > I'm on version 2022.10.5 and I'm also receiving the same error trying to use update_interval Any way to fix this?
> > 
> > 
> > Are you sure? 2022.10.1 is the latest version.
> 
> Yes I'm sure the current version is 2022.10.5

yeah, the latest Home Assistant version is 2022.10.5 but the latest Esphome version is 2022.10.1. 
can you show me some logs?

armonicfpv, 2022-10-29:
INFO Reading configuration /config/esphome/air-gradient-3.yaml...
Failed config

sensor.pmsx003: [source /config/esphome/air-gradient-3.yaml:180]
  platform: pmsx003
  type: PMS5003T
  uart_id: uart1
  pm_2_5: 
    name: Air-Gradient-3 - Particulate <2.5µm
    id: pm25
    filters: 
      - median: 
          window_size: 11
          send_every: 11
          send_first_at: 3
  temperature: 
    name: Air-Gradient-3 - PMS5003T Temperature
    id: temp3
    filters: 
      - median: 
          window_size: 11
          send_every: 11
          send_first_at: 3
      - offset: 0.8
  humidity: 
    name: Air-Gradient-3 - PMS5003T Humidity
    id: humidity3
    filters: 
      - median: 
          window_size: 11
          send_every: 11
          send_first_at: 1
  
  [update_interval] is an invalid option for [sensor.pmsx003]. Please check the indentation.
  update_interval: 60000ms [source /config/esphome/air-gradient-3.yaml:208]

armonicfpv, 2022-11-03:
Would be awesome if someone could look in to this problem

ssieb, 2022-11-03:
You haven't shown what esphome version you have.

ssieb, 2022-11-04:
https://esphome.io/changelog/2022.4.0.html#home-assistant-addon-repo

closed 2022-11-04 by ssieb