esphome/issues #3704: Error Compiling

johnmyster, 2022-10-19:
### The problem

Up until this weekend, I had a config running on the board.   Since having gotten this board setup, HASS and ESP32Home has likely been updated a few times.  I went to make changes to the ESP32 yaml, and now can no longer compile and install updated config.  I've simplified the config to no avail.  Now just trying to get a new ESP board adopted and updating with a simple starter config, no dice.  Further updates to HASS and ESP32H add on did not fix.  Purchased new boards, did not fix.  Wireless, USB to HASS server (RPi3), USB to PC all fail.  Can't even compile to download and use online updater.

Thank you for looking into this.  Am I the only one?

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.4

### What platform are you using?

ESP32

### Board

HiLetGo ESP32 Development Board

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-9c871c

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-9C871C"
    password: "FiK4nQZGWGry"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esphome-web-9c871c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-9c871c (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- Update @ 1.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/application.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/color.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/component.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/component_iterator.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/controller.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/entity_base.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/helpers.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/log.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/scheduler.cpp.o
Compiling /data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/util.cpp.o
xtensa-esp32-elf-g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/esphome-web-9c871c/.pioenvs/esphome-web-9c871c/src/esphome/core/util.cpp.o] Error 4
========================= [FAILED] Took 78.88 seconds =========================
```

und3f1ned, 2022-10-19:
Same here

`xtensa-esp32-elf-g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/hallhub/.pioenvs/hallhub/src/esphome/components/api/api_connection.cpp.o] Error 4`

ssieb, 2022-10-27:
A compile option has just been added to dev, so you can limit the number of processes.
https://github.com/esphome/esphome/pull/3952

closed 2022-10-27 by ssieb

johnmyster, 2022-10-27:
If I understand correctly, the ultimate solve is for me to move up to a RPi4 or other for my HASS server?

ssieb, 2022-10-27:
You will probably be happier with that, but with that new option, compiling should work.  You can also do the compiling on some other computer if you want.

johnmyster, 2022-10-31:
ssieb,

I just got back home and loaded the esp32-dev add on into HASS.  I'm a novice, but not seeing where to set the parallel limit.  

Looking under file editor, I don't see anything like the esphome/core/config.py path in my esphome folder.

ssieb, 2022-10-31:
https://deploy-preview-2391--esphome.netlify.app/components/esphome.html#configuration-variables
`compile_process_limit`

johnmyster, 2022-12-07:
I was able to set the process limit in the config file but still got an error during compile...every time.

I installed esphome on my PC and write/compile from there, rather than within HASS.  Works.  Happy.  I'm past the hassle.

pvheater, 2022-12-07:
You should fill in it in the node file. In my case the file name is 
watro.yaml. Not in configuration. yaml

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<https://kommunikationsdienste.t-online.de/redirects/email_app_android_sendmail_footer>

--- Original-Nachricht ---
Von: johnmyster
Betreff: Re: [esphome/issues] Error Compiling (Issue #3704)
Datum: 07. Dezember 2022, 2:24
An: esphome/issues
Cc: pvheater, Comment

I was able to set the process limit in the config file but still got an 
error during compile...every time.

I installed esphome on my PC and write/compile from there, rather than 
within HASS. Works. Happy. I'm past the hassle.

—
<https://github.com/esphome/issues/issues/3704#issuecomment-1340246981> ,