esphome/issues #3705: XPT2046 Touchscreen binary sensor stays on

Zoogara, 2022-10-19:
### The problem

Changed behaviour in 2022.10.0, since XPT2046 touchscreen rolled into sensor component.

On first touch, binary sensor turns ON.  In previous versions the sensor would then turn OFF, under 2022.10.0 the binary sensor stays ON and no further touch events are recorded:

[13:57:03][D][light:035]: 'ILI9341 Display Backlight' Setting:
[13:57:03][D][light:046]:   State: OFF
[13:57:03][D][light:084]:   Transition length: 1.0s
[13:57:05][D][binary_sensor:036]: 'touch_key0': Sending state ON
[13:57:05][D][light:035]: 'ILI9341 Display Backlight' Setting:
[13:57:05][D][light:046]:   State: ON
[13:57:05][D][light:084]:   Transition length: 1.0s
[13:57:05][D][homeassistant.text_sensor:017]: 'sensor.wind_direction': Got state '55'
[13:57:05][D][text_sensor:067]: 'Current Wind Dir': Sending state '55'
[13:57:05][D][homeassistant.text_sensor:017]: 'sensor.wind_speed': Got state '5.4'
[13:57:05][D][text_sensor:067]: 'Current Wind Speed': Sending state '5.4'
[13:57:26][D][homeassistant.text_sensor:017]: 'sensor.wind_direction': Got state '40'

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.4

### What platform are you using?

ESP32

### Board

ESP32 DEVkit

### Component causing the issue

Touchscreen binary sensor (XPT2046)

### Example YAML snippet

```yaml
# touchscreen - default calibration as screen is just one big button.
touchscreen:
  platform: xpt2046
  id: touchscreen1
  cs_pin: 21
  interrupt_pin: 27
  update_interval: 50ms
  report_interval: 1s
  threshold: 400
  calibration_x_min: 280
  calibration_x_max: 3860
  calibration_y_min: 3860
  calibration_y_max: 280
  swap_x_y: false

binary_sensor:
  - platform: touchscreen
    id: touch_key0
    x_min: 0
    x_max: 240
    y_min: 0
    y_max: 320
    on_press:
      then:
        # if backlight on, then swap screen page.
        # if backlight off - wake up and display page 1
        - lambda: |-
            id(idle_time) = id(my_time).now().timestamp;
        - if:
            condition:
              light.is_on: back_light
            then:
              if: # etc more code follows
```

### Anything in the logs that might be useful for us?

```txt
[13:57:03][D][light:035]: 'ILI9341 Display Backlight' Setting:
[13:57:03][D][light:046]:   State: OFF
[13:57:03][D][light:084]:   Transition length: 1.0s
[13:57:05][D][binary_sensor:036]: 'touch_key0': Sending state ON
[13:57:05][D][light:035]: 'ILI9341 Display Backlight' Setting:
[13:57:05][D][light:046]:   State: ON
[13:57:05][D][light:084]:   Transition length: 1.0s
[13:57:05][D][homeassistant.text_sensor:017]: 'sensor.wind_direction': Got state '55'
[13:57:05][D][text_sensor:067]: 'Current Wind Dir': Sending state '55'
[13:57:05][D][homeassistant.text_sensor:017]: 'sensor.wind_speed': Got state '5.4'
[13:57:05][D][text_sensor:067]: 'Current Wind Speed': Sending state '5.4'
[13:57:26][D][homeassistant.text_sensor:017]: 'sensor.wind_direction': Got state '40'
```

### Additional information

As an aside - `interrupt_pin:` is spelt `interupt_pin:` in the doco, hover text and examples.  Not relevant to this isssue.

nielsnl68, 2022-10-19:
Hi, i am the person that changed the xpt2046.
Sadly i'm not sure what you exactly mean.  When did the binary_sensor go OFF in the old version? It should go off when you release the screen. is that not the case for you?

Zoogara, 2022-10-19:
Correct - up until this version the sensor would turn OFF when you released the screen.  It no longer does.  I currently have a workaround by publishing the OFF state in lambda.

`    on_press:
      then:
        # if backlight on, then swap screen page.
        # if backlight off - wake up and display page 1
        - lambda: |-
            id(idle_time) = id(my_time).now().timestamp;
            id(touch_key0).publish_state(false);`

nielsnl68, 2022-10-19:
You can use `on_state` event for that.

bjornprins, 2022-10-19:
@Zoogara Thanks for finding and posting this workaround. I had the same issue after upgrading to 2022.10. State did not return to OFF by itself as before 2022.10. Hopefully this can be fixed in a new release.
This is my example with the lambda workaround:
```

- platform: touchscreen
    id: touch_anywhere_on_screen
    x_min: 0
    x_max: 240
    y_min: 0
    y_max: 320
    on_press:
      - script.execute: backlight_turn_on_for_20s
      - lambda: |-
          id(touch_anywhere_on_screen).publish_state(false);
```

@nielsnl68 Using `on_state` event did not help. Every subsequent press is not doing anything with states, because the state is still ON.