esphome/issues #3707: Total number of pulses not persistent for Pulse Counter Sensor

Bascht74, 2022-10-19:
### The problem

I am using the pulse counter sensor for my gas meter. 
I used the pulse_counter.set_total_pulses service to set the total value to match my gas meter. 

If I install an ESPHome update the total value will be resetted. 
Even that the ESP32 has the possibilty to store the data persistent this is not used or offered. 

If the meter is reset and I use the pulse_counter.set_total_pulses service again the results in Home Assistant on the energy dashboard and my helper entites are strange, because they regocnize a strong gain von 0 to the high number I entered. 

Can this be corrected? 
I would be great if the integration can store the total number like the gobal variables do. The persistant storage of the total counter could be an optional parameter for the sensor.
I don't know if that would be possible on ESP8266 but Tasmota does store energy sums as well and they survive a reboot / firmware upgrade.

If that isn't an error but more like an feature request, please let me know. 
As a work-a-round I am wondering if I could store the total numer of pulses in a global variable and restore it on boot of the device? 

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

not relevant

### What platform are you using?

ESP32

### Board

not relevant

### Component causing the issue

Pulse Counter Sensor

### Example YAML snippet

```yaml
sensor:
  - platform: pulse_counter
    id: gasverbrauch_aktuell
    name: "Gasverbrauch aktuell"
    pin:
      number: GPIO25
      mode:
        input: true
        pulldown: true
    count_mode:
      rising_edge: Increment
      falling_edge: Disable
    use_pcnt: true
    unit_of_measurement: "m³"
    state_class: "measurement"
    device_class: "gas"
    filters:
      - multiply: 0.01
    total:
      name: "Gasverbrauch gesamt"
      id: gasverbrauch_gesamt
      # persistent: true <- something like this would be very helpfull
      unit_of_measurement: "m³"
      state_class: "total_increasing"
      device_class: "gas"
      accuracy_decimals: 2
      filters:
        - multiply: 0.01
```

### Anything in the logs that might be useful for us?

```txt
./.
```

### Additional information

Great integration by the way. I really like the pulse_counter integration!

Diddlik, 2022-10-23:
I think you should use globals:
globals:
  - id: total_pulses
    type: int
    restore_value: true

so you can define template sensor:

binary_sensor:
  - platform: gpio
    id: internal_pulse_counter
    pin:
      number: D5
      mode: INPUT_PULLUP
      inverted: true
    name: "Live-Impuls"
    filters:
      - delayed_on: 10ms
    on_press:
      then:
        - lambda: id(total_pulses) += 1;

sensor:

  - platform: mqtt_subscribe
    name: "GasCounter Manual Set"
    id: set_total_pulses
    topic: $mqqt_topic/$devicename/setpulses
    on_value:
      then:
        - globals.set:
            id: total_pulses
            value: !lambda 'return x;'

  - platform: template
    name: "Gasverbrauch"
    device_class: gas
    unit_of_measurement: "m³"
    state_class: "total_increasing"
    icon: "mdi:fire"
    accuracy_decimals: 2
    lambda: |-
      return id(total_pulses) * 0.01;

oliverrahner, 2022-11-07:
I have the same issue.
I am very new to ESPHome, so please forgive any seemingly obvious oversight.

If I used the solution suggested by @Diddlik, I would loose all the benefits of using `pulse_counter`, so I'd rather not go that way. Or can I just move the same context over to `pulse_counter` instead of using plain `gpio`?

ssieb, 2022-11-07:
If you're using the energy dashboard, you shouldn't need to save the total.  The dashboard will detect the reset to 0 and continue adding from there.

ssieb, 2022-11-07:
Another option is to store the value in a global variable and restore it at boot.

jendrush, 2022-12-27:
Before I started using Home Assistant, I used(still active) Domoticz. In Domoticz I had an energy and water meter implemented using pulse counting with MySensors nodes. It worked really fine. In case of a restart, such a node would query Domoticz what the last state was and count from that point. Something like this is missing from HA and ESPHome for me. Maybe I missed something?

ageurtse, 2023-01-03:
i'm ran imto the same that a total value isn't restored after flashing or rebot/power failure.
so i went looking into a solution where the internal memory doesn't die after 10k writes.

feel free to look at https://github.com/ageurtse/ESP-Home-SO-counter-with-persistant-memory-module for the solution i have found. currently in the process to make code beter and also building a board for the hardware.