esphome/issues #3708: Rename failed and config lost

fencer, 2022-10-20:
### The problem

I used the rename function in the main ESPHome screen and it went ahead and compiled a new bin. For some reason wireless upload failed so I pressed the "Retry" button. This took me back to the rename screen but upon clicking on "Rename" dialogue went away and so did the device. 
I can still control the device from Home Assistant but cannot see it in ESPHome.

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.4

### What platform are you using?

ESP8266

### Board

sonoff

Olen, 2022-11-16:
Not sure it is related, but I have seen that when using mDNS and the rename-function, it seems like esphome will try to connect to the new hostname while applying the new config (which obviously don't work, as that name is not available in mDNS).
Also, unless you have 
```
esphome:
  name_add_mac_suffix: false
```
There will be great confusion and mismatch between what the name of the device is, what the mDNS name is and what your config is called, which again might lead to a new, blank config can be created.

Shadow-351, 2025-03-14:
This just happened to me, any update on this bug? ESPHome notified me that I had an underscore in a hostname and said it would cause problems. I used the "Rename Hostname" function to replace the underscore with a hyphen. I then tried to upload the config which failed mDNS and then ESPHome DELETED my YAML. This really needs to be fixed and the upload tool should allow manually entering the devices IP to upload.