esphome/issues #3709: XPT2046 touchscreen code broken

ro-explorer, 2022-10-20:
### The problem

Using interrupt pin is only registering the first touch, then the touchscreen it is simply dead.
Commenting out interrupt_pin and using just spi pooling it is registering touches but then ghost touches are showing up, always with random X/Y coordinates. 

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2022.10.3

### What platform are you using?

ESP32

### Board

mini D1 esp32

### Component causing the issue

xpt2046

### Example YAML snippet

```yaml
spi:
  clk_pin: GPIO21
  mosi_pin: GPIO22
  miso_pin: GPIO23

touchscreen:
  platform: xpt2046
  id: xpt2046_touchscreen
  cs_pin: 27
  interrupt_pin: 25
  update_interval: 20ms
  report_interval: never
  threshold: 200
  calibration_x_min: 260
  calibration_x_max: 3800
  calibration_y_min: 3900
  calibration_y_max: 380
  swap_x_y: false
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              id(xpt2046_touchscreen).x,
              id(xpt2046_touchscreen).y,
              id(xpt2046_touchscreen).x_raw,
              id(xpt2046_touchscreen).y_raw
              );

display:
  - platform: ili9341
    model: TFT 2.4
    cs_pin: GPIO5
    dc_pin: GPIO18
    reset_pin: GPIO19
    update_interval: 20ms
    id: my_lcd
```

### Anything in the logs that might be useful for us?

```txt
********** With interrupt_pin used ...

[17:55:18][I][app:102]: ESPHome version 2022.10.0 compiled on Oct 20 2022, 17:53:49
[17:55:26][I][cal:060]: x=111, y=150, x_raw=1904, y_raw=2249

That's it - no other registered touches.

********** With interrupt_pin commented out in the code above ...

Real touches (corners of the screen):
[17:58:53][I][cal:060]: x=0, y=0, x_raw=271, y_raw=3919
[17:59:02][I][cal:060]: x=0, y=320, x_raw=235, y_raw=372
[17:59:04][I][cal:060]: x=240, y=320, x_raw=3815, y_raw=371
[17:59:07][I][cal:060]: x=240, y=1, x_raw=3812, y_raw=3879

Ghost touches randomly showing up after leaving the screen alone
[17:59:11][I][cal:060]: x=146, y=150, x_raw=0, y_raw=0
[17:59:22][I][cal:060]: x=147, y=149, x_raw=2440, y_raw=2257
[17:59:46][I][cal:060]: x=145, y=152, x_raw=2411, y_raw=2220
[18:00:31][I][cal:060]: x=191, y=235, x_raw=0, y_raw=0
[18:00:45][I][cal:060]: x=199, y=236, x_raw=3198, y_raw=1301
[18:00:49][I][cal:060]: x=151, y=100, x_raw=2491, y_raw=2792
[18:00:55][I][cal:060]: x=193, y=239, x_raw=3112, y_raw=1264
[18:01:06][I][cal:060]: x=192, y=233, x_raw=3105, y_raw=1336
[18:01:12][I][cal:060]: x=144, y=110, x_raw=2391, y_raw=2680
[18:01:23][I][cal:060]: x=149, y=177, x_raw=2468, y_raw=1946
[18:01:29][I][cal:060]: x=142, y=160, x_raw=2363, y_raw=2130
[18:01:40][I][cal:060]: x=147, y=146, x_raw=2436, y_raw=2287
[18:01:44][I][cal:060]: x=146, y=153, x_raw=0, y_raw=0
[18:01:50][I][cal:060]: x=146, y=144, x_raw=2426, y_raw=2316
[18:02:02][I][cal:060]: x=150, y=176, x_raw=2475, y_raw=1959
```

bjornprins, 2022-10-21:
This is a know issue, see https://github.com/esphome/issues/issues/3705.
The state is not cleared after touching the screen.

ro-explorer, 2022-10-21:
Thanks for the input - using report_interval solved my problem. Still, using SPI pooling is still generating random ghost touches.
I'll stick to IRQ and report interval then,

nielsnl68, 2022-10-21:
I did the conversion to the new touchscreen platform.

Sadly i do not have the luxury to test the IRQ based on the info i got from the datasheet i suspected that the IRQ flag would got triggered as long the touchscreen was touched. From your info i seems this is not the case is it?

Do you have any way to show what happens to the IRQ and BUSY pins in respect to the CS and DOUT pins.

ro-explorer, 2022-10-21:
Sure - let me set-up a rig to measure them.

ro-explorer, 2022-10-21:
Back with some info: The BUSY pin on the XPT2046 controller is not wired to anything on the LCD interface. I can tap into it and measure its behaviour but I really do not understand what value will this add to the functionality.

In the datasheet the BUSY pin will just go high impedance when CS is not activating the interface. 

On the other hand, on the adafruit library, I found this piece of info:

"The interrupt pin always outputs a digital signal related to the touch controller signals, which is LOW when the display is touched. It also is driven low while software reads the touch position.
[...]
However, inside your interrupt function, if the display is no longer being touched, any attempt to read the touch position will cause the interrupt pin to create another falling edge. This can lead to an infinite loop of falsely triggered interrupts. Special care is needed to avoid triggering more interrupts on the low signal due to reading the touch position."

So, I guess this is what is happening here maybe? A continuous reading of the X-Y keeping the chip in the IRQ-LOW state and also generating ghost touches on spi pooling ? Just thinking out loud here. 

I'll do a rig to capture those 4 pins and graph them out (I'll be using the very ESP and LCD with the touch for this).

nielsnl68, 2022-10-21:
Yeah, there was indeed a bug while using the interrupt pin. Where the release action was not send at all.

From what i did understand from the datasheet was that you should wait reading the values like X and Y axis until the BUSY flag was down again after sending the command. 

I haven't checked the code from adafruit but i did checked some other libraries and they do almost the same thing with the interrupt pin, i belief that part should work.

I created a new PR fixing the issue and removed some redounded code at the same time, I hope you can check it out,
@ro-explorer and @bjornprins

ro-explorer, 2022-10-21:
Your black magic worked like a charm. I tested your modifications via external_components and everything is peachy. 
Many thanks. Next beer/coffee is on me !

nielsnl68, 2022-10-21:
check my github page and you find a way to send them to me ;)

ro-explorer, 2022-10-21:
Already did that hours ago :)

nielsnl68, 2022-10-21:
Thank you very much.

ro-explorer, 2022-10-24:
Sorry for the late reply - I do not see any ghost touches with SPI polling but in that scenario I get weird readings on the raw X/Y (while adjusted real X/Y remain pretty accurate. See x_raw, y_raw with 0 value below (from the log with IRQ commented out).

[17:16:00][I][cal:062]: x=150, y=189, x_raw=2464, y_raw=1807
[17:16:00][I][cal:062]: x=151, y=190, x_raw=0, y_raw=0
[17:16:00][I][cal:062]: x=149, y=205, x_raw=0, y_raw=0
[17:16:00][I][cal:062]: x=153, y=191, x_raw=2468, y_raw=1807
[17:16:01][I][cal:062]: x=150, y=191, x_raw=0, y_raw=0
[17:16:01][I][cal:062]: x=152, y=193, x_raw=0, y_raw=0
[17:16:01][I][cal:062]: x=150, y=192, x_raw=2476, y_raw=1784
[17:16:01][I][cal:062]: x=121, y=171, x_raw=0, y_raw=0
[17:16:02][I][cal:062]: x=148, y=191, x_raw=0, y_raw=0
[17:16:02][I][cal:062]: x=149, y=193, x_raw=2467, y_raw=1791
[17:16:03][I][cal:062]: x=150, y=193, x_raw=0, y_raw=0
[17:16:03][I][cal:062]: x=149, y=192, x_raw=0, y_raw=0
[17:16:03][I][cal:062]: x=150, y=190, x_raw=0, y_raw=0

ro-explorer, 2022-10-24:
Oups .. while I was writing the above, 5 ghost touches happened (so SPI polling now really working still - IRQ is spot-on though):

[17:16:51][I][cal:062]: x=137, y=232, x_raw=2284, y_raw=1345
[17:16:58][I][cal:062]: x=135, y=233, x_raw=0, y_raw=0
[17:18:20][I][cal:062]: x=135, y=231, x_raw=0, y_raw=0
[17:19:04][I][cal:062]: x=126, y=193, x_raw=0, y_raw=0
[17:19:06][I][cal:062]: x=138, y=233, x_raw=0, y_raw=0

ro-explorer, 2022-10-24:
I deleted the previous 2 entries where I was describing some weird behaviour but it was all fixed by increasing the touchscreen threshold to 400.
All good with SPI polling as well. 

Conclusion, the code is good to go.

nielsnl68, 2022-10-24:
Perfect. thanks for reporting

closed 2022-10-25 by jesserockz