esphome/issues #3710: Strobe Effect Expected Color Temperature with unit, got 100%.

SamPhoenix91, 2022-10-20:
### The problem

When trying to make a strobe effect including colour temperature;

`          colors:
            - state: true
              brightness: 100%
              color_temperature: 100%
              duration: 1s
`

It reports "Expected Color Temperature with unit, got 100%."
It happens with any percentage as far as I can tell and I cannot work out what input it will accept

### Which version of ESPHome has the issue?

v2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.1

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

light (hbridge)

### Example YAML snippet

```yaml
effects:
      - random:
      - strobe:
          name: Alternate
          colors:
            - state: true
              brightness: 100%
              color_temperature: 100%
              duration: 1s
            - state: true
              brightness: 100%
              color_temperature: 0%
              duration: 1s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/light-test.yaml...
Failed config

light.hbridge: [source /config/esphome/light-test.yaml:58]
  platform: hbridge
  id: mainlight
  name: Hbridge Lights
  pin_a: pina
  pin_b: pinb
  effects: 
    - random: {}
    - strobe: 
        name: Alternate
        colors: 
          - state: True
            brightness: 100%
            
            Expected Color Temperature with unit, got 100%.
            color_temperature: 100%
            duration: 1s
          - state: True
            brightness: 100%
            color_temperature: 0%
            duration: 1s
    - strobe: 
    - lambda: 
        name: Breathing
        update_interval: 4s
        lambda: |-
          static int state = 0;
          static int color = 1;
          auto call = id(mainlight).turn_on(); //the id of your light
          call.set_transition_length(3000);
          if (state == 0) 
          {
          call.set_brightness(1.0);
          }
          else if (state == 1)
          {
          call.set_brightness(0.01);
          }
          call.perform();
          state ++;
          if (state == 2){
          state = 0;
          }
```

bkaufx, 2022-10-20:
I'm sure it needs either mireds or Kelvin (K) as the units.

SamPhoenix91, 2022-10-20:
> I'm sure it needs either mireds or Kelvin (K) as the units.

I've checked and it does accept Kelvin at least, however documentation highly suggests that % is an acceptable format, either documentation needs changing or it is a bug.

[image]

bkaufx, 2022-10-20:
Huh. What pages are those screenshots from? If it's supposed to take a percentage it probably wants a float between 0.0 and 1.0.

SamPhoenix91, 2022-10-20:
> What pages are those screenshots from?

One is the prompt VS Code gives when hovering over the code
The other is from the Light Effects page on ESPHome https://esphome.io/components/light/index.html#strobe-effect

Also note that  brightness is expressed as `brightness: 100%` and it is accepted perfectly fine

bkaufx, 2022-10-20:
Yea I bet that is a typo in the documentation. I'll try to look at the code.

bkaufx, 2022-10-20:
Yea definitely a typo in the documentation.  This field uses the same config validator as all other color_temp fields and requires mireds or Kelvin.  I created a couple pull requests to fix this in the documentation and the VS code plugin.  Hopefully I did it right.

https://github.com/esphome/esphome-docs/pull/2375
https://github.com/esphome/esphome-vscode/pull/64

SamPhoenix91, 2022-10-20:
Awesome, thanks for the help