esphome/issues #3711: ESP32-C3-mini-1 OTA Update fails with "Error: Finishing update failed."

tilatrivia, 2022-10-21:
### The problem

OTA updates to an ESP32-C3-mini-1 are failing with the error "ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information." and the said device logs output "Error ending OTA!, error_code: 132" No other details about the error can be obtained other than it is something to do with the end of the upload process, and it may have to do with the MD5 hash not matching.

The error is being returned by [ota_backend_arduino_esp32.cpp:36] ArduinoESP32OTABackend::end() when [Updater.cpp:278] UpdateClass::end() returns false. The UpdateClass::end() function could return false because of an existing error, premature end, MD5 mismatch, or [Updater.cpp:249] UpdateClass::_verifyEnd() returning false. UpdateClass::_verifyEnd() could return false because of a partitioning issue. I cannot determine which issue it actually is because this code is part of the Arduino package and its actual error value is ignored.

I have tried newer versions of the Arduino platform (version: 2.0.3, platform_version: 4.4.0) but this is the only combo that successfully compiles, all others giving a variety of errors. The esp-idf platform won't compile as the web server and captive portal require Arduino.

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

ota

### Example YAML snippet

```yaml
esphome:
  name: vf-1
  platformio_options:
    board_build.variant: esp32c3
    board_build.f_flash: 40000000L # OTA fix?
    board_build.flash_mode: dio
    board_build.flash_size: 4MB # OTA fix?
  includes:
    - vf_light.h

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0

# Backend
logger:
  level: debug

debug:
  update_interval: 5s

text_sensor:
  - platform: debug
    device:
      name: "Device Info"

sensor:
  - platform: debug
    free:
      name: "Heap Free"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"

output:
(gpio pins)

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "vf-1"
    password: "vf"

captive_portal:

web_server:
  port: 80

# Components
light:
(custom light component from vf_light.h)
```

### Anything in the logs that might be useful for us?

```txt
### Upload Logs

> esphome upload .\vf-1.yaml
INFO Reading configuration .\vf-1.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Resolving IP address of vf-1.local
INFO  -> 192.168.1.16
INFO Uploading .\.esphome/build/vf-1\.pioenvs\vf-1\firmware.bin (889776 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
```

### Device Log
```
21:08:45 | [D] | [ota:143] | Starting OTA Update from ::FFFF:192.168.1.9...
21:08:46 | [D] | [ota:312] | OTA in progress: 0.1%
21:08:47 | [D] | [ota:312] | OTA in progress: 7.8%
21:08:48 | [D] | [ota:312] | OTA in progress: 14.8%
21:08:49 | [D] | [ota:312] | OTA in progress: 21.2%
21:08:50 | [D] | [ota:312] | OTA in progress: 23.0%
21:08:51 | [D] | [ota:312] | OTA in progress: 27.7%
21:08:52 | [D] | [ota:312] | OTA in progress: 35.2%
21:08:53 | [D] | [ota:312] | OTA in progress: 42.8%
21:08:54 | [D] | [ota:312] | OTA in progress: 47.5%
21:08:55 | [D] | [ota:312] | OTA in progress: 54.3%
21:08:56 | [D] | [ota:312] | OTA in progress: 62.1%
21:08:57 | [D] | [ota:312] | OTA in progress: 69.7%
21:08:58 | [D] | [ota:312] | OTA in progress: 74.5%
21:08:59 | [D] | [ota:312] | OTA in progress: 79.6%
21:09:01 | [D] | [ota:312] | OTA in progress: 86.9%
21:09:02 | [D] | [ota:312] | OTA in progress: 93.9%
21:09:03 | [D] | [ota:312] | OTA in progress: 96.5%
21:09:04 | [W] | [ota:328] | Error ending OTA!, error_code: 132
```

### Device Statistics
```
[21:20:40][D][debug:061]: ESPHome version 2022.10.0
[21:20:40][D][debug:065]: Free Heap Size: 209280 bytes
[21:20:40][D][debug:095]: Flash Chip: Size=0kB Speed=0MHz Mode=UNKNOWN
[21:20:40][D][debug:132]: Chip: Model=UNKNOWN, Features=WIFI_BGN,BLE, Cores=1, Revision=3
[21:20:40][D][debug:140]: ESP-IDF Version: v4.4.1-1-gb8050b365e
[21:20:40][D][debug:145]: EFuse MAC: 34:B4:72:B2:99:80
[21:20:40][D][debug:199]: Reset Reason: Power On Reset
[21:20:40][D][debug:244]: Wakeup Reason: Unknown
```
```

M1Sports20, 2022-12-07:
I am also seeing this problem, but am still debugging.  I feel like I need to setup partitions or something.  Still learning the esp32c3

mihsu81, 2022-12-20:
I've encountered the same issue on a **_lolin_c3_mini_**.
What solved it was flashing the factory bin with esptool v3.3 using the following command:
```.\esptool.exe --chip esp32c3 --port COM13 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_size detect 0x0 ble-gateway05-factory.bin```.
Adjust the port and firmware name accordingly.

hgelpke, 2022-12-29:
I have this same issue with a switchbot bulb. Even compiling manually and uploading via the web server fails (using both modern and legacy formats)

apnar, 2023-01-31:
Same issue on a Switchbot Plug Mini.  Had been working well on esp-idf including OTA updates.  I tried arduino (to get the web server component) which seemingly works well but now I can't seem to update it at all.

mellis, 2023-08-23:
I also have a Switchbot Plug that I can no longer OTA after flashing with onto arduino. @apnar were you able to fix your plugs with OTA?

Globolo001, 2024-03-29:
Any updates regarding this problem? I have the same on my S3-Zero

rtgree01, 2024-07-19:
I too am having problems updating via OTA with my Switchbot Plugs.  Previously I had Tasmota on them and could not do an OTA switch to ESPHome. I had to pop them open and flash with a serial port. Could that have anything to do with this problem?
First time I saw this problem was with ESPHome 6.4. I've since updated to 7.0 and still have the problem. I tried removing the captive portal and web server and using the esp-idf framework. No success.

apnar, 2024-07-19:
I never got my switch on ardunio to OTA update.  I have a handful of other ones on esp-idf though and they OTA update perfectly from within ESPHome.