esphome/issues #3712: Wifi Password containing causes error when installing updates

BobRathke, 2022-10-21:
### The problem

After recently changing my wifi password I can no longer update my esp32 boards. After changing secrets Validation completes with warning and completes the initial upload to the board. after reboot board will not connect to wifi. all boards where running fine for weeks before changing my wifi password.

Validation warnings included along with secrets file.

**secrets file:**
# Your Wi-Fi SSID and password
wifi_ssid: "TaxGuest"
wifi_password: 'TaxGuest$2022'

**validation log:**
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-28d188.yaml...
WARNING Found 'TaxGuest$2022' (see wifi->password) which looks like a substitution, but '2022' was not declared
WARNING Found 'TaxGuest$2022' (see wifi->password) which looks like a substitution, but '2022' was not declared
INFO Configuration is valid!
substitutions:
  name: esp32-bluetooth-proxy-28d188
  wifi_ap_password: ''
esphome:
  name: esp32-bluetooth-proxy-28d188
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: '1.0'
  build_path: .esphome/build/esp32-bluetooth-proxy-28d188
  platformio_options: {}
  includes: []
  libraries: []
  min_version: 2022.10.0
esp32:
  board: esp32dev
  framework:
    version: 1.0.6
    source: ~3.10006.0
    platform_version: platformio/espressif32 @ 3.5.0
    type: arduino
  variant: ESP32
wifi:
  ap:
    password: ''
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  networks:
  - ssid: TaxGuest
    password: TaxGuest$2022
    priority: 0.0
  use_address: esp32-bluetooth-proxy-28d188.local
api:
  port: 6053
  password: ''
  reboot_timeout: 15min
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
ota:
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10
improv_serial: {}
captive_portal: {}
dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
    duration: 5min
    continuous: true
bluetooth_proxy:
  active: true
button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic
  disabled_by_default: false
  icon: mdi:restart-alert
  device_class: restart

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.0

### What platform are you using?

ESP32

### Board

generic

### Component causing the issue

validation installation of updates

ssieb, 2022-10-24:
I can't reproduce this.  What is your actual config?

NODeeJay, 2022-11-20:
Did you find a solution?

The problem is maybe the dollar sign, as it is used in substitutions as well - basically that's what the waring tells you. Do you mind giving it a try with either changing your WiFi password to something without a $ sign or adding a second WiFi with a password that does not have a $.

ssieb, 2022-11-20:
I can't even reproduce the warning.  See what the serial logs have to say about it not connecting to wifi.

BobRathke, 2022-11-20:
Please confirm when I should dump the logs. I should change the secrets password and then validate and dump the serial logs?

BobRathke, 2022-11-20:
I created an additional wifi net without a "$" in the pass word and everything works as expected. I have 50+ devices so  I would rather not have to change them all. A second wifi network is also not ideal.

NODeeJay, 2022-11-20:
Well, especcially the dollar is an issue as it is used in substitutions. You may play around with different parenthesis to get it working but in the end the easiest solution is to replace the dollar by another special character (maybe exclude: # $ ! ' " () {} [] - there should be still enough special chars left to fulfill a proper password security).
And regarding the 50+ devices, this is a good reminder for proper change management and tests before migrating :-)

Or/And regarding the 2nd WiFi: It's not a bad practice to separate IoT devices from the rest with their own WiFi and subnet for a couple of reasons (ESPs don't like a couple of WiFi security settings like WPA3 (yet), when a device got taken over or goes rogue it does not spill into your internal network, reduce traffic to ESP as broadcasts are only per subnet thus increase battery/decrease energy consumption, etc.)

ssieb, 2022-11-20:
> Please confirm when I should dump the logs. I should change the secrets password and then validate and dump the serial logs?

With the wifi secrets set to the value that isn't working, watch the serial logs to see what's happening.

ssieb, 2022-11-20:
> Edit: it would be a nice PR to update the docs with the caracters not to be used in the password

There's no problem.  I've tested it with a `$` and it worked.  I don't even get a warning.

ssieb, 2022-11-20:
@BobRathke also which esphome version are you using?  Those numbers don't look correct.

NODeeJay, 2022-11-20:
I tested @BobRathke's config after removing the duplicity in ESPHome dashboard (official HASS addon) and locally with Windows 10, Python 3.10.4 & ESPHome 2022.12.0-dev get both times the error:
```
INFO Reading configuration /config/esphome/test.yaml...
Validate test.yaml ✅
INFO Reading configuration /config/esphome/test.yaml...
WARNING Found 'TaxGuest$2022' (see wifi->password) which looks like a substitution, but '2022' was not declared
```

But it seems to be used correctly in runtime and it's just cosmetical, at least regarding this thread: https://community.home-assistant.io/t/warning-found-see-wifi-password-which-looks-like-a-substitution-but-b7-was-not-declared/373129/5

@BobRathke can you just give it a try and see if the ESP connects? And if you could check the serial log as @ssieb recommended after flashing to see if it does would be good.

closed 2023-03-05 by jesserockz