esphome/issues #3714: Assigning static IP wirelessly does not work

Zocker1999NET, 2022-10-21:
### The problem

ESPHome loses access to the ESP32 when adding a "static_ip" to the config without rolling it out. It found the ESP before by using mDNS. When removing the static_ip config part, Dashboard can find the ESP again (which took about 5 minutes).

It could be really helpful if ESPHome Dashboard could fallback to mDNS discovery if the ESP cannot be reached by its static IP so it is possible to rollout a static IP configuration wireless.

Due to an issue with another ESP8266, I run ESPHome Dashboard with `ESPHOME_DASHBOARD_USE_PING=true`, but that didn't prevent the dashboard from detecting the ESP32 without a static IP.

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu v2

### Example YAML snippet

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # manual IP (stored in FB)
  manual_ip:
    static_ip: 10.11.11.84
    gateway: 10.11.11.1
    subnet: 255.255.255.0
    dns1: 10.11.11.10
```

Fischelsberger, 2022-10-26:
You normally need to add "use_address" [see here](https://esphome.io/components/wifi.html#:~:text=server%20to%20use.-,use_address,-(Optional%2C%20string)).

so it should be like:
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # manual IP (stored in FB)
  use_address: the.old.ip.address
  manual_ip:
    static_ip: 10.11.11.84
    gateway: 10.11.11.1
    subnet: 255.255.255.0
    dns1: 10.11.11.10
```
Just compile and upload it afterwards.
After the Device is connected again you need to remove "use_address" again and need to reupload it again.

ssieb, 2022-10-26:
You don't need to re-upload after removing the `use_address`.  That parameter is only for the dashboard, not the device.

Zocker1999NET, 2022-11-02:
Oh, didn't read about the `use_address` config variable, good to know, thanks.

Then this may still be a feature request, if this is feasible to you, because it would improve the user experience when adding a static IP.

Or another idea could be to inform the user about `use_address` if he/she/it added `static_ip` recently and the ESP dashboard has problems on finding the ESP by itself (if that is easy enough to detect).

closed 2023-03-18 by github-actions[bot]