esphome/issues #3715: Generation of encryption key does not work (in popup)

RubenKelevra, 2022-10-22:
### The problem

When hovering the cursor over the encryption key section in the yaml file, a popup is supposed to generate an encryption key in the browser and show it for easy local generation.

Instead a link is presented an something followed by some syntax issues in the source code creating "[[[":

[image]

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

ESPHome's yaml editor

### Example YAML snippet

```yaml
Does not matter.
```

### Anything in the logs that might be useful for us?

```txt
No
```

RubenKelevra, 2023-02-20:
Please deactivate that stupid stale bot which is closing issues which haven't even be read by the team -.-

ChristianWaechter, 2023-12-04:
This is still not fixed!