esphome/issues #3717: CL-S10 i2c issues and no devices discovered in HA

botts7, 2022-10-22:
### The problem

After installing Bluetooth-proxy on CL-S10 and setting it up in Home Assistant

It is showing as connected etc, but get the following in the logs. 

I have hardware revision v1.0

[00:29:12][C][i2c.arduino:038]: I2C Bus:
[00:29:12][C][i2c.arduino:039]:   SDA Pin: GPIO15
[00:29:12][C][i2c.arduino:040]:   SCL Pin: GPIO13
[00:29:12][C][i2c.arduino:041]:   Frequency: 50000 Hz
[00:29:12][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[00:29:12][I][i2c.arduino:054]: Results from i2c bus scan:
[00:29:12][I][i2c.arduino:056]: Found no i2c devices!

After about 3 or 4 minutes I get additional logs, but no devices are found

[00:34:07][I][ota:109]: Boot seems successful, resetting boot loop counter.
[00:34:07][D][esp32.preferences:113]: Saving 1 preferences to flash...
[00:34:07][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[00:34:08][D][esp32_ble_tracker:264]: Starting scan...

Any ideas?

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP32

ssieb, 2022-10-25:
I'm not sure what you're expecting from filing this issue.  This isn't a bug report.  You should ask in the esphome Discord server or one of the forums.
Are there devices within range?
What is your config?

botts7, 2022-10-25:
After further investigation it is working. The reason for thinking there was an issue was the esp logs and intergration does not provide any user feedback that there is any sensor data being transmitted to home assistant. After removing devices that were added via other integrations and removing any other Bluetooth intergrations. I was able to confirm it was working. Turning on debug logs for Bluetooth in HA I was able to confirm it was functioning correctly. I would suggest some user confirmation of transmission of the data within the epshome logs. The only information the logs give is that the esp device is being seen by esphome. Not showing what sensor values are being captured

ssieb, 2022-10-25:
The proxy doesn't decode any data.  That's the point.  It's just passing everything to HA.  It should be showing you in the log the devices it sees though.

botts7, 2022-10-25:
Yes understood!

botts7, 2022-10-25:
Yes understood.  Still think there should be some kind of user confirmation that the undecided data have been passed on to HA ( verbose log seems to kind off show that)

closed 2022-10-25 by botts7