esphome/issues #3720: web_server.local: true -> "Uncaught ReferenceError: assignment to undeclared variable url"

jangrewe, 2022-10-23:
### The problem

When enabling `web_server.local` including the JS/CSS assets and going to the web interface, the page is blank and the console shows a Javascript error:

```
 Uncaught ReferenceError: assignment to undeclared variable url
    st http://box.local/:1
    <anonymous> http://box.local/:1
growbox.local:1:28727
    st http://box.local/:1
    <anonymous> http://box.local/:1
```

I also tested this on a different device running `2022.9.0-dev`, which worked just fine on that version, but after updating it to `2022.10.0` it showed the same issue.

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

web_server

### Example YAML snippet

```yaml
web_server:
  port: 80
  local: true
```
```

zibous, 2022-10-26:
Same Problem here:
`esphome-2022.10.1`

**DO NOT WORK**
```
web_server:
  port: 80
  version: 2
  local: true
```

`local: false` is working.

pruwait, 2022-10-27:
Same Problem.
EspHome 2022.9.4 is working with local: true

RoboMagus, 2022-10-27:
It seems that the current workflow for [generating the header](https://github.com/esphome/esphome-webserver/blob/main/scripts/make_header.sh) used in `local: true` deployment of the webserver does not include the js file itself.

foosinn, 2022-11-07:
Same issue here, thanks for sharing!

closed 2022-11-09 by jesserockz