esphome/issues #3722: GPIO binary sensor initial state is Unknown when using filters and delayed_off

spikeygg, 2022-10-24:
### The problem

I am using a binary sensor PIR GPIO like this:

```
binary_sensor:
  - platform: gpio
    pin: D2
    device_class: motion
    name: "Motion detected in Garage"
    publish_initial_state: true
    filters:
      - delayed_off: 5min
```

and I noticed that when I reboot the device, it comes up with `Unknown` state. When I comment out the `filters:` section, it comes up as `Clear`. I feel like this is a bug.

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

PIR sensor

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    pin: D2
    device_class: motion
    name: "Motion detected in Garage"
    publish_initial_state: true
    filters:
      - delayed_off: 5min
```

degrashopper, 2023-02-10:
I can confirm this. Having the same problem when using delayed_off

clau-bucur, 2023-07-24:
Same here. Seems only affected by `delayed_off`.

Was strugling with this issue for so long, trying to add some `on_boot` trigger to re-publish the entity state but with no luck.
Then commenting out the `delayed_off` filter fixed the issue.

Clearly looks like a bug:
```
binary_sensor:
  - platform: gpio
    name: Weather raining ESP
    id: weather_raining_esp
    device_class: moisture
    pin:
      number: 5
      inverted: true
      mode:
        input: true
    filters:
       - delayed_on: 10s
      #  - delayed_off: 600s
```

masterd89, 2023-08-28:
Can confirm too ( on 2023.6.3 ). Strange thing is that i have binary inputs with 1s off delay which don't show this problem.
Also, as suggested, `publish_initial_state: true` seems to fix the issue. Having the initial state being reported after delay is correct: if you are setting a off delay, you want the sensor to be in off state for at least that time before updating it's state, and so at startup you need to wait that time to know in which state your input is!

Bascht74, 2024-02-12:
I have the same problem, but without using any filters. 
My HA sensor is "off", and ESPHOME reports is as "unknown", until the state is changed to "on", after that everything works as expected...

ssieb, 2024-02-12:
> I have the same problem, but without using any filters. My HA sensor is "off", and ESPHOME reports is as "unknown", until the state is changed to "on", after that everything works as expected...

That is not the same problem.  This issue is about GPIO binary sensors, not HA.

ssieb, 2024-02-12:
@Bascht74 create a new issue or come ask on Discord.  Don't hijack another issue.

illigtr, 2024-02-20:
I have exactly the same issue.  Using filters causes binary sensors to start off in an "unknown" state. Removing the filters and the sensors works as it should. This happens with gpio *and* template platforms of the binary sensor. I have tried on_boot to publish an initial state, but that simply does not work. I have no workaround to offer until the code is fixed.

illigtr, 2024-02-20:
I have found a workaround solution that works for me. Insert TWO invert in your filter at the top, as these are pipe-lined. The problem currently is that the default state at boot appears to be "false", even though the GUI shows "unknown". This means that the delayed_off timer is RUNNING...

I tested this by testing with boot_up code, like this:

`#  on_boot:
#    priority: -100
#    then:
#      - delay: 1s
#      - lambda: id(motion).publish_state(true);
#      - delay: 1s
#      - lambda: id(motion).publish_state(false);
`

If one attempts to publish a true state, it works! If you publish a false state it won't! It appears that delay_off filter CANNOT be interrupted. Publishing as state is like the sensor changing value BEFORE hitting the filter pipeline... Delay_off ignores any state changes until the timeout value is reached.

Now for the workaround. Here is my example, tested and working as expected. 

`  - platform: template
    name: "Motion"
    device_class: motion
    publish_initial_state: true
    id: motion
    filters:
      - invert: 
      - invert:
      - delayed_off: 5 min`

Enjoy!

illigtr, 2024-02-20:
Ok, after several failed attempts, I now have a functional workaround. Not pretty, but workarounds usually aren't.

The issue with previous workaround attempts is that we simply cannot intercept or cancel the delay_off: filter. However, we CAN modify the value of the delay_off: with lambda. 

To do this however, we need to capture when the entity rebooted. The easiest was the up_time sensor. It returns the number of seconds since last reboot.

Ok, Good so far, but still doesn't work. The last part of the puzzle is to force the state of the binary sensor twice. We can do this in boot_up.

So here is the complete, functional code:

```
esphome:
  ...
  on_boot:
    priority: -100
    then:
      - delay: 1s
      - lambda: id(motion).publish_state(true);
      - delay: 1s
      - lambda: id(motion).publish_state(false);

sensor:
    # Display uptime
  - platform: uptime
    name: Uptime
    id: up_time

binary_sensor:
  - platform: template
    name: "Motion"
    device_class: motion
    publish_initial_state: true
    id: motion
    filters:
#      ...old code
#      - delayed_off: 5 min

#      ...new code / workaround / up_time is in seconds, return values must be in milliseconds
       - delayed_off: !lambda "if ( id(up_time).state < 5 ) return 0; else return 5 * 60 * 1000;"

``` 

Enjoy.