esphome/issues #3723: light component default_transition_length not taking min_power into consideration

kpelzel, 2022-10-24:
### The problem

I have an rgbw light component that has ledc outputs. My white output is not able to dim lower than 10%, so I set the min_power of that output to 0.1 and set zero_means_zero to true to get the light to actually turn off. The problem is that if I toggle the light from on to off, it will dim down to 10% and sit there before finally turning off. 

I can exaggerate this issue by setting min_power to 0.5 and the default_transition_length to 2s in which case the light will dim to 50% for 1 second and sit at 50% for the next second before finally turning off. Ideally this should dim to 50% for the full two seconds and then turn off.

It seems like the transition time isn't taking min_power into consideration. 

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP32

### Board

esp32-d0wd-v3

### Component causing the issue

ledc

### Example YAML snippet

```yaml
output:
  - platform: ledc
    id: output_red
    pin: GPIO14
  - platform: ledc
    id: output_green
    pin: GPIO26
  - platform: ledc
    id: output_blue
    pin: GPIO13
  - platform: ledc
    id: output_white
    pin: GPIO15
    min_power: 0.5
    zero_means_zero: true

light:
  - platform: rgbw
    name: ${devicename}
    id: ${deviceid}
    red: output_red
    green: output_green
    blue: output_blue
    white: output_white
    default_transition_length: 2s
```

closed 2023-03-08 by github-actions[bot]

mrmultifunk, 2023-03-20:
I'm seeing the exact same thing. Did you manage to solve the problem?

kpelzel, 2023-03-20:
Nope and I haven't had time to dig into it myself unfortunately.

ssieb, 2023-03-20:
The light knows nothing of the output settings, so it can't take that into account.  Maybe the output could have an option to rescale the set value to fit between the min and max range.  For now, you could use a template output to do that.

kpelzel, 2023-03-20:
> Maybe the output could have an option to rescale the set value to fit between the min and max range.

I think this is pretty much what I was looking for. How would you use a template output to accomplish this?

ssieb, 2023-03-20:
```yaml
output:
  - platform: template
    id: fakeoutput
    write_action:
      - lambda: |-
          if (state > 0)
            state = (max_value - min_value) * state + min_value;
          id(realoutput).set_level(state);
```
Replace `max_value` and `min_value` with the actual values.

ssieb, 2023-03-21:
I checked and the output core already does exactly that calculation, so I don't know what you were seeing before.  If you use that template, you will get really weird results. :-)  If you still think it's not working, then come to the discord server and we can discuss it.

mrmultifunk, 2023-03-22:
Maybe the issue is linked to the rgbw(w) component?

ssieb, 2023-03-22:
It can't have anything to do with the light because the light only sends the level it wants.  It knows nothing about the output.

mrmultifunk, 2023-03-22:
That makes sense.

I have attached the verbose logoutput from the turn off command below, if it can be helpful in any way:

```
[22:52:10][D][light:035]: 'Tuya light ESP' Setting:
[22:52:10][D][light:046]:   State: OFF
[22:52:10][D][light:084]:   Transition length: 8.0s
[22:52:10][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[22:52:10][V][json:056]: Size after shrink 156 bytes
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62258 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19255 on channel 3
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62255 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19254 on channel 3
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62248 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19252 on channel 3
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62233 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19248 on channel 3
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62206 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19241 on channel 3
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62166 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19230 on channel 3
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62111 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19215 on channel 3
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:10][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:10][V][ledc.output:110]: Setting duty: 62036 on channel 4
[22:52:10][V][ledc.output:110]: Setting duty: 19195 on channel 3
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:11][V][ledc.output:110]: Setting duty: 61955 on channel 4
[22:52:11][V][ledc.output:110]: Setting duty: 19173 on channel 3
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:11][V][ledc.output:110]: Setting duty: 61844 on channel 4
[22:52:11][V][ledc.output:110]: Setting duty: 19143 on channel 3
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:11][V][ledc.output:110]: Setting duty: 61708 on channel 4
[22:52:11][V][ledc.output:110]: Setting duty: 19106 on channel 3
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 0
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 1
[22:52:11][V][ledc.output:110]: Setting duty: 0 on channel 2
[22:52:11][V][ledc.output:110]: Setting duty: 61563 on channel 4
[22:52:11][V][ledc.output:110]: Setting duty: 19067 on channel 3
[... 756 lines cut ...]
              call.set_rgb(1.0, 0.86, 0.0); // yellow
            } else if (state == 3) {
              call.set_rgb(0.75, 1.0, 0.0); // chartreuse
            } else if (state == 4) {
              call.set_rgb(0.0, 1.0, 0.0); // green
            } else if (state == 5) {
              call.set_rgb(0.0, 1.0, 0.5); // spring green
            } else if (state == 6) {
              call.set_rgb(0.0, 1.0, 1.0); // cyan
            } else if (state == 7) {
              call.set_rgb(0.0, 0.5, 1.0); // azure
            } else if (state == 8) {
              call.set_rgb(0.0, 0.0, 1.0); // blue
            } else if (state == 9) {
              call.set_rgb(0.5, 0.0, 1.0); // violet
            } else if (state == 10) {
              call.set_rgb(1.0, 0.0, 1.0); // magenta
            } else if (state == 11) {
              call.set_rgb(1.0, 0.0, 0.5); // rose
            }
            call.perform();
            state++;
            if (state == 12)
              state = 0;
```

ssieb, 2023-03-23:
Are these outputs directly driving the LEDs or is there some other chip in between?  Something to try is to put a template output between the light and the ledc output to log what the actual value is.  Also, depending on how the LEDs are actually driven, you might need to use `gamma_correct: 1` on the light.

mrmultifunk, 2023-03-25:
Sorry for the late reply. The channels are driven by constant current ic's. I tried to correct the gamma factor as you suggested, and guess what, it is now turning of in a following a nice linear curve! Thank you very much for your time 👌

kpelzel, 2023-03-25:
Can you post your final config?

mrmultifunk, 2023-03-25:
Sure, here you go:

```yaml
output:
  - platform: ledc
    pin: GPIO25
    id: red
    frequency: 1220Hz
    #min_power: 0.05
    zero_means_zero: True
  - platform: ledc
    pin: GPIO21
    id: green
    frequency: 1220Hz
    #min_power: 0.05
    zero_means_zero: True
  - platform: ledc
    pin: GPIO19
    frequency: 1220Hz
    id: blue
    #min_power: 0.05
    zero_means_zero: True
  - platform: ledc
    pin: GPIO32
    id: warm_white
    frequency: 1220Hz
    #min_power: 0.05
    max_power: 0.8
    zero_means_zero: True
  - platform: ledc
    pin: GPIO27
    id: cold_white
    frequency: 1220Hz
    #min_power: 0.05
    max_power: 0.7
    zero_means_zero: True

light:
  - platform: rgbww
    name: ${friendly_name}
    red: red
    green: green
    blue: blue
    cold_white: cold_white
    warm_white: warm_white
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    color_interlock: True
    id: thelight
    default_transition_length: 1000ms
    gamma_correct: 1.0 #0.6
    effects:
      - random:
      - random:
          name: 'Random Slow'
          update_interval: 30s
          transition_length: 7.5s
      - flicker:
      - lambda:
          name: Color Cycle
          update_interval: 11s
          lambda: |-
            static int state = 0;
            auto call = id(thelight).turn_on();
            call.set_transition_length(10000);
            if (state == 0) {
            call.set_rgb(1.0, 0.0, 0.0); // red
            } else if (state == 1) {
            call.set_rgb(1.0, 0.5, 0.0); // orange
            } else if (state == 2) {
            call.set_rgb(1.0, 0.86, 0.0); // yellow
            } else if (state == 3) {
            call.set_rgb(0.75, 1.0, 0.0); // chartreuse
            } else if (state == 4) {
            call.set_rgb(0.0, 1.0, 0.0); // green
            } else if (state == 5) {
            call.set_rgb(0.0, 1.0, 0.5); // spring green
            } else if (state == 6) {
            call.set_rgb(0.0, 1.0, 1.0); // cyan
            } else if (state == 7) {
            call.set_rgb(0.0, 0.5, 1.0); // azure
            } else if (state == 8) {
            call.set_rgb(0.0, 0.0, 1.0); // blue
            } else if (state == 9) {
            call.set_rgb(0.5, 0.0, 1.0); // violet
            } else if (state == 10) {
            call.set_rgb(1.0, 0.0, 1.0); // magenta
            } else if (state == 11) {
            call.set_rgb(1.0, 0.0, 0.5); // rose
            }
            call.perform();
            state++;
            if (state == 12)
            state = 0;
```