esphome/issues #3728: Teleinfo: old linky with USB port does not report any data

lchabert, 2022-10-25:
### The problem

I have an old and first one version of the linky power meter. I dont have any I1/I2/A ports, but only a USB port.

I have created a usb connector and linked to USB as described below:
* USB GND => ESP32 GND
* USB White data cable => ESP32 GPIO16

When i start ESPHome with teleinfo module, no data is reported into home-assistant.
I have tried to enable UART RX input into MQTT ... no message decoded by ESP32.

But when i connect my USB cable into an UART/USB convertor on my linux, linky data is displayed on minicom.

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.10.3

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

teleinfo

### Example YAML snippet

```yaml
substitutions:
  device_name: esp32-tic
  friendly_name: esp32-tic
  time_timezone: "Europe/Paris"

esphome:
  name: ${device_name}
  platform: ESP32
  board: esp32doit-devkit-v1
  on_boot:
    priority: 600
    then:
      - logger.log: "Hello World"

# Enable logging
debug:
  update_interval: 5s

mqtt:
  broker: 192.168.10.25
  username: home-assistant
  password: xxxxxxxxxx

logger:
  level: VERBOSE
  on_message:
    level: VERBOSE
    then:
      - mqtt.publish:
          topic: debug/topic
          payload: !lambda |-
            return "Triggered on_message with level " + to_string(level) + ", tag " + tag + " and message " + message;

# Enable Home Assistant API
#api:

ota:

wifi:
  ssid: "FBPRO-LCHA"
  password: "rxxxxxxr"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Linky Fallback Hotspot"
    password: "SlDqUsXARcjg"

web_server:
  port: 80

uart:
  id: uart_bus
  rx_pin: GPIO16
  baud_rate: 1200
  parity: EVEN
  data_bits: 7
  stop_bits: 1
  debug:
    direction: RX
    dummy_receiver: true
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

### Anything in the logs that might be useful for us?

```txt
[20:45:55][I][wifi:516]: WiFi Connected!
[20:45:55][C][wifi:360]:   Local MAC: 58:BF:25:8B:C4:DC
[20:45:55][C][wifi:361]:   SSID: 'FBPRO-LCHA'
[20:45:55][C][wifi:362]:   IP Address: 192.168.10.107
[20:45:55][C][wifi:364]:   BSSID: 70:FC:8F:F2:2D:50
[20:45:55][C][wifi:365]:   Hostname: 'esp32-tic'
[20:45:55][C][wifi:367]:   Signal strength: -48 dB ▂▄▆█
[20:45:55][V][wifi:369]:   Priority: 0.0
[20:45:55][C][wifi:371]:   Channel: 1
[20:45:55][C][wifi:372]:   Subnet: 255.255.255.0
[20:45:55][C][wifi:373]:   Gateway: 192.168.10.254
[20:45:55][C][wifi:374]:   DNS1: 192.168.10.25
[20:45:55][C][wifi:375]:   DNS2: 0.0.0.0
[20:45:55][D][wifi:525]: Disabling AP...
[20:45:55][C][web_server:092]: Setting up web server...
[20:45:55][C][ota:089]: Over-The-Air Updates:
[20:45:55][C][ota:090]:   Address: esp32-tic.local:3232
[20:45:55][W][ota:099]: Last Boot was an unhandled reset, will proceed to safe mode in 5 restarts
[20:45:55][C][mqtt:029]: Setting up MQTT...
[20:45:55][I][mqtt:176]: Connecting to MQTT...
[20:45:55][I][mqtt:216]: MQTT Connected!
[20:45:56][I][app:062]: setup() finished successfully!
[20:45:56][V][mqtt:415]: Publish(topic='esp32-tic/status' payload='online' retain=1)
[20:45:56][I][app:102]: ESPHome version 2022.9.4 compiled on Oct 25 2022, 20:45:20
[20:45:56][C][wifi:502]: WiFi:
[20:45:56][C][wifi:360]:   Local MAC: 58:BF:25:8B:C4:DC
[20:45:56][C][wifi:361]:   SSID: 'FBPRO-LCHA'
[20:45:56][C][wifi:362]:   IP Address: 192.168.10.107
[20:45:56][C][wifi:364]:   BSSID: 70:FC:8F:F2:2D:50
[20:45:56][C][wifi:365]:   Hostname: 'esp32-tic'
[20:45:56][C][wifi:367]:   Signal strength: -50 dB ▂▄▆█
[20:45:56][V][wifi:369]:   Priority: 0.0
[20:45:56][C][wifi:371]:   Channel: 1
[20:45:56][C][wifi:372]:   Subnet: 255.255.255.0
[20:45:56][C][wifi:373]:   Gateway: 192.168.10.254
[20:45:56][C][wifi:374]:   DNS1: 192.168.10.25
[20:45:56][C][wifi:375]:   DNS2: 0.0.0.0
[20:45:56][C][logger:275]: Logger:
[20:45:56][C][logger:276]:   Level: VERBOSE
[20:45:56][C][logger:277]:   Log Baud Rate: 115200
[20:45:56][C][logger:278]:   Hardware UART: UART0
[20:45:56][C][uart.arduino_esp32:107]: UART Bus:
[20:45:56][C][uart.arduino_esp32:109]:   RX Pin: GPIO16
[20:45:56][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[20:45:56][C][uart.arduino_esp32:113]:   Baud Rate: 1200 baud
[20:45:56][C][uart.arduino_esp32:114]:   Data Bits: 7
[20:45:56][C][uart.arduino_esp32:115]:   Parity: EVEN
[20:45:56][C][uart.arduino_esp32:116]:   Stop bits: 1
[20:45:56][C][web_server:125]: Web Server:
[20:45:56][C][web_server:126]:   Address: esp32-tic.local:80
[20:45:56][C][mdns:100]: mDNS:
[20:45:56][C][mdns:101]:   Hostname: esp32-tic
[20:45:56][V][mdns:102]:   Services:
[20:45:56][V][mdns:104]:   - _http, _tcp, 80
[20:45:56][C][ota:089]: Over-The-Air Updates:
[20:45:56][C][ota:090]:   Address: esp32-tic.local:3232
[20:45:56][W][ota:099]: Last Boot was an unhandled reset, will proceed to safe mode in 5 restarts
[20:45:56][C][mqtt:065]: MQTT:
[20:45:56][C][mqtt:067]:   Server Address: 192.168.10.25:1883 (192.168.10.25)
[20:45:56][C][mqtt:068]:   Username: 'home-assistant'
[20:45:56][C][mqtt:069]:   Client ID: 'esp32-tic-58bf258bc4dc'
[20:45:56][C][mqtt:071]:   Discovery prefix: 'homeassistant'
[20:45:56][C][mqtt:072]:   Discovery retain: YES
[20:45:56][C][mqtt:074]:   Topic Prefix: 'esp32-tic'
[20:45:56][C][mqtt:076]:   Log Topic: 'esp32-tic/debug'
[20:45:56][C][mqtt:079]:   Availability: 'esp32-tic/status'
[20:45:56][C][debug:049]: Debug component:
[20:45:56][D][debug:061]: ESPHome version 2022.9.4
[20:45:56][D][debug:065]: Free Heap Size: 243684 bytes
[20:45:56][D][debug:095]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[20:45:56][D][debug:132]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=1
[20:45:56][D][debug:140]: ESP-IDF Version: v3.3.5-1-g85c43024c
[20:45:56][D][debug:145]: EFuse MAC: 58:BF:25:8B:C4:DC
[20:45:56][D][debug:199]: Reset Reason: Power On Reset
[20:45:56][D][debug:244]: Wakeup Reason: Unknown
[20:50:52][I][ota:109]: Boot seems successful, resetting boot loop counter.
[20:50:52][D][esp32.preferences:113]: Saving 1 preferences to flash...
[20:50:52][V][esp32.preferences:125]: sync: key: 233825507, len: 4
[20:50:52][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

ssieb, 2022-10-26:
The GitHub issues are not for support.  Please ask questions like this in the esphome discord server or one of the forums.
However, the most likely cause is that GPIO16 does not have any interrupt support so the software serial won't work.

lchabert, 2022-10-27:
Ok thx for precision. I will take a look on discord.
So you recommand to use GPIO 3 ? But GPIO3 is used for debug and ESP32 programming no ?

ssieb, 2022-10-27:
oh, sorry!  I just realized it's an ESP32, not an ESP8266.  GPIO16 is fine, so it's something else.

closed 2023-03-12 by github-actions[bot]