esphome/issues #3729: Cannot establish ble connection in 2022.10.1

hcoohb, 2022-10-26:
### The problem

Since 2022.10.1, HA.esphome.bluetooth seems to always report ble_connections_free=0 and therefore no ble connection can be established.
I am using a custom component that needs an continuous connection to the ble device to receive notifications and send commands.

This issue did not happened in esphome 2022.10.0, so I assume this is related to https://github.com/esphome/esphome/pull/3935

### Which version of ESPHome has the issue?

2022.10.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

bluetooth_proxy

### Anything in the logs that might be useful for us?

```txt
HA log:

2022-10-26 10:34:14.226 DEBUG (MainThread) [custom_components.yeelight_bt.yeelightbt] BLE_device details: {'source': 'esp-bt', 'connector': HaBluetoothConnector(client=<class 'homeassistant.components.esphome.bluetooth.client.ESPHomeClient'>, source='esp-bt', can_connect=<function _async_can_connect_factory.<locals>._async_can_connect at 0x7f31bdc40b80>)}
2022-10-26 10:34:14.292 DEBUG (MainThread) [custom_components.yeelight_bt.yeelightbt] Connecting now to F8:24:41:E6:3E:39: XMCTD_:...
2022-10-26 10:34:14.292 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.328 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.329 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.407 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.414 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.432 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.438 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.443 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] Checking if esp-bt can connect, available=True, ble_connections_free=0
2022-10-26 10:34:14.444 ERROR (MainThread) [custom_components.yeelight_bt.yeelightbt] Connection: BleakError: F8:24:41:E6:3E:39 - F8:24:41:E6:3E:39: Failed to connect: No backend with an available connection slot that can reach address F8:24:41:E6:3E:39 was found
```

virtualm2000, 2022-10-26:
Going back to 2022.10.0 fixes this problem.

reniewk, 2022-10-27:
Not for me.  Rolled back to 2022.10.0, but still no slots available.

astrandb, 2022-10-28:
Same problem here running bluetooth proxy. Discovery works but I cannot control my device. Reverting to 20222.10.0 makes it work again.

virtualm2000, 2022-10-28:
try this patch in esphome/components/esp32_ble_client/ble_client_base.cpp

bool BLEClientBase::parse_device(const espbt::ESPBTDevice &device) {
  if (device.address_uint64() != this->address_)
    return false;
  if ((this->state_ != espbt::ClientState::IDLE)&&(this->state_ != espbt::ClientState::SEARCHING))
    return false;

  ESP_LOGD(TAG, "Found device at MAC address [%s]", device.address_str().c_str());
  this->set_state(espbt::ClientState::DISCOVERED);

  auto addr = device.address_uint64();
  this->remote_bda_[0] = (addr >> 40) & 0xFF;
  this->remote_bda_[1] = (addr >> 32) & 0xFF;
  this->remote_bda_[2] = (addr >> 24) & 0xFF;
  this->remote_bda_[3] = (addr >> 16) & 0xFF;
  this->remote_bda_[4] = (addr >> 8) & 0xFF;
  this->remote_bda_[5] = (addr >> 0) & 0xFF;
  this->remote_addr_type_ = device.get_address_type();
  return true;
}

astrandb, 2022-10-28:
Thanks @virtualm2000 that solves the problem for me.

virtualm2000, 2022-10-28:
> Thanks @virtualm2000 that solves the problem for me.

You're welcome! I think the connection management for bluetooth proxy is still a WIP, so this patch is not the proper solution.

Cameleon99, 2022-10-29:
Also having this issue, but I use Home Assistant to flash the boards, is there any way to downgrade to 2022.10.0 easily?

astrandb, 2022-10-29:
I am away from my dev system, but I know that a fix is waiting in upcoming release 2022.10.2

closed 2023-03-14 by github-actions[bot]