esphome/issues #3730: PULSE_METER broken after updating to 2022.10.0

telefonmann, 2022-10-26:
### The problem

I had a TCRT5000 running as PULSE_METER for a few months now with very! accurate readings of my Ferraris energy meter. But after updating to 2022.10.0 (coming from 2022.9.2) the PULSE_METER went crazy. I got extra-pulses and also the calculated power was nonsense, sometimes showing > 200k watts.

I now rolled back to pulse-meter version 2022.9.2 using "external_components" and the problem is gone - see screenshot.

[image]

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP8266

### Board

d1_mini_pro

### Component causing the issue

pulse_meter

rschueler, 2022-10-29:
I have the same issue. I used the PULSE_METER to measure gas and water consumption for month and it was very reliable before. It missed just a few pulses during this long time.
The gas and water meter have very slow and long pulses (less than 3 pulses per minute at max) and it could happen that the GPIO stays high for long time, if the meter just stops at the trigger digit. 

My original settings were:
- internal_filter_mode: PULSE
- internal_filter: 400ms
- timeout: 1min

Right after the upgrade only very few pulses were captured and then I tried to change the settings to make it work again, but without success. With the current settings I am getting 80-90% of the pulses for the water meter but the gas meter is counting pulses that are not there.

Is there a way for me to debug the problem and find the correct settings?

Current yaml:

```
esphome:
  name: esputility
  on_boot:
    priority: 500
    then:
    - pulse_meter.set_total_pulses:
        id: water_pulse_meter_id
        value: !lambda "return id(int_total_water);"
    - pulse_meter.set_total_pulses:
        id: gas_pulse_meter_id
        value: !lambda "return id(int_total_gas);"

preferences:
 flash_write_interval: 24h
  
globals:
  - id: int_total_water
    type: int
    restore_value: yes
    initial_value: '123456'
  - id: int_total_gas
    type: int
    restore_value: yes
    initial_value: '123456'
        
esp8266:
  board: nodemcuv2
  #restore_from_flash: True

dallas:
  - pin: GPIO2
  
sensor:
  - platform: pulse_meter
    id: water_pulse_meter_id
    pin: GPIO5
    internal_filter_mode: EDGE 
    internal_filter: 24s
    name: "Water Meter"
    icon: "mdi:faucet"
    unit_of_measurement: 'l/min'
    accuracy_decimals: 3
    filters:
      - multiply: 10
    total:
      name: "Water Total"
      id: water_total
      icon: "mdi:water"
      unit_of_measurement: "m³"
      state_class: "total_increasing" 
      accuracy_decimals: 2
      filters:
        - multiply: 0.01
      on_value:
        then:
          - lambda: "id(int_total_water) = id(water_total).state * 100;"

  - platform: pulse_meter
    id: gas_pulse_meter_id
    pin: GPIO4
    internal_filter_mode: EDGE 
    internal_filter: 24s
    name: "Gas Meter"
    icon: "mdi:fire"
    unit_of_measurement: 'm³/h'
    timeout: 1min
    accuracy_decimals: 3
    filters:
      - multiply: 0.6
    total:
      name: "Gas Total"
      id: gas_total
      icon: "mdi:fire"
      unit_of_measurement: "m³"
      device_class: 'gas'
      state_class: "total_increasing" 
      accuracy_decimals: 2
      filters:
        - multiply: 0.01
      on_value:
        then:
          - lambda: "id(int_total_gas) = id(gas_total).state * 100;"

````

jkorinth, 2022-11-01:
May be related to #3746 .

Claus-Schumacher, 2022-11-04:
I guess, I could have the same issue, too. I'm trying to read out a Ferraris electricity meter and do not manage to reliably count pulses.

With the bare TCRT5000 aligned in front of the electricity meter I get reliable detection of the passing by of the red segment on the wheel's circumference.

As soon as I hook up the TCRT5000 to the ESP, an 8266 in my case (ESP01S), it mainly randomly counts impulses, whereas the TCRT5000's red LED still blinks correctly. 

**ESPHOME version**
- 2022.10.2

**HOME ASSISTANT version**
- Home Assistant 2022.11.1
- Supervisor 2022.10.2
- Operating System 9.3
- Frontend 20221102.1 - latest

**BOARD**
- ESP01S

**Component causing the issue**
- pulse_meter

**CODE**
``` yaml
esphome:
  name: ${devicename}
  platform: esp8266
  board: esp01_1m
  on_boot:
    - text_sensor.template.publish:
        id: ${devicename}_version
        state: ${version_date} (V${version_number})
wifi:
  ssid: ${ssid}
  password: ${password}
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret fallback_ap_ssid
    password: !secret fallback_ap_password

# Enable Web-Server
web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

text_sensor:
  - platform: template
    name: "flashed"
    id: ${devicename}_version

sensor:
  - platform: pulse_meter
    pin: ${pin_data}
    name: "pulses"
    internal_filter: 550ms
    internal_filter_mode: PULSE
    total:
      name: "total pulses"
```

@telefonmann: Could you please post the code snipped on how to utilise "external_components" to replace the current version of a component with a former one. Thank You!

telefonmann, 2022-11-04:
@Claus-Schumacher sure! This is my currently working yaml:

```
esphome:
  name: d1-ferraris

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "booh!"

ota:
  password: "booh!"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  domain: .home.domain

text_sensor:
  - platform: template
    name: D1 Ferraris Uptime
    id: uptime_d1_ferraris
    icon: mdi:clock-start

sensor:
  - platform: pulse_meter
    pin: D5
    name: "Stromverbrauch Aktuell"
    device_class: power
    state_class: measurement
    unit_of_measurement: "W"
    internal_filter: 60ms
    internal_filter_mode: PULSE
    timeout: 20min
    filters:
      - multiply: 800
    total:
      name: "Stromverbrauch Total"
      unit_of_measurement: "kWh"
      accuracy_decimals: 3
      filters:
        - multiply: 0.0133333
  - platform: uptime
    name: Uptime Sensor D1 Ferraris
    id: uptime_sensor_d1_ferraris
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_d1_ferraris
            state: !lambda |-
              int seconds = round(id(uptime_sensor_d1_ferraris).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();

external_components:
  - source: github://esphome/esphome@2022.9.2
    components: [pulse_meter]

```

telefonmann, 2022-11-04:
@Claus-Schumacher it took quite some time to fiddle with the sensor position and the internal_filter time. For best results I ended up placing the sensor rotated like this:

[image]

github-k8n, 2022-11-05:
Same issue here, had correct readings for my heat meter (which outputs a clean 100ms pulse for each kWh) but since lately it seems to lose pulses (since around a month so very probably related to updating to 2022.10.x)
(And the water meter value also seems to be all over the place however I don't have a kind of reference for it as I just recently installed it, however the LED on the inductive proximity meter used does not show any random blinking or missing pulses)

ssieb, 2022-11-06:
Possibly caused by https://github.com/esphome/esphome/pull/3321?

janrg, 2022-11-07:
I also believe this to be caused by PR 3321. I started using the pulse meter a few days ago and couldn't get it to work properly. Whenever the gas meter I was reading stopped at the point where I got a signal, the pulse meter kept counting up and up and up. After finding this issue, I pulled in the version of pulse meter as of PR 3082 as external_component and since then it's worked flawlessly.

Claus-Schumacher, 2022-11-10:
First of all I'd like to thank @telefonmann for teaching me how to use `external_components`!

I went through all releases since 2022.9.2 and can confirm that my pulse meter setup

``` yaml
sensor:
  - platform: pulse_meter
    pin: ${pin_data}
    name: "pulses"
    internal_filter: 60ms
    internal_filter_mode: PULSE
    total:
      name: "total pulses"
```

runs on the following releases:

- [ ] 2022.11.0b2
- [ ] 2022.11.0b1
- [ ] 2022.10.2
- [ ] 2022.10.1
- [ ] 2022.10.0
- [ ] 2022.10.0b2
- [ ] 2022.10.0b1
- [x] 2022.9.4
- [x] 2022.9.3
- [x] 2022.9.2

I was surprised to learn that I could go back to releases before the major release I'm currently using, i.e., using a component of `2022.9.2` in `2022.10.2`. Therefore, I'm wondering if using `pulse_meter` of `2022.11.0bx` on `esphome 2022.10.2` is at all a valid test?

cstaahl, 2022-11-10:
I'd like to understand what is happening here. 

@telefonmann and @Claus-Schumacher
Could you add a binary_sensor on the same pin and share a snippet of the log where these issues occur? I have similar setup on my end with both an electricity meter and a gas meter that work very reliably.

Claus-Schumacher, 2022-11-11:
@cstaahl: As you suggested, I added a `binary_senor` on the same pin. 

The full `yaml` looks like this:

```yaml
esphome:
  name: ${devicename}
  platform: esp8266
  board: esp01_1m
  on_boot:
    - text_sensor.template.publish:
        id: ${devicename}_version
        state: ${version_date} (V${version_number})
wifi:
  ssid: ${ssid}
  password: ${password}
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret fallback_ap_ssid
    password: !secret fallback_ap_password

# Enable Web-Server
web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

text_sensor:
  - platform: template
    name: "flashed"
    id: ${devicename}_version

binary_sensor:
  - platform: gpio
    name: foo
    pin: ${pin_data}

sensor:
  - platform: pulse_meter
    pin: ${pin_data}
    name: "pulses"
    internal_filter: 60ms
    internal_filter_mode: PULSE
    total:
      name: "total pulses"
```

Here comes the log snippet:

```log
[17:13:41][I][app:102]: ESPHome version 2022.10.2 compiled on Nov 11 2022, 17:13:03
[17:13:41][C][wifi:502]: WiFi:
[17:13:41][C][wifi:360]:   Local MAC: [retracted]
[17:13:41][C][wifi:361]:   SSID: [redacted]
[17:13:41][C][wifi:362]:   IP Address: 192.168.1.192
[17:13:41][C][wifi:363]:   BSSID: [redacted]
[17:13:41][C][wifi:365]:   Hostname: 'tcrt5000_testing'
[17:13:41][C][wifi:367]:   Signal strength: -37 dB ▂▄▆█
[17:13:41][C][wifi:371]:   Channel: 6
[17:13:41][C][wifi:372]:   Subnet: 255.255.255.0
[17:13:41][C][wifi:373]:   Gateway: 192.168.1.254
[... 236 lines cut ...]
...
```
The `pulse_meter` exactly counts up my finger touches of the tcrt5000.

When touching the sensor once and leaving it alone from then on, the log looks like this:

```
[17:37:23][D][binary_sensor:036]: 'foo': Sending state OFF
[17:37:25][D][binary_sensor:036]: 'foo': Sending state ON
[17:37:25][D][sensor:126]: 'total pulses': Sending state 1.00000 pulses with 0 decimals of accuracy
[17:37:27][D][binary_sensor:036]: 'foo': Sending state OFF
[17:37:27][D][binary_sensor:036]: 'foo': Sending state ON
[17:37:33][D][binary_sensor:036]: 'foo': Sending state OFF
[17:37:33][D][binary_sensor:036]: 'foo': Sending state ON
[17:37:37][D][binary_sensor:036]: 'foo': Sending state OFF
[17:37:37][D][binary_sensor:036]: 'foo': Sending state ON
[17:37:41][D][binary_sensor:036]: 'foo': Sending state OFF
[17:37:41][D][binary_sensor:036]: 'foo': Sending state ON
[17:37:51][D][binary_sensor:036]: 'foo': Sending state OFF
[17:37:51][D][binary_sensor:036]: 'foo': Sending state ON
...
```
Apparently the `binary_sensor` still picks up state changes due to jitter or whatever but the pulse counter actually counts only my single touch. 

I hope that helps you to better understanding the issue. Pls let me know if I can do more!

Claus-Schumacher, 2022-11-12:
It makes a difference which TCRT5000 sensor device you use!

On the image @telefonmann posted on 4/Nov 23:47 there is a TCRT5000 setup with both an analogue and digital output (`A0`, `D0`), with the digital one connected.

Originally I used this sensor which has only on output:

[image]

When using the following sensor with pin `D0` connected I get the same weird behaviour:

[image]

When connecting pin `A0` instead, the "blue" sensor behaves normal, also when running it on `esphome@2022.10.2`.

@telefonmann, could you pls switch to `A0` to check if you can confirm my observations?

github-k8n, 2022-11-12:
Ok, after a few days I can confirm that using the external component trick to downgrade pulsemeter to 2022.9.2 works excellent. The values are spot on, no deviation at all from the real device (heat meter and water meter) at all, while with 2022.10.x it had really wrong values.
The heat meter has an output pulling to ground for 100ms for each pulse connected to an esp8266 nodemcuv2 on D2, the water meter uses a DC 5V NPN NO LJ18A3-8-Z/BX-5V proximity sensor connected to an esp8266 d1 mini on D2.
Both use "internal_filter_mode: PULSE" with 90ms for the heat meter and 100ms for the water meter.

telefonmann, 2022-11-13:
@cstaahl I am back on 2022.10.0 since yesterday evening with the change you asked for:

```
#external_components:
#  - source: github://esphome/esphome@2022.9.2
#    components: [pulse_meter]

binary_sensor:
  - platform: gpio
    pin: D5
    name: "pm_test"
```

And our "guest" is also back, partying all night:

[image]

You can find the log here: [https://drive.google.com/file/d/1zU8AW4gDTacKHOS2KGIG_gK5FJynByOg/view?usp=sharing](url).

Hope this helps, if you need anything else, please let me know...

telefonmann, 2022-11-13:
> It makes a difference which TCRT5000 sensor device you use!
> 
> On the image @telefonmann posted on 4/Nov 23:47 there is a TCRT5000 setup with both an analogue and digital output (`A0`, `D0`), with the digital one connected.
> 
> Originally I used this sensor which has only a single output:
> 
> [image]
> 
> When using the following sensor with pin `D0` connected I get the same weird behaviour:
> 
> [image]
> 
> When connecting pin `A0` instead, the "blue" sensor behaves normal, also when running it on `esphome@2022.10.2`.
> 
> @telefonmann, could you pls switch to `A0` to check if you can confirm my observations?

@Claus-Schumacher Can't confirm that. I just switched to A0 and I get no pulses at all...

Claus-Schumacher, 2022-11-13:
To complete my observations:

After switching to `A0` instead of `D0` on the TCRT5000 (ESP-01s on esphome@2022.10.2) I could reproducibly count pulses when triggering the TCRT5000 with my fingertip. However, with the TCRT5000 mounted in front of the Ferraris electricity meter it actually did see the red segment passing by, BUT the pin on the ESP didn't go low. The blue indicator LED dimmed down in brightness a bit, though.

I guess there's always enough reflected light coming from the glass pane in front of the electricity meter which prevents the `pulse_meter` component from detecting a proper pulse. 

Bottom line: I am at the exact same point where @telefonmann currently is: A working Ferraris digitiser with esphome@2022.9.4 . As soon as I use 2022.10.x behaviour is as posted above.

[image]

cstaahl, 2022-11-15:
Having looked at the logs provided by @telefonmann and @Claus-Schumacher (thank you for measuring and posting) and having tried to reproduce the issue, I am not convinced yet that the issue is caused by an error in the code. I suspect you both had dormant issues that were masked by the incorrect behaviour of the previous version of the pulse_meter.

@Claus-Schumacher: You seem to have substantial jitter when your sensor does not pull the signal low. From what I can tell (I don't know the value of your ${pin_data} parameter), you are not configuring the input pin to use a pullup resistor. Try adding `pullup: true` to the mode part of your pin configuration. Due to the jitter being from high to to low, the high pulses are always long enough so that the pulse_meter considers them (PULSE filtermode only looks at the length of the high pulses). Only the off states are very short in your case.

@telefonmann: You shouldn't connect the A0 pin but the D0 pin. I believe the A0 pin outputs the input the linear comparator that you can adjust with the screw on your PCB. 
Your issue seems to be a different one. I suspect it is related to "real output by the sensor", but caused by incorrect detection of the turning wheel of the ferrarris meter. With my TCRT5000 I have observed similar behaviour where the sensor blinks somewhere along the non colored part of the spinning wheel. Try observing it and see if you can notice any wrong detections as it spins around. These blinks (e.g. at 21:58:18, 21:58:23 and 22:49:36) are long enough so that they aren't filtered out. 
I would try increasing the value of your `internal_filter` parameter (I don't know how close you currently are to the maximum power you are expecting measure).

github-k8n, 2022-11-15:
@cstaahl that argument does not make any sense in regards to the issues I observed with the heat meter output.
That is not some LED shining on a spinning wheel but instead a clean 100ms output signal from the meter itself (using an internal  MOSFET i guess)
What kind of dormant issues should be maked by the previous "incorrect" behavior? Pullup is configured and the pulse is "inverted" meaning that the heat meter pulls it to ground for the 100ms pulse. (see my config above)
And similar to @Claus-Schumacher and @telefonmann  the issues go away once reverting back to  2022.9.2 for pulsemeter

cstaahl, 2022-11-15:
@github-k8n It is what I suspect from the log snippets provided by @telefonmann and @Claus-Schumacher. I am not ruling out errors in the code yet...

The previous version does not accurately detect certain type of pulses. In certain cases that leads to missed pulses, in other cases "dirt effects" are masked as they are not detected.
 
Unfortunately the output from these type of sensor is not always as clean as we would expect them to be. 
Check out what is described in https://github.com/esphome/feature-requests/issues/1354 for instance.

I saw your configuration, if you add binary sensor to your configuration and share log snippets where you observe the issues, I can better understand what might be causing the unwanted behaviour on your end.

github-k8n, 2022-11-15:
Not really eager to screw up my numbers again ... however I now connected an oscilloscope to the signal (very clean +3V line while no pulse is happening) waiting for the next impulse to happen. Once i got a nice picture of the signal I could maybe connect a second ESP in parallel (in that case an ESP32) using the new pulsemeter version.

telefonmann, 2022-11-16:
@cstaahl Did that! I changed the value to ```internal_filter: 180ms``` Now the good new is: Those massive bad readings from the nights before got better. Also those crazy high values are gone. But there is still some spike of 2.46kW at 02:19 which does not make sense. And even worse it now seems to drop pulses while we have high power consumption. The maximum is now ~9kW while it should be at at least 12kW. The total values on my Ferraris meter and HA are already out of sync after a few hours.

[image]

Now I feel stuck at this point. I think I should set the internal_filter even higher to get rid of the last spike(s) but also need to set it lower. Is there anything else I can do to make it work? The new log can be found [here](https://drive.google.com/file/d/1PxozACZa-4MlbMkygujBIYB8CpdBYwTI/view?usp=sharing). Thanks in advance!!!

cstaahl, 2022-11-16:
@telefonmann 
Have you tried improving the mounting of the sensor any further? I struggled **a lot** when mounting mine and I don't ever want to remove it again.

In your newest log the meter seems to miss every second pulse starting from around 09:21 up until 09:25. This would mean that your actual power consumption should be ~18-19 kW. Does that seem right?

The issue is that the pulses you are trying to detect are roughly of the same length as some of the disturbances you get. One thing that comes to my mind would be to handle the filtering differently for the high and low state. Similarly as with the [binary sensor](https://esphome.io/components/binary_sensor/index.html#delayed-on). The current implementation of the pulse_meter is equivalent of the `delayed_on_off` filter of the binary_sensor.

If you are willing to try it out, I can create a branch with that option. It would have two parameters for the internal_filter. One for the high pulse and a second for the low pulse.

telefonmann, 2022-11-16:
> @telefonmann Have you tried improving the mounting of the sensor any further? I struggled **a lot** when mounting mine and I don't ever want to remove it again.

@cstaahl  Same here. If I really had to move the sensor to make it work again, I think I would rather stay on 2022.9.x. 

I also wonder how this could improve things. If the main problem is the increased sensitivity, could it be a problem that the Ferraris wheel does have this black-mirror-black... pattern?

[image]

Then it would make sense to me that, if the wheel is slow, the black parts trigger a pulse. But if it is actually able to detect that pattern, to me that would seem like a very good placement of the sensor...?
 
> In your newest log the meter seems to miss every second pulse starting from around 09:21 up until 09:25. This would mean that your actual power consumption should be ~18-19 kW. Does that seem right?

Should have been around 12-14kW

> The issue is that the pulses you are trying to detect are roughly of the same length as some of the disturbances you get. One thing that comes to my mind would be to handle the filtering differently [...]

Sure, I can test it!

One question about the log: Most of the time it shows an on->off->pulse-info sequence like this 
with the pulse-info on the binary_sensor-OFF. The sensors seem to be in sync.
```
[21:19:28][D][binary_sensor:036]: 'pm_test': Sending state ON
[21:19:37][D][binary_sensor:036]: 'pm_test': Sending state OFF
[21:19:37][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 228.39415 W with 2 decimals of accuracy
[21:19:37][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.04000 kWh with 3 decimals of accuracy
```

But here things are mixed up and the pulse-info happens when the binary_sensor turns ON! Is it true that there 
must have been an ON-OFF cycle which only has been picked up by the pulse_meter and not by the binary_sensor?
If so, what magic did the binary_sensor use to filter it out?
```
[02:19:38][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 2464.31860 W with 2 decimals of accuracy
[02:19:38][D][sensor:126]: 'Stromverbrauch Total': Sending state 1.21333 kWh with 3 decimals of accuracy
[02:19:38][D][binary_sensor:036]: 'pm_test': Sending state ON
[02:19:57][D][binary_sensor:036]: 'pm_test': Sending state OFF
```

github-k8n, 2022-11-17:
Changed water meter to 2022.11.0 now including pulse meter and I see stuff not working correctly, in this case it seems to lose/not recognize the pulses. Combined with the observation from @telefonmann this again leads to the values being all over the place (too much, too litte, who knows...

```
[08:52:54][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[08:53:43][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:00][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[09:31:00][D][pulse_meter:034]: No pulse detected for 2286s, assuming 0 pulses/min
[09:31:03][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:07][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[09:31:10][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:10][D][sensor:126]: 'water_meter_counter': Sending state 183868.00000 L with 0 decimals of accuracy
[09:31:10][D][esp8266.preferences:236]: Saving preferences to flash...
[09:31:14][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[09:31:17][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:17][D][sensor:126]: 'water_meter': Sending state 5.54017 L/min with 2 decimals of accuracy
[09:31:17][D][sensor:126]: 'water_meter_counter': Sending state 183869.00000 L with 0 decimals of accuracy
[09:31:20][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[09:31:22][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:22][D][sensor:126]: 'water_meter': Sending state 10.21450 L/min with 2 decimals of accuracy
[09:31:22][D][sensor:126]: 'water_meter_counter': Sending state 183870.00000 L with 0 decimals of accuracy
[09:31:27][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[09:31:28][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:28][D][sensor:126]: 'water_meter': Sending state 10.44750 L/min with 2 decimals of accuracy
[09:31:28][D][sensor:126]: 'water_meter_counter': Sending state 183871.00000 L with 0 decimals of accuracy
[09:31:34][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[09:31:35][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:35][D][sensor:126]: 'water_meter': Sending state 8.83392 L/min with 2 decimals of accuracy
[09:31:35][D][sensor:126]: 'water_meter_counter': Sending state 183872.00000 L with 0 decimals of accuracy
[09:31:39][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
[09:31:42][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
[09:31:42][D][sensor:126]: 'water_meter': Sending state 8.86394 L/min with 2 decimals of accuracy
[09:31:42][D][sensor:126]: 'water_meter_counter': Sending state 183873.00000 L with 0 decimals of accuracy
[09:32:10][D][esp8266.preferences:236]: Saving preferences to flash...
```

cstaahl, 2022-11-17:
> But here things are mixed up and the pulse-info happens when the binary_sensor turns ON! Is it true that there must have been an ON-OFF cycle which only has been picked up by the pulse_meter and not [...]
@telefonmann 
Yes this is wrong. The interrupt is definitely triggered causing the state variable to get out of sync. I'll look into this and compare how this is handled in binary_sensor.

@github-k8n 

> ```
> [08:52:54][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
> [08:53:43][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
> [09:31:00][D][binary_sensor:036]: [...]
This is obviously wrong. Maybe due to the same root cause as in telefonmanns issues. Where there any pulses detected prior to this point in time?

> ```
> [09:31:20][D][binary_sensor:036]: 'water_meter_binsensor': Sending state ON
> [09:31:22][D][binary_sensor:036]: 'water_meter_binsensor': Sending state OFF
> [09:31:22][D][sensor:126]: 'water_meter': [...]
This part seems correct to me. Or are these values erroneous?

github-k8n, 2022-11-17:
> This part seems correct to me. Or are these values erroneous?

Nope, that part is good, I just let it run a little bit to show that it works (apart from the times when it doesn't :D )

cstaahl, 2022-11-18:
I created a branch that fixes an issue where the state tracker of the pulse_meter may get desynchronized.
I would appreciate if any one experiencing issues since the  2022.10.0 version could test it see if it resolves your issue. You can test it by modifying your configuration to contain the following:
```
external_components:
  - source: github://cstaahl/esphome@fix_pulse_meter_state_tracker
    components: [ pulse_meter ]
```

@telefonmann (or anyone else for that matter), if the above does not fix your issue, I've added a feature to setup filtering for the on and off states independently. It is in a separate branch (that contains the above fixes as well). If you're willing to test it, use the code below to pull the code and configure the filter. Let me know if it improves things on your end.
```
external_components:
  - source: github://cstaahl/esphome@pulse_meter_unsymmetric_pulse_mode
    components: [ pulse_meter ]
```

Example `yaml` configuration:

```
sensor:
  - platform: pulse_meter
    name: 'foo'
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
    internal_filter:
        on: 60ms
        off: 150ms
    internal_filter_mode: "PULSE"
    timeout: 600s
```

telefonmann, 2022-11-18:
@cstaahl I just deployed the branch ```fix_pulse_meter_state_tracker``` but now the pulses are gone. The log just shows the binary_sensor:

````[17:17:41][D][binary_sensor:036]: 'pm_test': Sending state ON
[17:17:49][D][binary_sensor:036]: 'pm_test': Sending state OFF
[17:20:21][D][binary_sensor:036]: 'pm_test': Sending state ON
[17:20:26][D][binary_sensor:036]: 'pm_test': Sending state OFF
[17:22:02][D][binary_sensor:036]: 'pm_test': Sending state ON
[17:22:03][D][binary_sensor:036]: 'pm_test': Sending state OFF
[17:23:53][D][binary_sensor:036]: 'pm_test': Sending state ON
[17:23:58][D][binary_sensor:036]: 'pm_test': Sending state OFF
[17:26:03][D][binary_sensor:036]: 'pm_test': Sending state ON
[17:26:08][D][binary_sensor:036]: 'pm_test': Sending state OFF
````

cstaahl, 2022-11-18:
@telefonmann 
Could you share the `yaml` you are using now, with all the modifications?

telefonmann, 2022-11-18:
> 

@cstaahl This is the current yaml...

`````
esphome:
  name: d1-ferraris

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "booh!"

ota:
  password: "booh!"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  domain: .home.domain

text_sensor:
  - platform: template
    name: D1 Ferraris Uptime
    id: uptime_d1_ferraris
    icon: mdi:clock-start

sensor:
  - platform: pulse_meter
    pin: D5
    name: "Stromverbrauch Aktuell"
    device_class: power
    state_class: measurement
    unit_of_measurement: "W"
    internal_filter: 60ms
    internal_filter_mode: PULSE
    timeout: 20min
    filters:
      - multiply: 800
    total:
      name: "Stromverbrauch Total"
      unit_of_measurement: "kWh"
      accuracy_decimals: 3
      filters:
        - multiply: 0.0133333
  - platform: uptime
    name: Uptime Sensor D1 Ferraris
    id: uptime_sensor_d1_ferraris
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_d1_ferraris
            state: !lambda |-
              int seconds = round(id(uptime_sensor_d1_ferraris).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();

external_components:
  - source: github://cstaahl/esphome@fix_pulse_meter_state_tracker
    components: [pulse_meter]

#external_components:
#  - source: github://esphome/esphome@2022.9.2
#    components: [pulse_meter]

binary_sensor:
  - platform: gpio
    pin: D5
    name: "pm_test"

cstaahl, 2022-11-18:
@telefonmann 
I am really puzzled... With a copy of your `yaml`, i get this output (I am using ESPHome 2022.10.0 btw) 
```
[19:58:52][D][binary_sensor:036]: 'pm_test': Sending state ON
[19:58:55][D][binary_sensor:036]: 'pm_test': Sending state OFF
[19:58:55][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 16800.83984 W with 2 decimals of accuracy
[19:58:55][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.53333 kWh with 3 decimals of accuracy
[19:58:56][D][binary_sensor:036]: 'pm_test': Sending state ON
[19:58:58][D][binary_sensor:036]: 'pm_test': Sending state OFF
[19:58:58][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 12868.63281 W with 2 decimals of accuracy
[19:58:58][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.54667 kWh with 3 decimals of accuracy
[19:58:58][D][binary_sensor:036]: 'pm_test': Sending state ON
[19:59:01][D][binary_sensor:036]: 'pm_test': Sending state OFF
[19:59:01][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 17731.80664 W with 2 decimals of accuracy
[19:59:01][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.56000 kWh with 3 decimals of accuracy
[19:59:02][D][binary_sensor:036]: 'pm_test': Sending state ON
[19:59:04][D][binary_sensor:036]: 'pm_test': Sending state OFF
[19:59:04][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 14379.86719 W with 2 decimals of accuracy
[19:59:04][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.57333 kWh with 3 decimals of accuracy
[19:59:05][D][bnary_sensor:036]: 'pm_test': Sending state ON
[19:59:07][D][binary_sensor:036]: 'pm_test': Sending state OFF
[19:59:07][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 16026.71094 W with 2 decimals of accuracy
[19:59:07][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.58667 kWh with 3 decimals of accuracy
[19:59:08][D][binary_sensor:036]: 'pm_test': Sending state ON
[19:59:10][D][binary_sensor:036]: 'pm_test': Sending state OFF
[19:59:10][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 13868.82520 W with 2 decimals of accuracy
[19:59:10][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.60000 kWh with 3 decimals of accuracy
```
I am grasping a straws here, maybe `esphome clean <CONFIG>` will change things?

funtax, 2022-11-18:
Hey there, I am completely new here and cannot guarantee for correctness.
It looks like i get the same issue like @telefonmann with an almost identical yaml and setup.

I am on ESPhome 2022.11.11.

Looking forward to figure this out, will check further tomorrow.

telefonmann, 2022-11-18:
@cstaahl 

This is so strange. I thought I didn't have any pulses but I actually did. It just took a while until the pulses started. Unfortunately I don't have any logs from that time. Only the HA graph shows the values between 18:30 (30 min after first deployment) and 20:30 (redeployment with clean before build):

[image]

After the second deployment the log showed one pulse in 30 minutes:

````
[20:49:02][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:49:03][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:50:42][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:50:48][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:51:17][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:51:18][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:51:36][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:51:37][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:51:55][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:51:56][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:52:14][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:52:15][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:52:33][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:52:34][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:52:52][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:52:53][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:53:11][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:53:12][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:53:30][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:53:31][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:53:50][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:53:50][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:54:09][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:54:09][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:54:28][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:54:28][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:54:47][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:54:48][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:55:06][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:55:07][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:55:25][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:55:26][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:55:44][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:55:45][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:56:03][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:56:04][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:56:04][D][sensor:126]: 'Stromverbrauch Total': Sending state 0.01333 kWh with 3 decimals of accuracy
[20:56:22][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:56:23][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:56:52][D][binary_sensor:036]: 'pm_test': Sending state ON
[20:56:59][D][binary_sensor:036]: 'pm_test': Sending state OFF
[20:59:55][D][binary_sensor:036]: 'pm_test': Sending state ON
[21:00:03][D][binary_sensor:036]: 'pm_test': Sending state OFF
[21:03:41][D][binary_sensor:036]: 'pm_test': Sending state ON
[21:03:53][D][binary_sensor:036]: 'pm_test': Sending state OFF
[21:08:10][D][binary_sensor:036]: 'pm_test': Sending state ON
[21:08:21][D][binary_sensor:036]: 'pm_test': Sending state OFF
[21:12:36][D][binary_sensor:036]: 'pm_test': Sending state ON
[21:12:48][D][binary_sensor:036]: 'pm_test': Sending state OFF
[21:17:03][D][binary_sensor:036]: 'pm_test': Sending state ON
[21:17:15][D][binary_sensor:036]: 'pm_test': Sending state OFF
[21:21:31][D][binary_sensor:036]: 'pm_test': Sending state ON
[21:21:42][D][binary_sensor:036]: 'pm_test': Sending state OFF

cstaahl, 2022-11-19:
@telefonmann 
I've been unable to reproduce anything similar to what you have. As a first try could you please change your configuration of the pin from
`pin: D5`
to 
```
pin:
  number: D5
  mode:
    input: true
    pullup: true
```
If that does not work, I've added some further log messages. You can activate them by changing the logger configuration to 
```
logger:
  level: VERY_VERBOSE
  logs:
    sensor.filter: DEBUG
    scheduler: DEBUG 
```
This will really clutter your output and certainly have an influence on performance. But it may help me understand what is happening with your sensor and ESP. I reduced the log level of some components. You may have to do that to further components if they also fill the log.

funtax, 2022-11-19:
Good morning.
Yes mentioned yesterday, it seems like I have the same issue like @telefonmann with ESPhome 2022.11.11, a NodeMCU v3 and the TCR5000 (4pins).
Right now I am testing with this yaml and it's still not working as expected:

```
sensor:
  - platform: pulse_meter
    pin: 
      number: D5
      mode:
        input: true
        pullup: true
    unit_of_measurement: W
    name: 'Energieverbrauch Alle (Aktuell)'
    accuracy_decimals: 0
    device_class: "power"
    state_class: measurement
    internal_filter: 60ms
    internal_filter_mode: PULSE
    timeout: 20min
    filters:
      - multiply: 666.6666666666667 # (60s/90 pulses per kWh * 1000)
      #- heartbeat: 10s
    icon: 'mdi:flash'
    id: energieverbrauch_komplett_w

  - platform: total_daily_energy
    name: "Energieverbrauch Alle (Tag)"
    filters:
      - multiply: 0.001
    unit_of_measurement: 'kWh'
    power_id: energieverbrauch_komplett_w
    
time:
  - platform: homeassistant
    id: homeassistant_time
    
external_components:
  #- source: github://esphome/esphome@2022.9.2
  - source: github://cstaahl/esphome@fix_pulse_meter_state_tracker
    components: [pulse_meter]

binary_sensor:
  - platform: gpio
    pin: D5
    name: "pm_test"
```

It produces this output:

> [09:44:44][D][binary_sensor:036]: 'pm_test': Sending state ON
> [09:44:47][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [09:46:07][D][binary_sensor:036]: 'pm_test': Sending state ON
> [09:46:10][D][binary_sensor:036]: 'pm_test': Sending state OFF

Reverting back to 2022.9.2 produces this log:

> [09:36:13][D][binary_sensor:036]: 'pm_test': Sending state ON
> [09:36:13][D][sensor:126]: 'Energieverbrauch Alle (Aktuell)': Sending state 467.15878 W with 0 decimals of accuracy
> [09:36:13][D][sensor:126]: 'Energieverbrauch Alle (Tag)': Sending state 0.69136 kWh with 2 decimals of accuracy
> [09:36:16][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [09:37:38][D][binary_sensor:036]: 'pm_test': Sending state ON
> [09:37:38][D][sensor:126]: 'Energieverbrauch Alle (Aktuell)': Sending state 466.98969 W with 0 decimals of accuracy
> [09:37:38][D][sensor:126]: 'Energieverbrauch Alle (Tag)': Sending state 0.70247 kWh with 2 decimals of accuracy
> [09:37:42][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [09:37:46][I][ota:113]: Boot seems successful, resetting boot loop counter.
> [09:39:03][D][binary_sensor:036]: 'pm_test': Sending state ON
> [09:39:03][D][sensor:126]: 'Energieverbrauch Alle (Aktuell)': Sending state 474.98605 W with 0 decimals of accuracy
> [09:39:03][D][sensor:126]: 'Energieverbrauch Alle (Tag)': Sending state 0.71358 kWh with 2 decimals of accuracy
> [09:39:06][D][binary_sensor:036]: 'pm_test': Sending state OFF
>

funtax, 2022-11-19:
Here are some extended logs by using 
```
logger:
  level: VERY_VERBOSE
  logs:
    sensor.filter: DEBUG
    scheduler: DEBUG 
```

> [10:14:56][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:14:56][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:00][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:00][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:01][D][binary_sensor:036]: 'pm_test': Sending state ON
> [10:15:01][VV][api.service:070]: send_binary_sensor_state_response: BinarySensorStateResponse {
>   key: 2768481743
>   state: YES
>   missing_state: NO
> }
> [10:15:01][VV][pulse_meter:035]: Correcting wrong sensor state
> [10:15:03][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:03][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:04][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [10:15:04][VV][api.service:070]: send_binary_sensor_state_response: BinarySensorStateResponse {
>   key: 2768481743
>   state: NO
>   missing_state: NO
> }
> [10:15:04][VV][pulse_meter:035]: Correcting wrong sensor state
> [10:15:11][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:11][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:15][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:15][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:18][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:18][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:26][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:26][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:27][VV][api.socket:198]: ESPHome Logs 2022.11.1 (192.168.1.96): Connection closed
> [10:15:27][W][api.connection:083]: ESPHome Logs 2022.11.1 (192.168.1.96): Connection closed
> [10:15:27][V][api:114]: Removing connection to ESPHome Logs 2022.11.1 (192.168.1.96)
> [10:15:30][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:30][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:33][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:33][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:45][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:45][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:15:48][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:15:48][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:00][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:00][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:03][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:03][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:15][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:15][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:18][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:18][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:30][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:30][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:33][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:33][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:33][D][binary_sensor:036]: 'pm_test': Sending state ON
> [10:16:33][VV][api.service:070]: send_binary_sensor_state_response: BinarySensorStateResponse {
>   key: 2768481743
>   state: YES
>   missing_state: NO
> }
> [10:16:33][VV][pulse_meter:035]: Correcting wrong sensor state
> [10:16:37][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [10:16:37][VV][api.service:070]: send_binary_sensor_state_response: BinarySensorStateResponse {
>   key: 2768481743
>   state: NO
>   missing_state: NO
> }
> [10:16:37][VV][pulse_meter:035]: Correcting wrong sensor state
> [10:16:45][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:45][VV][api.service:043]: send_ping_response: PingResponse {}
> [10:16:48][VV][api.service:470]: on_ping_request: PingRequest {}
> [10:16:48][VV][api.service:043]: send_ping_response: PingResponse {}

cstaahl, 2022-11-19:
@funtax 
Thank you for giving this a try as well. I noticed that latest version containing the more detailed log messages had not been pulled. Could you please change your external component configuration to 
```
external_components:
 #- source: github://esphome/esphome@2022.9.2 
 - source: github://cstaahl/esphome@fix_pulse_meter_state_tracker
    components: [pulse_meter]
    refresh: 0s
```
You can remove the `refresh: 0s` after you have run it once. Otherwise it will pull in the latest changes tomorrow as this value defaults to 1 day.

The `api.service` seems very active in the log. Could you also change the logger configuration to 
```
logger:
  level: VERY_VERBOSE
  logs:
    sensor.filter: DEBUG
    scheduler: DEBUG
    api.service: DEBUG
```
Could you then let the sensor cycle 10 or so on-off pulses and post the log?

funtax, 2022-11-19:
@cstaahl 
It looks like this new version solved the issue in general?
You can see that new values are recognized and they look "ok" as the oven us currently in usage.

> [12:33:56][V][api.connection:898]: Hello from client: 'Home Assistant 2022.11.3 (192.168.1.96)' | API Version 1.7
> [12:33:56][D][api.connection:917]: Home Assistant 2022.11.3 (192.168.1.96): Connected successfully
> [12:33:56][VV][time:027]: Got epoch 1668857636
> [12:33:56][D][time:044]: Synchronized time: 2022-11-19 12:33:56
> [12:34:04][VV][pulse_meter:097]: Interrupt registered
> [12:34:04][VV][pulse_meter:124]: Pin value is 1
> [12:34:04][VV][pulse_meter:135]: Signal has been stable long enough
> [12:34:04][VV][pulse_meter:154]: Exiting ISR
> [12:34:04][VV][pulse_meter:097]: Interrupt registered
> [12:34:04][VV][pulse_meter:124]: Pin value is 1
> [12:34:04][VV][pulse_meter:130]: Value equals state, exiting early
> [12:34:04][D][binary_sensor:036]: 'pm_test': Sending state ON
> [12:34:04][VV][pulse_meter:097]: Interrupt registered
> [12:34:04][VV][pulse_meter:124]: Pin value is 0
> [12:34:04][VV][pulse_meter:135]: Signal has been stable long enough
> [12:34:04][VV][pulse_meter:154]: Exiting ISR
> [12:34:04][VV][pulse_meter:097]: Interrupt registered
> [12:34:04][VV][pulse_meter:124]: Pin value is 0
> [12:34:04][VV][pulse_meter:130]: Value equals state, exiting early
> [12:34:04][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [12:34:04][V][sensor:076]: 'Energieverbrauch Alle (Aktuell)': Received new state 3.463803
> [12:34:04][D][sensor:126]: 'Energieverbrauch Alle (Aktuell)': Sending state 2309.20215 W with 0 decimals of accuracy
> [12:34:05][V][sensor:076]: 'Energieverbrauch Alle (Tag)': Received new state 3561.033936
> [12:34:05][D][sensor:126]: 'Energieverbrauch Alle (Tag)': Sending state 3.56103 kWh with 2 decimals of accuracy
> [12:34:23][VV][pulse_meter:097]: Interrupt registered
> [12:34:23][VV][pulse_meter:124]: Pin value is 1
> [12:34:23][VV][pulse_meter:135]: Signal has been stable long enough
> [12:34:23][VV][pulse_meter:154]: Exiting ISR
> [12:34:23][VV][pulse_meter:097]: Interrupt registered
> [12:34:23][VV][pulse_meter:124]: Pin value is 1
> [12:34:23][VV][pulse_meter:130]: Value equals state, exiting early
> [12:34:23][D][binary_sensor:036]: 'pm_test': Sending state ON
> [12:34:24][VV][pulse_meter:097]: Interrupt registered
> [12:34:24][VV][pulse_meter:124]: Pin value is 0
> [12:34:24][VV][pulse_meter:135]: Signal has been stable long enough
> [12:34:24][VV][pulse_meter:154]: Exiting ISR
> [12:34:24][VV][pulse_meter:097]: Interrupt registered
> [12:34:24][VV][pulse_meter:124]: Pin value is 0
> [12:34:24][VV][pulse_meter:130]: Value equals state, exiting early
> [12:34:24][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [12:34:24][V][sensor:076]: 'Energieverbrauch Alle (Aktuell)': Received new state 3.072354
> [12:34:24][D][sensor:126]: 'Energieverbrauch Alle (Aktuell)': Sending state 2048.23584 W with 0 decimals of accuracy
> [12:34:24][V][sensor:076]: 'Energieverbrauch Alle (Tag)': Received new state 3572.185547
> [12:34:24][D][sensor:126]: 'Energieverbrauch Alle (Tag)': Sending state 3.57219 kWh with 2 decimals of accuracy
> [12:34:43][VV][pulse_meter:097]: Interrupt registered
> [12:34:43][VV][pulse_meter:124]: Pin value is 1
> [12:34:43][VV][pulse_meter:135]: Signal has been stable long enough
> [12:34:43][VV][pulse_meter:154]: Exiting ISR
> [12:34:43][VV][pulse_meter:097]: Interrupt registered
> [12:34:43][VV][pulse_meter:124]: Pin value is 1
> [12:34:43][VV][pulse_meter:130]: Value equals state, exiting early
> [12:34:43][D][binary_sensor:036]: 'pm_test': Sending state ON
> [12:34:44][VV][pulse_meter:097]: Interrupt registered
> [12:34:44][VV][pulse_meter:124]: Pin value is 0
> [12:34:44][VV][pulse_meter:135]: Signal has been stable long enough
> [12:34:44][VV][pulse_meter:154]: Exiting ISR
> [12:34:44][VV][pulse_meter:097]: Interrupt registered
> [12:34:44][VV][pulse_meter:124]: Pin value is 0
> [12:34:44][VV][pulse_meter:130]: Value equals state, exiting early
> [12:34:44][D][binary_sensor:036]: 'pm_test': Sending state OFF
> [12:34:44][V][sensor:076]: 'Energieverbrauch Alle (Aktuell)': Received new state 2.983294
> [12:34:44][D][sensor:126]: 'Energieverbrauch Alle (Aktuell)': Sending state 1988.86243 W with 0 decimals of accuracy
> [12:34:44][V][sensor:076]: 'Energieverbrauch Alle (Tag)': Received new state 3583.293457
> [12:34:44][D][sensor:126]: 'Energieverbrauch Alle (Tag)': Sending state 3.58329 kWh with 2 decimals of accuracy
> [12:35:03][VV][pulse_meter:097]: Interrupt registered
> [12:35:03][VV][pulse_meter:124]: Pin value is 1
[... 239 lines cut ...]
    icon: 'mdi:flash'
    id: energieverbrauch_komplett_w

  - platform: total_daily_energy
    name: "Energieverbrauch Alle (Tag)"
    filters:
      - multiply: 0.001
    unit_of_measurement: 'kWh'
    power_id: energieverbrauch_komplett_w
    
time:
  - platform: homeassistant
    id: homeassistant_time
    
external_components:
  #- source: github://esphome/esphome@2022.9.2
  - source: github://cstaahl/esphome@fix_pulse_meter_state_tracker
    components: [pulse_meter]
    refresh: 0s

binary_sensor:
  - platform: gpio
    pin: D5
    name: "pm_test"
```

cstaahl, 2022-11-19:
@funtax That's great news! I hope that the addition of `refresh: 0s` solves it for @telefonmann as well. Anyone else experiencing issues since 2022.10.0 and willing to test this branch, please let me know if it solves the issues you've been encountering.

```
external_components:
  - source: github://cstaahl/esphome@fix_pulse_meter_state_tracker
    components: [pulse_meter]
    refresh: 0s
```

telefonmann, 2022-11-19:
@cstaahl I also deployed the latest version of fix_pulse_meter_state_tracker today. And though I did not add the input/pullup config (yet) I can confirm it works a lot better than before. binary_sensor and pulse_meter are almost completely in sync. 

Most of the time the pattern in the log looks like this:

````
[17:52:25][VV][pulse_meter:097]: Interrupt registered
[17:52:25][VV][pulse_meter:124]: Pin value is 1
[17:52:25][VV][pulse_meter:135]: Signal has been stable long enough
[17:52:25][VV][pulse_meter:154]: Exiting ISR
[17:52:25][VV][pulse_meter:097]: Interrupt registered
[17:52:25][VV][pulse_meter:124]: Pin value is 1
[17:52:25][VV][pulse_meter:130]: Value equals state, exiting early
[17:52:25][D][binary_sensor:036]: 'pm_test': Sending state ON
[17:52:26][VV][pulse_meter:097]: Interrupt registered
[17:52:26][VV][pulse_meter:124]: Pin value is 0
[17:52:26][VV][pulse_meter:135]: Signal has been stable long enough
[17:52:26][VV][pulse_meter:154]: Exiting ISR
[17:52:26][VV][pulse_meter:097]: Interrupt registered
[17:52:26][VV][pulse_meter:124]: Pin value is 0
[17:52:26][VV][pulse_meter:130]: Value equals state, exiting early
[17:52:26][D][binary_sensor:036]: 'pm_test': Sending state OFF
[17:52:26][V][sensor:076]: 'Stromverbrauch Aktuell': Received new state 3.293085
[17:52:26][D][sensor:126]: 'Stromverbrauch Aktuell': Sending state 2634.46777 W with 2 decimals of accuracy
[17:52:26][V][sensor:076]: 'Stromverbrauch Total': Received new state 267.000000
[17:52:26][D][sensor:126]: 'Stromverbrauch Total': Sending state 3.55999 kWh with 3 decimals of accuracy

````

But there are some cases (~50 in 5 hours) where the pulse_meter seems to be in the wrong state:

````
[17:53:17][VV][pulse_meter:097]: Interrupt registered
[17:53:17][VV][pulse_meter:124]: Pin value is 1
[17:53:17][VV][pulse_meter:135]: Signal has been stable long enough
[17:53:17][VV][pulse_meter:154]: Exiting ISR
[17:53:17][VV][pulse_meter:097]: Interrupt registered
[17:53:17][VV][pulse_meter:124]: Pin value is 1
[17:53:17][VV][pulse_meter:130]: Value equals state, exiting early
[17:53:17][D][binary_sensor:036]: 'pm_test': Sending state ON
[17:53:22][VV][pulse_meter:097]: Interrupt registered
[17:53:22][VV][pulse_meter:124]: Pin value is 1
[17:53:22][VV][pulse_meter:130]: Value equals state, exiting early
[17:53:22][VV][pulse_meter:097]: Interrupt registered
[17:53:22][VV][pulse_meter:124]: Pin value is 0
[17:53:22][VV][pulse_meter:150]: Signal has NOT been stable long enough
[17:53:22][VV][pulse_meter:154]: Exiting ISR
[17:53:22][VV][pulse_meter:097]: Interrupt registered
[17:53:22][VV][pulse_meter:124]: Pin value is 0
[17:53:22][VV][pulse_meter:150]: Signal has NOT been stable long enough
[17:53:22][VV][pulse_meter:154]: Exiting ISR
[17:53:22][VV][pulse_meter:097]: Interrupt registered
[17:53:22][VV][pulse_meter:124]: Pin value is 0
[17:53:22][VV][pulse_meter:150]: Signal has NOT been stable long enough
[17:53:22][VV][pulse_meter:154]: Exiting ISR
[17:53:22][VV][pulse_meter:097]: Interrupt registered
[17:53:22][VV][pulse_meter:124]: Pin value is 0
[17:53:22][VV][pulse_meter:150]: Signal has NOT been stable long enough
[17:53:22][VV][pulse_meter:154]: Exiting ISR
[17:53:22][D][binary_sensor:036]: 'pm_test': Sending state OFF
[17:53:22][VV][pulse_meter:035]: Correcting wrong sensor state to low

````

Will now deploy the version with added input/pullup config to see if that makes a difference.

cstaahl, 2022-11-20:
@telefonmann 
I've been able reproduce what you posted as well. I am working on an update that involves a larger change and will then test it for a couple of days before I push another update to the branch.

cstaahl, 2022-11-30:
Hi all

I have now implemented a larger change to the way the pulses are registered and filtered. This **should** produce a more stable behavior. 
I would really appreciate, if you could this change by applying the snippet below in your config.

```
 - source: github://cstaahl/esphome@fix_pulse_meter_state_tracker
    components: [pulse_meter]
    refresh: 0s
```

It fixes the issue that I reproduced from @telefonmann's log and I hope it solves the more general root cause. I have had it running for several days without missing any pulses, however my sensor worked before this fix as well.... 

Let me know what the results are on your end.

funtax, 2022-12-02:
Hi @cstaahl,
thanks for your effort. I will update my configuration and check the results over the next days.

telefonmann, 2022-12-02:
@cstaahl Many thanks from me too! I deployed it yesterday and it looks very promising! Will leave it running for a few days and see if Ferraris and pulse_meter stay in sync...

rschueler, 2022-12-07:
Hi @cstaahl, I updated to your PR last weekend and both meters (gas and water) are still perfect in sync. Great job! Thanks.

telefonmann, 2022-12-08:
Same here. Ferraris and pulse_meter are perfectly in sync for 6 days now with the initial configuration of ``internal_filter: 60ms`` which I used with 2022.9.2. Many thanks @cstaahl

github-k8n, 2022-12-13:
Can also confirm both meters are now still spot on after a few days, thanks @cstaahl ! Hope it's going to be merged soon.

cstaahl, 2022-12-16:
I've created a PR for this fix now: https://github.com/esphome/esphome/pull/4199

The PR includes some  minor changes in addition to the branch previously posted here. I have been testing them extensively on my end for the last days. 

Please test the PR as well and comment on the results directly in the PR to facilitate a swift merge of this fix.

```
external_components:
  - source: github://pr#4199
    components: [ pulse_meter ]
```

closed 2022-12-22 by jesserockz