esphome/issues #3732: script.wait action cannot be used in lambda function

LucasCZE, 2022-10-26:
### The problem

According to [the documentation](https://esphome.io/guides/automations.html#script-wait-action) the `script.wait` action can be used also in a lambda function. But in reality such code cannot be compiled.

I don't think it is specific to the ESPHome version I'm currently using or platform/board specific. There is no wait() declaration in the script.h file.

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Shelly 2.5

### Component causing the issue

Script

### Example YAML snippet

```yaml
- lambda: |-
    ...
    id(malfunction_or_obstacle_detected_script).execute();
    id(malfunction_or_obstacle_detected_script).wait();
    ...
```

### Anything in the logs that might be useful for us?

```txt
/config/blind-bedroom.yaml: In lambda function:
/config/blind-bedroom.yaml:1137:54: error: 'class esphome::script::SingleScript' has no member named 'wait'
 1137 |                 id(malfunction_or_obstacle_detected_script).wait();
      |                                                      ^~~~
*** [.pioenvs/blind-bedroom/src/main.cpp.o] Error 1
```

ssieb, 2022-10-26:
That's right.  That method doesn't exist and *can't* exist.  It would crash the device.

closed 2022-10-26 by ssieb

LucasCZE, 2022-10-26:
I see, that makes sense. I didn't think about it that way.

But the documentation is still wrong. You have only removed `id(my_script).wait();` from the snippet. But what about this information?

[image]

This doesn't make any sense in the section for the script.wait action. This should definitely be removed also.

ssieb, 2022-10-26:
Good point.  There's a specific section for execute (which also had an error in the lambda example).  I fixed both.