esphome/issues #3733: EZO component for EC sensor returns only first value.

orrious, 2022-10-26:
### The problem

The EZO component does not return all values read for EC.

By default the EC sensor returns four comma separated values: EC, TDS, Salinity, and Specific Gravity, however only the EC value is returned.

4 separate sensors should be created (EC, TDS, Salinity, Specific Gravity) each providing their respective values.
 

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Home Assistant 2022.7.7

### What platform are you using?

ESP8266

### Board

Wemos D1

### Component causing the issue

ezo

### Example YAML snippet

```yaml
dallas:
  - pin: D7

i2c:
  - id: bus_a
    sda: D2
    scl: D1
    scan: True

sensor:
  - platform: dallas
    address: 0x7a3c01b556acf728
    id: dht
    name: "DH Temperature"
    on_value:
      - lambda: id(ec_ezo).set_tempcomp_value(id(dht).state);

  - platform: ezo
    id: ec_ezo
    address: 100
    accuracy_decimals: 0
    unit_of_measurement: "cm"
    update_interval: 10s
    name: "EC"
```

inspectorgrow, 2023-02-28:
I am interested in the other readings.

contactpjb, 2023-05-21:
I am also interested in multiple readings, also using the EC sensor.

cpetzel, 2023-06-08:
I too am interested in the other values

TheRealFalseReality, 2023-08-22:
Would love to see more readings, especially since the HUM sensors has more than one secondary sensors, temp. and dew.