esphome/issues #3736: Total daily energy sensor resetting at wrong time despite correct time zone

frazzyfin, 2022-10-27:
### The problem

The total daily energy sensor in home assistant for my energy monitor running ESPHome is resetting at 1am instead of midnight. My timezone is currently BST (UTC+1).

I have tried using the sntp time component and the homeassistant time component. I think the correct time zone is selected because when I look at the logs in ESPHome each log line shows the real current time.

ESPHome is running in a docker container and I have the timezone passed into the container (- /etc/localtime:/etc/localtime:ro).

### Which version of ESPHome has the issue?

v2022.10.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.10.4

### What platform are you using?

ESP32

### Board

Emporia Vue 2

### Component causing the issue

emporiavue2

### Additional information

See log time and windows clock are the same:

[image]

Home assistant sensor resetting at 1am: 

[image]

ssieb, 2022-10-27:
The log timestamps are from the receiving side.  They don't show what time the ESP has.  Add the timezone to the ESP config.

frazzyfin, 2022-10-27:
Thanks that worked. Though I wonder why the timezone was being picked up correctly through the container or homeassistant.

closed 2022-10-27 by frazzyfin