esphome/issues #3740: Sprinkler state not stopped after full cycle is completed, running disabled valve

quhar, 2022-10-29:
### The problem

Let me start by thanking you for amazing work that is sprinkler controller. 

For last few weeks I have 2 issues with one of the controllers. (I don't remember when it exactly started). 

Both of them happen occasionally, but both of them happened last night. 

1. I disabled the 'Front Lawn' valve in the HA UI weeks ago, however occasionally this valve still runs when I ran full cycle. This happened at first run that night at 01:00. In the log messages below I did stop it when I realized it was running. First screenshot below shows that the valve was disabled when it this valve started.

2. Sometimes, after the full cycle completes the switch for this controller is till `on`, all the valves are off, including the pump, but controllers main switch stays on. The next time schedule wants to run full cycle it doesn't do anything because controller is still active. The second screenshot below shows that this happened more than once in the history I have in my HA. This happened during the ran at 04:00.

Thanks for the help!

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

sprinkler

### Example YAML snippet

```yaml
sprinkler:
  - id: lawn_sprinkler_ctrlr
    main_switch: "Lawn Sprinklers"
    auto_advance_switch: "Lawn Sprinklers Auto Advance"
    valve_overlap: 5s
    valves:
      - valve_switch: "Front Lawn"
        enable_switch: "Enable Front Lawn"
        pump_switch_id: mcp_sw_0
        run_duration: 300s
        valve_switch_id: mcp_sw_1
      - valve_switch: "Back Lawn"
        enable_switch: "Enable Back Lawn"
        pump_switch_id: mcp_sw_0
        run_duration: 300s
        valve_switch_id: mcp_sw_2

  - id: drip_sprinkler_ctrlr
    main_switch: "Drip Irrigation"
    auto_advance_switch: "Drip Irrigation Auto Advance"
    valve_open_delay: 5s
    valves:
      - valve_switch: "Front Drip"
        enable_switch: "Enable Front Drip"
        pump_switch_id: mcp_sw_0
        run_duration: 900s
        valve_switch_id: mcp_sw_3
      - valve_switch: "Side Drip"
        enable_switch: "Enable Side Drip"
        pump_switch_id: mcp_sw_0
        run_duration: 900s
        valve_switch_id: mcp_sw_4
      - valve_switch: "Back Drip"
[... 392 lines cut ...]
2022-10-28T04:05:41-0700 : garage/debug : [VV][scheduler:196]: Running interval 'update' with interval=60000 last_execution=149870006 (now=149930008)
2022-10-28T04:05:41-0700 : garage/debug : [D][sensor:127]: 'Sprinklers Duration': Sending state 4.50000  with 1 decimals of accuracy
2022-10-28T04:05:41-0700 : garage/debug : [VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
2022-10-28T04:05:41-0700 : garage/sensor/sprinklers_duration/state : 4.5
2022-10-28T04:05:46-0700 : garage/debug : [VV][scheduler:196]: Running interval 'update' with interval=60000 last_execution=149875434 (now=149935434)
2022-10-28T04:05:46-0700 : garage/debug : [D][sensor:127]: 'Garage ESP32 Uptime': Sending state 149935.43750 s with 0 decimals of accuracy
2022-10-28T04:05:46-0700 : garage/debug : [VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
2022-10-28T04:05:46-0700 : garage/sensor/garage_esp32_uptime/state : 149935
2022-10-28T04:05:49-0700 : garage/debug : [VV][scheduler:196]: Running interval 'update' with interval=900000 last_execution=149038578 (now=149938578)
2022-10-28T04:05:53-0700 : garage/debug : [VV][api.service:470]: on_ping_request: PingRequest {}
2022-10-28T04:05:53-0700 : garage/debug : [VV][api.service:043]: send_ping_response: PingResponse {}
```
```

### Additional information

'Front Lawn' was disabled, yet it run when full cycle was started.

![image](https://user-images.githubusercontent.com/4906585/198821867-ad28ca0e-9b51-41ed-a606-6dc1c94c6dc6.png)

'Lawn sprinklers' should run for 4.5 minutes each valve, but here it stayed  `on` until manually disabled. 

![image](https://user-images.githubusercontent.com/4906585/198821128-3a43fda2-a587-4ff6-8543-10586cf51332.png)

![image](https://user-images.githubusercontent.com/4906585/198821918-19621a11-e37a-42cb-aeba-40aa5d989333.png)

kbx81, 2022-11-24:
I think I know what's going on. I'm working on some updates to the component so I'll roll the fix into the PR. 😄

nununo, 2022-11-24:
The same is happening to me. Thank you @kbx81!

kbx81, 2022-11-24:
@nununo could you also post your `sprinkler` config, if you don't mind?

nununo, 2022-11-24:
Sure!

Here it is:

```yaml
#This is the ESPHome script, it goes to the esphome folder
# Adapted from here: https://github.com/bruxy70/Irrigation-with-display
substitutions:
  rel1_gpio: GPIO12
  rel2_gpio: GPIO5
  rel3_gpio: GPIO4
  rel4_gpio: GPIO15
  key1_gpio: GPIO0
  key2_gpio: GPIO9
  key3_gpio: GPIO10
  key4_gpio: GPIO14
  rx_gpio: GPIO3
  tx_gpio: GPIO1

esphome:
  name: regador
  comment: Sonoff 4ch PRO controlling irrigation solenoid valves
  includes: regador.h
  on_boot:
    priority: 600
    then:
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on: blue_led_light

esp8266:
  board: esp8285

uart:
  rx_pin: $rx_gpio
  tx_pin: $tx_gpio
  baud_rate: 9600

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

captive_portal:

api:
  password: !secret api_password
  reboot_timeout: 0s
  services:
    - service: next_valve
      then:
        - sprinkler.next_valve: controller

ota:
  password: !secret ota_password

web_server:
  port: 80

logger:

sprinkler:
  - id: controller
    main_switch: regador
    auto_advance_switch: regador_auto_advance
    valves:
      - valve_switch: regador_rega_pomar
        valve_switch_id: regador_valve_pomar
        enable_switch: regador_active_pomar
[... 173 lines cut ...]
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();

text_sensor:
# Esphome installed version
  - platform: version
    name: regador_esphome_version

# Uptime in human-readable format
  - platform: template
    name: regador_uptime_human
    id: regador_uptime_human
    icon: mdi:clock-start
    entity_category: diagnostic
```

Good luck!

kbx81, 2022-11-28:
I believe I've identified the issue(s) causing this behavior -- I have [a branch](https://github.com/kbx81/esphome/tree/sprinkler-v2) in which I'm working on this as well as a couple of additional features. I'll open a PR probably within the next week or two, but you are welcome to test the branch (if you feel up to it...or wait for the PR) and confirm if it is fixed for you. 😄

nununo, 2022-12-22:
Hello @kbx81,

I finally got some time to test this but I don't know how to install and use this particular branch. I have macOS and installed ESPHome with `brew install esphome`. Can you point me to documentation on how to run esphome from your branch?

Thanks and sorry for this.

Regards.

randybb, 2022-12-22:
@nununo: add this to your yaml and it will pull this PR
```yaml
external_components:
  - source: github://pr#4159
    components: sprinkler 
```

kbx81, 2022-12-24:
Hopefully what @randybb mentioned helps you @nununo -- let me know how it goes and feel free to ping me if you have any questions. Also, the updated docs preview is available [here](https://deploy-preview-2510--esphome.netlify.app/components/sprinkler.html).

nununo, 2022-12-24:
Hello @kbx81,

I'm happy to inform you that, after replicating the error again just to be sure that both problems were still happening, I recompiled it again with this PR and both problems seem to be gone. Everything is now working as expected.

Thank you!
And thank you also to @randybb for teaching me how to do it.

kbx81, 2022-12-24:
Awesome! Thanks for testing & confirming. Once we merge this PR, you'll be able to remove that bit of config from your device config. 😄

closed 2023-02-22 by kbx81