esphome/issues #3744: ESP32-C3 in DIO mode should allow use of GPIO12 and GPIO13

davidmonro, 2022-11-01:
### The problem

I'm using a "luatos.com/t/esp32c3" board (airm2m_core_esp32c3 is the board type).
Attempting to use GPIO12 or GPIO13 in the config produces the following error:

This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIHD).

However, this board uses DIO flash mode, so  GPIO12 and GPIO13 are not actually used for the flash interface, and should be available as general purpose IO pins.

(Google translated from the maker of the board at https://wiki.luatos.com/chips/esp32c3/board.html:
IO12 (GPIO12) and IO13 (GPIO13) are multiplexed for SPI signals SPIHD and SPIWP in QIO mode. In order to increase the number of available GPIOs, the development board chooses the DIO mode of 2-wire SPI. IO12 and IO13 are not connected to flash, and use their own compilation The software needs to pay attention to configure the flash as DIO mode.)

Currently, gpio_esp32_c3.py  just defines pins 12-17 as the PSRAM pins and doesn't pay attention the flash mode used for the board.

I don't know if it is possible to access the board config from within this code?

If not, perhaps it could be handled like the S3R8 and S3R8V chips where it throws a warning rather than an error for those pins?

### Which version of ESPHome has the issue?

2022.9.4 (but git dev branch is the same)

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

airm2m_core_esp32c3

davidmonro, 2023-03-01:
This is still an issue.

roommaster, 2023-06-23:
I‘m using the same board and want it be solved.

fluppie, 2023-09-22:
Yes, using the Arduino IDE I can use the onboard leds of this board without any problem.
```
#define ledPin                 13  // D5
#define ledPinDataRequest      12  // D4
```

Doing this in ESPHome gives the famous error.
```
  This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIWP).
  This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIHD).
```
```
esphome:
  name: ${device_name}
  friendly_name: ${device_name}
  comment: $device_comment
  name_add_mac_suffix: false
  project:
    name: "Generator"
    version: "1"
  platformio_options:
    board_build.flash_mode: dio

esp32:
#  board: airm2m_core_esp32c3
  board: esp32-c3-devkitm-1
  framework:
#    type: arduino
    type: esp-idf

output:
  - platform: gpio
    pin: 13  # D5
    id: LedPin

  - platform: gpio
    pin: 12  # D4
    id: LedPinDataRequest
```

Soviet-Steve, 2023-12-07:
yep, not stale

closed 2023-12-21 by jesserockz

hldh214, 2024-06-21:
FYI, since [2024.6.0](https://github.com/esphome/esphome/releases/tag/2024.6.0), the GPIO validation behavior has changed. Previously(https://github.com/esphome/esphome/pull/5982), setting the board to "airm2m_core_esp32c3" could avoid this issue. Now, you must also set `ignore_pin_validation_error` for each pin. for example:

```yaml
output:
  - platform: ledc
    pin: 
      number: LED1_BUILTIN
      ignore_pin_validation_error: true
```

Refs:
https://github.com/esphome/esphome/commit/f0ec900e4841a32023491f8482af3c4a07432adb

ZDZX-T, 2024-06-26:
> FYI, since [2024.6.0](https://github.com/esphome/esphome/releases/tag/2024.6.0), the GPIO validation behavior has changed. Previously([esphome/esphome#5982](https://github.com/esphome/esphome/pull/5982)), setting the board to "airm2m_core_esp32c3" could avoid this issue. Now([esphome/esphome#5615](https://github.com/esphome/esphome/pull/5615)), you must also set `ignore_pin_validation_error` for each pin. for example:
> 
> ```yaml
> output:
>   - platform: ledc
>     pin: 
>       number: LED1_BUILTIN
>       ignore_pin_validation_error: true
> ```
> 
> Refs: [esphome/esphome@f0ec900](https://github.com/esphome/esphome/commit/f0ec900e4841a32023491f8482af3c4a07432adb)

感谢！这个方法较为优雅地起了作用。我认为这个解决方法应该添加到错误提示中，以帮助人们更快地找到解决方法。
Thanks! This approach worked gracefully.  I think this solution should be added to the error message to help people find a solution faster.

cropse, 2024-09-28:
Thanks for the solution, but maybe we should add this solution in the error log? I encounter this issue, but in the error log well help a lot to begin with.