esphome/issues #3745: factory_reset will not reset total_daily_energy + manual reset documentation

skleeschulte, 2022-11-01:
### The problem

I need to manually reset a total_daily_energy sensor.

This is what I tried:
- Temporarily setting the `restore` property to `false` - after removing the restore property, the old stored value is back.
- Publishing a zero-value from a lambda function: `id(total_daily_energy_sensor).publish_state(0);` - the value is set to 0, but the next time the observed power sensor changes, the stored value is used again.
- Adding a `factory_reset` switch and toggling that from home assistant - while the log reads "Resetting to factory defaults..." and "Cleaning up preferences in flash...", the stored value is still there.

What worked was calling this lambda function: `id(total_daily_energy_sensor).publish_state_and_save(0);` (but only if `restore` is not `false`)

However, IMHO at least that the factory_reset did not work is a bug. In addition, it might be helpful to document how to do the manual reset.

### Which version of ESPHome has the issue?

2022.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP8266

### Board

D1 Mini

bkaufx, 2022-11-04:
If you add

```
esp8266:
  restore_from_flash: true
```

then will factory_reset clear it?

https://esphome.io/components/esp8266.html

I believe what is happening is that without restore_from_flash being set under the esp8266: section, ESPHome is still storing the value in non-volatile memory, just in a different area (RTC memory?) that is not subject to the factory_reset function.  I can see why this is confusing behavior in general.  

I could see the benefit of a function to reset the cumulative energy value, but I think it would just do the same thing as publish_state_and_save(0), so not sure if anything needs to be added.  Maybe documenting how to clear the value would be nice.

I think that maybe the factory_reset function could be modified to clear out the area of flash memory that is used when restore: is true under the total_daily_energy sensor but restore_from_flash: is false under the esp8266: section.

If restore is false, then just resetting the device should clear the value.

The fact that you are changing configurations confuses things.  You are never going to be able to erase a stored value from a different configuration, unless it happens just by chance.  So if you go to restore: false and try to clear, then when you change back to restore: true that stored value will never have been cleared.  You have to stay in the desired configuration and clear while in that configuration.

camsaway, 2022-11-30:
I think I'm hitting the same/similar issue.

Perhaps the restore: parameter could have a reset option that invokes the function properly?

bkaufx, 2022-11-30:
Or did you just on_boot clear the value once, then remove that code?

camsaway, 2022-11-30:
I had to leave the code there. Tried taking it away and the old value came
back.

Do we know where the value is saved?  I'm happy to use a utility to wipe
that memory.

Sent from my mobile phone so may be briefer than normal...

Regards

Cameron Gray
079 528 4802
***@***.***

https://www.linkedin.com/in/camerongray76