esphome/issues #3748: tr-502msv transmitting not working

ale1800, 2022-11-01:
### The problem

Hi everyone, I have this RF remote from LIDL: tr-502msv that controls some RF sockets:
[image]
I can receive his messages:
[image]
But I can't reproduce them to turn on and off the plugs.
This is my configuration:
[image]
If I activate the switch nothing happens. If I change protocol to another one, like 5 or 6, I can receive something but without the last "1"
Is it a bug or I'm just interpreting in a wrong way the remote messages?
Thanks in advance

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

rf receiver / rf transmitter

### Example YAML snippet

```yaml
remote_receiver:
  - id: RF_RX
    pin: 
      number: GPIO5
      inverted: True
    dump: 
      - rc_switch
    tolerance: 35%
    filter: 200us
    idle: 4ms
    buffer_size: 3kb

remote_transmitter:
  - id: RF_TX
    pin: GPIO18
    carrier_duty_percent: 100%

switch:
  - platform: template
    name: RF Power Button
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          transmitter_id: RF_TX
          protocol: 2
          code: '10110000000000010001'
```

zcftr29, 2023-02-26:
Hi [ale1800](https://github.com/ale1800),

I have a couple of set of these RF sockets (purchased from Lidl many years ago), and I would like to get them working with HA using my Broadlink RM Pro.  Unfortunately my controller has died so I have no way of learning the RF codes.  Is there any way you could provide me with a list of b64 codes to see if I can get them working again?

Many thanks in advance

ale1800, 2023-02-26:
> Hi [ale1800](https://github.com/ale1800),
>
> I have a couple of set of these RF sockets (purchased from Lidl many years ago), and I would like to get them working with HA using my Broadlink RM Pro. [...]

Hi @zcftr29, sure! But I only have the binary commands.
These are the commands to turn on and off the socket. It's not a standard protocol 2. I think that these commands are the "general ones", those who toggle all the sockets. If you want, as soo as I can, I can send you the  code for every button of the remote

```
    on_turn_on:
      - remote_transmitter.transmit_rc_switch_raw:
          transmitter_id: RF_TX
          protocol:
            pulse_length: 650
            inverted: True
            sync: [1, 10]
            zero: [1 ,2]
            one:  [2, 1]
          code: "10110000000000010001"
          repeat: 
            times: 10
            wait_time: 30ms
    on_turn_off:
      - remote_transmitter.transmit_rc_switch_raw:
          transmitter_id: RF_TX
          protocol:
            pulse_length: 650
            inverted: True
            sync: [1, 10]
            zero: [1 ,2]
            one:  [2, 1]
          code: "10110000000000000000"
          repeat: 
            times: 10
            wait_time: 30ms 
```

closed 2023-02-26 by ale1800

zcftr29, 2023-02-26:
Hi [ale1800](https://github.com/ale1800),
Thanks for the prompt reply... I don't think these codes can be used with the Broadlink though.  If there is any way you can capture them in the format that the RM Pro uses that would be amazing!.  I note that you have an RM3 mini, but am I right in thinking that only does IR?  I did find this project (https://github.com/lbschenkel/broadlink-bridge) which might be able to use some raw LIRC codes I have found elsewhere for an Aldi remote which might work (https://gathering.tweakers.net/forum/list_message/29516055#29516055)

ale1800, 2023-02-26:
@zcftr29 Yeah, the RM3 mini is IR only, so I can't help you on that side

zcftr29, 2023-02-26:
Thanks all the same...