esphome/issues #3751: BLEWrite and Climate Preset causes panic

camsaway, 2022-11-01:
### The problem

After being forced to change climate sensor config to use preset, my device comes online but no sensors report.

I connected cable and see it's going into a boot loop. (See log output).

The immediate commands before are set default preset
which triggers an update with BLEWrite.

As BLE is not yet connected, I suspect this is causing the failure.

### Which version of ESPHome has the issue?

2022.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

9.2

### What platform are you using?

ESP32

### Board

NodeMCU Clone

### Component causing the issue

blewrite & climate sensor

### Example YAML snippet

```yaml
esphome:
  name: grainfather2

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging - NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE
logger:
  level: VERBOSE

mqtt:
  broker: !secret homeassistantIP
  username: esphome
  password: !secret mqtt_password
#  on_message:
#    topic: helper/input_boolean/esphome_sleepoff_pool
#    then:
#      - binary_sensor.template.publish:
#          id: prevent_deep_sleep
#          state: !lambda 'if(x=="on"){return 1;}else{return 0;}'

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  manual_ip:     #Optional Manual IP
#    static_ip: 192.168.1.84
#    gateway: 192.168.1.250
[... 409 lines cut ...]
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:243]: Log initialized
[C][ota:465]: There have been 2 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[V][esp32.preferences:125]: sync: key: 233825507, len: 4
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written,                                                                  0 failed
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[V][esp32.preferences:058]: nvs_get_blob('2662315803'): ESP_ERR_NVS_NOT_FOUND -                                                                  the key might not be set yet
[I][thermostat.climate:1001]: Custom preset start requested
[D][GFCode:274]: Writing some value
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandl                                                                 ed.
Core 1 register dump:
PC      : 0x400eb234  PS      : 0x00060b30  A0      : 0x800ea230  A1      : 0x3f                                                                 fccc00
A2      : 0x0000003c  A3      : 0x3ffcea24  A4      : 0x00000000  A5      : 0x00                                                                 000000
A6      : 0x3ffccbf4  A7      : 0x00000051  A8      : 0x00000024  A9      : 0x00                                                                 000018
A10     : 0x00000005  A11     : 0x3f405e94  A12     : 0x00000112  A13     : 0x3f                                                                 405ea6
A14     : 0x3ffcf770  A15     : 0x3ffcf52c  SAR     : 0x00000004  EXCCAUSE: 0x00                                                                 00001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xff                                                                 ffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400eb234:0x3ffccc00 0x400ea22d:0x3ffccc40 0x400ea1bb:0x3ffccc60 0x4                                                                 01f50d3:0x3ffccc80 0x400e1a06:0x3ffccca0 0x400e2f5e:0x3ffcccc0 0x400e32e1:0x3ffc                                                                 cce0 0x400e34af:0x3ffccd20 0x400e3534:0x3ffccdc0 0x401f5745:0x3ffccde0 0x401f583                                                                 d:0x3ffcce00 0x400e8971:0x3ffcce20 0x400ecf31:0x3ffcce70 0x400f9c3a:0x3ffcd030 0                                                                 x400909d6:0x3ffcd050
```

camsaway, 2022-11-01:
Potentially raised in error.

closed 2022-11-01 by camsaway

tomfool1985, 2022-11-17:
Hello camsaway, can you help with an issue that i have, i am trying to do the same that you did with your gf3,  how you that the total lenght is 19bits?

camsaway, 2022-11-18:
Hi @tomfool1985 - I do still have some issues and some panics with some of my code (I just don't have time to investigate properly at the moment). What is your exact issue?

The documentation I referenced, https://github.com/kingpulsar/Grainfather-Bluetooth-Protocol, indicates that all messages must be exactly 19 characters (not bits) - with any unused characters padded with space.

My main problem is converting/casting between the different data types being used in the BLEWrite action. There are some data types used in ESPhome and this integration that I'm not used to working with. (and I'm not a programmer, I just hack my way around)

tomfool1985, 2022-11-18:
Thanks Cameron,

my issue is that i am not able to write without respond (operation code 0x12), do you habe6any idea how to do that?

thanks

Ottenere Outlook per Android<https://aka.ms/AAb9ysg>

camsaway, 2022-11-19:
Sorry. Not familiar. Where do you see that? Can you share your YAML config
and the error logs?

But perhaps do so as a new thread on esphome forum, not here on code bugs.
My username there is the same.

Sent from my mobile phone so may be briefer than normal...

Regards

Cameron Gray
079 528 4802
***@***.***

https://www.linkedin.com/in/camerongray76