esphome/issues #3752: Validation error for dallas component if update_interval set

sebcaps, 2022-11-01:
### The problem

Configure dallas component in yaml file.
Set mandatory Id to a correct on (eg : ``D3``)
Add ``update_interval``option => **Validation** error raised

See screenshot below

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

dallas

### Example YAML snippet

```yaml
dallas:
  - pin: D3
  - update_interval : 60s
```

### Anything in the logs that might be useful for us?

```txt
**Config without update_interval**
![image](https://user-images.githubusercontent.com/24374878/199302730-0461191b-f156-404f-8b5f-425ef3a18f12.png)

**Config with update_interval**
![image](https://user-images.githubusercontent.com/24374878/199303007-b6b30973-2a37-4d19-b075-016c461cfcfc.png)
```

ssieb, 2022-11-01:
Your yaml is wrong.  You're making a second dallas component by putting that `-` there.  GitHub isn't for support.  Use Discord or one of the forums.

sebcaps, 2022-11-01:
My bad :confused:
This is not my first esphome yaml, but still have trouble with yaml...
Sorry for the inconvenience

closed 2022-11-01 by sebcaps