esphome/issues #3755: Homeassistant is getting unresponsive when updating esp8266

liakjim, 2022-11-02:
### The problem

What title says. Only when updating esp8266's. Esp32's are updating just fine. I have to remove power to my raspberry in order to have my system run again. 

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2022.10.5

### What platform are you using?

ESP8266

### Board

d1 mini

ssieb, 2022-11-02:
Assuming you are using a pi3, use dev esphome with https://next.esphome.io/components/esphome.html#:~:text=compile_process_limit

liakjim, 2022-11-03:
> Assuming you are using a pi3, use dev esphome with https://next.esphome.io/components/esphome.html#:~:text=compile_process_limit

I am using raspberry pi4 2GB.

ssieb, 2022-11-03:
Ok, same thing, too little memory.  Use 1.

liakjim, 2022-11-03:
> Ok, same thing, too little memory. Use 1.

Just to confirm. 
I need to add compile_process_limit=1 inside each 8266's yaml?

liakjim, 2022-11-03:
Ok, i can confirm that ```compile_process_limit: 1``` did the trick. I had to install dev esphome add-on in order to accept this setting.

closed 2022-11-03 by liakjim