esphome/issues #3756: Pairing attempt failed with an unhandled exception pairing Level Lock through HomeKit Controller

steve28, 2022-11-03:
### The problem

When I try to pair my Level Lock through HomeKit Controller integration, I get an unhandled exception error and pairing times out.

### Which version of ESPHome has the issue?

10.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.0

### What platform are you using?

ESP32

### Board

Nodemcu ESP32

### Component causing the issue

Bluetooth Proxy

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-3c80c0
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
```

### Anything in the logs that might be useful for us?

```txt
Pairing attempt failed with an unhandled exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 83, in _async_wrap_bluetooth_operation
    return await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 389, in read_gatt_char
    return await self._client.bluetooth_gatt_read(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 598, in bluetooth_gatt_read
    resp = await self._send_bluetooth_message_await_response(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 443, in _send_bluetooth_message_await_response
    raise BluetoothGATTAPIError(BluetoothGATTError.from_pb(resp[0]))
aioesphomeapi.core.BluetoothGATTAPIError: Bluetooth GATT Error address=D2:3E:18:F4:88:9E handle=68 error=133 description=Error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/config_flow.py", line 479, in async_step_pair
    self.finish_pairing = await discovery.async_start_pairing(self.hkid)
  File "/usr/local/lib/python3.10/site-packages/bleak_retry_connector/__init__.py", line 652, in _async_wrap_bluetooth_connection_error_retry
    return await func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/discovery.py", line 151, in async_start_pairing
    salt, pub_key = await self._async_start_pairing(alias)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/discovery.py", line 141, in _async_start_pairing
    return await drive_pairing_state_machine(
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 230, in drive_pairing_state_machine
    decoded = await _pairing_char_write(client, char, iid, request)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 184, in _pairing_char_write
    data = await char_write(client, None, None, handle, iid, next_write)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 167, in char_write
    pdu_status, data = await ble_request(
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 73, in ble_request
    return await _read_pdu(client, decryption_key, handle, tid)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 109, in _read_pdu
    data = await client.read_gatt_char(handle)
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 571, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 71, in _async_wrap_bluetooth_connected_operation
    return next(iter(done)).result()
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 87, in _async_wrap_bluetooth_operation
    raise BleakError(str(err)) from err
bleak.exc.BleakError: Bluetooth GATT Error address=D2:3E:18:F4:88:9E handle=68 error=133 description=Error
```

This is then followed by:

```
Logger: homeassistant.components.homekit_controller.config_flow
Source: components/homekit_controller/config_flow.py:479 
Integration: homekit_controller (documentation, issues) 
First occurred: 6:57:21 PM (1 occurrences) 
Last logged: 6:57:21 PM

Pairing attempt failed with an unhandled exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bleak_retry_connector/__init__.py", line 543, in establish_connection
    await client.connect(
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/models.py", line 306, in connect
    return await super().connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 190, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/connection.py", line 47, in establish_connection
    return await retry_establish_connection(
  File "/usr/local/lib/python3.10/site-packages/bleak_retry_connector/__init__.py", line 616, in establish_connection
    _raise_if_needed(name, description, exc)
  File "/usr/local/lib/python3.10/site-packages/bleak_retry_connector/__init__.py", line 491, in _raise_if_needed
    raise BleakAbortedError(f"{msg}: {ABORT_ADVICE}") from exc
bleak_retry_connector.BleakAbortedError: A223N0DP (D2:3E:18:F4:88:9E) - D2:3E:18:F4:88:9E -> esp32-bluetooth-proxy-3c80c0: Failed to connect: [org.bluez.Error.Failed] le-connection-abort-by-local: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/config_flow.py", line 479, in async_step_pair
    self.finish_pairing = await discovery.async_start_pairing(self.hkid)
  File "/usr/local/lib/python3.10/site-packages/bleak_retry_connector/__init__.py", line 652, in _async_wrap_bluetooth_connection_error_retry
    return await func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/discovery.py", line 151, in async_start_pairing
    salt, pub_key = await self._async_start_pairing(alias)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/discovery.py", line 120, in _async_start_pairing
    await self._ensure_connected()
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/discovery.py", line 91, in _ensure_connected
    self.client = await establish_connection(
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/connection.py", line 57, in establish_connection
    raise AccessoryDisconnectedError(ex) from ex
aiohomekit.exceptions.AccessoryDisconnectedError: A223N0DP (D2:3E:18:F4:88:9E) - D2:3E:18:F4:88:9E -> esp32-bluetooth-proxy-3c80c0: Failed to connect: [org.bluez.Error.Failed] le-connection-abort-by-local: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference
````
```

stomko11, 2022-11-03:
I experienced the same issue. I wanted to try my Tedee lock (paired via homekit controller) - I have it paired to my HA via USB BT dongle. I also have esp active BT proxy set in my network, so what I did was the usual:
- removed current HomeKit Controller integration
- removed BT dongle and restarted HA
- in my Tedee android app, i have removed device from my account, which sets lock to pairing mode automatically
- Device got auto discovered in HA
- When selecting "configure", I am prompted for HomeKit PIN, but there is error/warning in red saying: "An unhandled error occured while attempting to pair with this device. This may be temporary failure or your device may not be supported currently." When I see this, device just won't pair, and I am continously being prompted for PIN
- switching back to BT dongle - no such error/warning is present

austwhite, 2022-11-04:
I wonder if my issue is related to yours.  I just logged a new issue then saw this one.
https://github.com/esphome/issues/issues/3761

mrrobotnic, 2022-11-12:
Same setup issue with an August Smart Lock Pro 3rd Generation where it cannot complete setup using the BT Proxy.