esphome/issues #3757: ESPHome integration "Setup another instance.." does not work

jmBrosch, 2022-11-03:
### The problem

When adding an ESPHome Integration the menu item "Setup another instance of ESPHome" does not show any action.
Action: Home Assistant -> Settings -> Devices & Services -> Integrations -> ADD INTEGRATION -> ESPHome -> "Setup another instance of ESPHome" : nothing happens.
It worked before the latest update (ok with Home Assistant 2022.10.5 and ESPHome 2022.10.1, not ok with Home Assistant 2022.11.0 and ESPHome 2022.10.2). Tested with Google Chrome and Firefox. 

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.0

### What platform are you using?

ESP32

### Board

ESP32 D1 mini and all others

### Component causing the issue

Integration

### Example YAML snippet

```yaml
Independent from YAML File
```

### Additional information

[image]

tbone004, 2022-11-11:
Mine is also doing the same thing. Is there anyone looking into the issue yet, I see it is not assigned yet.

Thanks

ssieb, 2022-11-11:
What are you expecting it to do?

tbone004, 2022-11-12:
Past versions allowed you to add a esphome device that was not picked up by discovery. In my case I want to add a device that wasn't in the list, so normally I had an option to give the IP and Port of the device to add. When clicking the link at the bottom now does nothing.

I get there might be an issue of why the device didn't get pick up by discovery, in my case I rebooted the device and it showed up so I can add it from the list. But even though that worked for me, there is now no manual way to do this unless I am looking in the wrong area.

nagyrobi, 2022-11-12:
Yep, that link does not work. Pretty annoying, because if the list above populates with discovered devices, manual addition is impossible.

jesserockz, 2022-11-12:
This is a home assistant core issue (and should be reported there). 
I can reproduce this myself on my home assistant instance, but I do not have any other discovered integrations to test it's only esphome that is doing it.

jesserockz, 2022-11-12:
Ok, just confirmed it's happening with wled too.
So please report on ha core issues as it is not esphome specific.

jesserockz, 2022-11-13:
https://github.com/home-assistant/frontend/issues/14288

closed 2022-11-13 by jesserockz