esphome/issues #3760: ESP-WROOM-32 DevKit WiFi Problem

anti-spy, 2022-11-03:
### The problem

The situation is confusing.
My esp32 won't connect to wifi. My network initially had one ESP32, everything worked.

When connecting a new one, I ran into this problem.
Reset the router, did not help. Tried to configure the router, I thought the problem.

After 3 days. I realized that the problem is not with the network. In ESP32
Funny how the first ESP32 I had now can't connect either. But if you register AP Mobile, it will be able to connect to it.

I put tasmota, she connected to the network. I installed an empty firmware for ESPHome, it connected. I installed the full firmware of ESPHome, it did not connect. I tried to put it again empty - it did not connect. Put tasmota - everything works again.

**Chip info**
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: C0:49:EF:CA:62:1C

**Log Tasmota**

[20:22:40]00:00:00.375 Project tasmota - Tasmota Version 12.2.0(tasmota)-2_0_5(2022-10-17T08:39:59)
[20:22:41]00:00:01.042 WIF: Connecting to AP1 VIP NETWORK in mode 11n as tasmota-CA621C-0540...
[20:22:44]00:00:04.047 WIF: Connected
[20:22:45]00:00:04.260 HTP: Web server active on tasmota-CA621C-0540 with IP address 192.168.100.101

**Log 1 empty firmware ESPHome**

[20:05:41][C][wifi:037]: Setting up WiFi...
[20:05:42][D][wifi:384]: Starting scan...
[20:05:43][D][wifi:399]: Found networks:
[20:05:43][I][wifi:443]: - 'VIP NETWORK' (50:64:2B:A9:80:6D) ▂▄▆█
[20:05:43][D][wifi:444]:     Channel: 9
[20:05:43][D][wifi:445]:     RSSI: -64 dB
[20:05:43][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[20:05:45][W][wifi_esp32:492]: Event: Disconnected ssid='VIP NETWORK' reason='Probe Request Unsuccessful'
[20:05:45][W][wifi:557]: WiFi network can not be found anymore.
[20:05:45][W][wifi:583]: Restarting WiFi adapter...
[20:05:45][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[20:05:46][I][wifi:516]: WiFi Connected!
[20:05:46][C][wifi:360]:   Local MAC: C0:49:EF:CA:62:1C
[20:05:46][C][wifi:361]:   SSID: 'VIP NETWORK'
[20:05:46][C][wifi:362]:   IP Address: 192.168.100.101
[20:05:46][C][wifi:364]:   BSSID: 50:64:2B:A9:80:6D
[20:05:46][C][wifi:365]:   Hostname: 'burner'
[20:05:46][C][wifi:367]:   Signal strength: -56 dB ▂▄▆█
[20:05:46][C][wifi:371]:   Channel: 9
[20:05:46][C][wifi:372]:   Subnet: 255.255.255.0
[20:05:46][C][wifi:373]:   Gateway: 192.168.100.1
[20:05:46][C][wifi:374]:   DNS1: 192.168.100.1
[20:05:46][C][wifi:375]:   DNS2: 0.0.0.0
[20:05:46][D][wifi:525]: Disabling AP...
[20:05:46][C][ota:089]: Over-The-Air Updates:
[20:05:46][C][ota:090]:   Address: burner.local:3232
[20:05:46][C][ota:093]:   Using Password.

**Log Full firmware ESPHome**

[21:06:00][D][wifi:399]: Found networks:
[21:06:00][I][wifi:443]: - 'VIP NETWORK' (50:64:2B:A9:80:6D) ▂▄▆█
[21:06:00][D][wifi:444]:     Channel: 9
[21:06:00][D][wifi:445]:     RSSI: -77 dB
[21:06:00][D][wifi:448]: - 'Termostat Fallback Hotspot' (EC:62:60:94:D0:89) ▂▄▆█
[21:06:00][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[21:06:01][W][wifi_esp32:495]: Event: Disconnected ssid='VIP NETWORK' bssid=50:64:2B:A9:80:6D reason='Auth Expired'
[21:06:01][W][wifi:547]: Error while connecting to network.
[21:06:01][W][wifi:583]: Restarting WiFi adapter...
[21:06:01][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[21:06:02][W][wifi_esp32:495]: Event: Disconnected ssid='VIP NETWORK' bssid=50:64:2B:A9:80:6D reason='Auth Expired'
[21:06:02][W][wifi:547]: Error while connecting to network.
[21:06:02][W][wifi:583]: Restarting WiFi adapter...

**Again log empty firmware ESPHome**

[21:09:18][I][logger:243]: Log initialized
[21:09:18][C][ota:465]: There have been 0 suspected unsuccessful boot attempts.
[21:09:18][D][esp32.preferences:113]: Saving 1 preferences to flash...
[21:09:18][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[21:09:18][I][app:029]: Running through setup()...
[21:09:18][C][wifi:037]: Setting up WiFi...
[21:09:19][D][wifi:384]: Starting scan...
[21:09:20][D][wifi:399]: Found networks:
[21:09:20][I][wifi:443]: - 'VIP NETWORK' (50:64:2B:A9:80:6D) ▂▄▆█
[21:09:20][D][wifi:444]:     Channel: 9
[21:09:20][D][wifi:445]:     RSSI: -67 dB
[21:09:20][D][wifi:448]: - 'Termostat Fallback Hotspot' (EC:62:60:94:D0:89) ▂▄▆█
[21:09:20][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[21:09:22][W][wifi_esp32:495]: Event: Disconnected ssid='VIP NETWORK' bssid=50:64:2B:A9:80:6D reason='Auth Expired'
[21:09:22][W][wifi:547]: Error while connecting to network.
[21:09:22][W][wifi:583]: Restarting WiFi adapter...
[21:09:22][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[21:09:23][W][wifi_esp32:495]: Event: Disconnected ssid='VIP NETWORK' bssid=50:64:2B:A9:80:6D reason='Auth Expired'
[21:09:23][W][wifi:547]: Error while connecting to network.
[21:09:23][W][wifi:583]: Restarting WiFi adapter...
[21:09:29][D][wifi:384]: Starting scan...
[21:09:33][D][wifi:399]: Found networks:
[21:09:33][I][wifi:443]: - 'VIP NETWORK' (50:64:2B:A9:80:6D) ▂▄▆█
[21:09:33][D][wifi:444]:     Channel: 9
[21:09:33][D][wifi:445]:     RSSI: -61 dB
[21:09:33][D][wifi:448]: - 'Termostat Fallback Hotspot' (EC:62:60:94:D0:89) ▂▄▆█
[21:09:33][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[21:09:34][W][wifi_esp32:495]: Event: Disconnected ssid='VIP NETWORK' bssid=50:64:2B:A9:80:6D reason='Auth Expired'
[21:09:34][W][wifi:547]: Error while connecting to network.
[21:09:34][W][wifi:583]: Restarting WiFi adapter...
[21:09:34][I][wifi:255]: WiFi Connecting to 'VIP NETWORK'...
[21:09:35][W][wifi_esp32:495]: Event: Disconnected ssid='VIP NETWORK' bssid=50:64:2B:A9:80:6D reason='Auth Expired'
[21:09:35][W][wifi:547]: Error while connecting to network.
[21:09:35][W][wifi:583]: Restarting WiFi adapter...

**Connecting full firmware log ESPHome to a mobile AP (SSID with a dot)**

[21:21:28][D][wifi:384]: Starting scan...
[21:21:32][D][wifi:399]: Found networks:
[21:21:32][I][wifi:443]: - 'VIP NETWORK.' (0A:C5:E1:1A:CE:70) ▂▄▆█
[21:21:32][D][wifi:444]:     Channel: 9
[21:21:32][D][wifi:445]:     RSSI: -33 dB
[21:21:32][D][wifi:448]: - 'VIP NETWORK' (50:64:2B:A9:80:6D) ▂▄▆█
[21:21:32][I][wifi:255]: WiFi Connecting to 'VIP NETWORK.'...
[21:21:36][I][wifi:516]: WiFi Connected!
[21:21:36][C][wifi:360]:   Local MAC: C0:49:EF:CA:62:1C
[21:21:36][C][wifi:361]:   SSID: 'VIP NETWORK.'
[21:21:36][C][wifi:362]:   IP Address: 192.168.43.192
[21:21:36][C][wifi:364]:   BSSID: 0A:C5:E1:1A:CE:70
[21:21:36][C][wifi:365]:   Hostname: 'burner'
[21:21:36][C][wifi:367]:   Signal strength: -33 dB ▂▄▆█
[21:21:36][C][wifi:371]:   Channel: 9
[21:21:36][C][wifi:372]:   Subnet: 255.255.255.0
[21:21:36][C][wifi:373]:   Gateway: 192.168.43.1
[21:21:36][C][wifi:374]:   DNS1: 192.168.43.1
[21:21:36][C][wifi:375]:   DNS2: 0.0.0.0
[21:21:36][D][wifi:525]: Disabling AP...
[21:21:36][C][web_server:092]: Setting up web server...
[21:21:36][C][ota:089]: Over-The-Air Updates:
[21:21:36][C][ota:090]:   Address: burner.local:3232
[21:21:36][C][ota:093]:   Using Password.
[21:21:36][C][api:025]: Setting up Home Assistant API server...
[21:21:36][I][app:062]: setup() finished successfully!
[21:21:36][D][sensor:127]: 'WiFi Signal oil-burner': Sending state -33.00000 dBm with 0 decimals of accuracy
[21:21:36][I][app:102]: ESPHome version 2022.10.2 compiled on Nov  3 2022, 19:17:52
[21:21:36][C][wifi:502]: WiFi:
[21:21:36][C][wifi:360]:   Local MAC: C0:49:EF:CA:62:1C
[21:21:36][C][wifi:361]:   SSID: 'VIP NETWORK.'
[21:21:36][C][wifi:362]:   IP Address: 192.168.43.192
[21:21:36][C][wifi:364]:   BSSID: 0A:C5:E1:1A:CE:70
[21:21:36][C][wifi:365]:   Hostname: 'burner'
[21:21:36][C][wifi:367]:   Signal strength: -33 dB ▂▄▆█
[21:21:36][C][wifi:371]:   Channel: 9
[21:21:36][C][wifi:372]:   Subnet: 255.255.255.0
[21:21:36][C][wifi:373]:   Gateway: 192.168.43.1
[21:21:36][C][wifi:374]:   DNS1: 192.168.43.1
[21:21:36][C][wifi:375]:   DNS2: 0.0.0.0

Other devices on the network (esp8266) are not experiencing network issues.
**I don't understand why.**

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP32

### Board

ESP-WROOM-32 DevKit

### Component causing the issue

Wi-Fi

jjvandenberg, 2022-11-20:
Exact same issue. Will connect to ISP provider's AP, will not connect to Aerohive AP121's.

anti-spy, 2022-12-04:
If you remove:

**captive_portal:** 
and 
**ap:**

Then sometimes the device appears on the network. 50% to 50%
Problem in ESPHome

jehy, 2022-12-12:
Removing captive_portal an ap helped me too. With those enabled I was unable to connect to wifi.

anti-spy, 2022-12-23:
I repurchased esp32. She works.

closed 2022-12-24 by anti-spy