esphome/issues #3761: Lock not connecting to Bluetooth Using YaleAccess Bluetooth integration

austwhite, 2022-11-04:
### The problem

I have logged this in https://github.com/home-assistant/core/issues/81468 but @bdraco looked over the logs and advised me to log this as an ESPHome isssue.
When using Bluetooth Proxy, my August Lock, using the Yale Access Bluetooth integration, is unable to connect to the ESPHome Bluetooth Proxy.   I have the Proxy less than 4 feet from the lock and still it will not connect.
Logs show connection attempts, but they are failing.
I have included the last logging done.  Hopefully it will be useful

### Which version of ESPHome has the issue?

2022.10.2 - but current DEV version

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.1

### What platform are you using?

ESP32

### Component causing the issue

Bluetooth Proxy

### Example YAML snippet

```yaml
substitutions:
  name: "kitchenbleproxy"
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
logger:
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
2022-11-04 20:37:37.158 DEBUG (MainThread) [yalexs_ble.push] Waiting for advertisement callbacks for FRONT DOOR (78:9C:85:01:CB:3D)
2022-11-04 20:37:42.671 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] kitchenbleproxy: Connecting scanner version=2, connectable=True
2022-11-04 20:37:43.077 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenbleproxy: BLE connection limits: used=0 free=3 limit=3
2022-11-04 20:37:43.240 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.LightState'>, 174803132): LightState(key=174803132, state=True, brightness=0.9019608, color_mode=3, color_brightness=1.0, red=1.0, green=1.0, blue=1.0, white=1.0, color_temperature=0.0, cold_white=1.0, warm_white=1.0, effect='')
2022-11-04 20:37:43.306 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.LightState'>, 2472414331): LightState(key=2472414331, state=True, brightness=0.9019608, color_mode=3, color_brightness=1.0, red=1.0, green=1.0, blue=1.0, white=1.0, color_temperature=0.0, cold_white=1.0, warm_white=1.0, effect='')
2022-11-04 20:37:43.307 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.LightState'>, 3184812465): LightState(key=3184812465, state=False, brightness=1.0, color_mode=3, color_brightness=1.0, red=1.0, green=1.0, blue=1.0, white=1.0, color_temperature=0.0, cold_white=1.0, warm_white=1.0, effect='')
2022-11-04 20:37:43.312 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.LightState'>, 3184812470): LightState(key=3184812470, state=False, brightness=1.0, color_mode=3, color_brightness=1.0, red=1.0, green=1.0, blue=1.0, white=1.0, color_temperature=0.0, cold_white=1.0, warm_white=1.0, effect='')
2022-11-04 20:37:43.326 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.SensorState'>, 1013851870): SensorState(key=1013851870, state=39.0, missing_state=False)
2022-11-04 20:37:43.412 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.SwitchState'>, 320112106): SwitchState(key=320112106, state=False)
2022-11-04 20:38:02.988 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.SensorState'>, 1013851870): SensorState(key=1013851870, state=39.0625, missing_state=False)
2022-11-04 20:38:30.266 DEBUG (MainThread) [homeassistant.components.bluetooth] Triggering bluetooth usb discovery
2022-11-04 20:38:35.368 DEBUG (MainThread) [yalexs_ble.push] Waiting for advertisement callbacks for FRONT DOOR (78:9C:85:01:CB:3D)
2022-11-04 20:38:36.158 DEBUG (MainThread) [homeassistant.components.bluetooth] Rediscovered adapters: {}
2022-11-04 20:39:02.985 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.SensorState'>, 1013851870): SensorState(key=1013851870, state=39.0, missing_state=False)
2022-11-04 20:39:28.557 DEBUG (MainThread) [homeassistant.components.esphome] kitchenbleproxy: 192.168.0.120 disconnected, running disconnected callbacks
2022-11-04 20:39:28.866 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] kitchenbleproxy: Connecting scanner version=2, connectable=True
2022-11-04 20:39:28.948 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenbleproxy: BLE connection limits: used=0 free=3 limit=3
2022-11-04 20:39:29.154 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: F9:C8:D8:80:3C:96 AdvertisementData(local_name='S', manufacturer_data={880: b'\x05\xe4\x00\x00Kitchen0 '}, tx_power=-127, rssi=-80) connectable: True match: set() rssi: -80
2022-11-04 20:39:29.156 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: 31:9A:74:40:6C:82 AdvertisementData(manufacturer_data={6: b'\x01\t \x02l\x0f\x8e\x109\t\x08\xe7\xe5\xaf\x99\xa5\xf2\xf2&\xc5\xd55\x96g;N\x06'}, tx_power=-127, rssi=-58) connectable: True match: set() rssi: -58
2022-11-04 20:39:29.255 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: D4:84:57:4B:35:D3 AdvertisementData(local_name='net', manufacturer_data={1704: b'\x01\x01\x002\xd4\x84WK5\xd3'}, service_uuids=['0000ff90-0000-1000-8000-00805f9b34fb', '0000ff80-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-83) connectable: True match: set() rssi: -83
2022-11-04 20:39:29.281 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: 51:65:DF:75:88:27 AdvertisementData(manufacturer_data={224: b'\x14\x02\xca\x92n\xad'}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-70) connectable: True match: set() rssi: -70
2022-11-04 20:39:29.563 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: 4E:4E:4E:7A:8A:45 AdvertisementData(manufacturer_data={224: b'\x00 \xca8\xda\xac'}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-61) connectable: True match: set() rssi: -61
2022-11-04 20:39:29.564 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: FC:4F:A2:51:AD:E2 AdvertisementData(local_name='Bathroom globe 2', service_data={'0000fe0f-0000-1000-8000-00805f9b34fb': b'\x1b\x10l\x96\x02'}, service_uuids=['0000fe0f-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-82) connectable: True match: set() rssi: -82
2022-11-04 20:39:29.564 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: 78:9C:85:01:CB:3D AdvertisementData(local_name='AugSW38', manufacturer_data={76: b'\x06\xad\x01\x07\x87\xca\xe2\x86\xaa\x06\x00\x01\x00\x01\x02', 22: b',\x0e\xa5\xcc\xc9\xd5[\xc1D\x8d\x8aE<\x90\x14u\xd5k\x00\x00\x00\x00\x00'}, tx_power=-127, rssi=-76) connectable: True match: {'homekit_controller'} rssi: -76
[... 132 lines cut ...]
2022-11-04 20:40:09.089 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenbleproxy: BLE connection limits: used=1 free=2 limit=3
2022-11-04 20:40:09.090 DEBUG (MainThread) [bleak_retry_connector] FRONT DOOR (78:9C:85:01:CB:3D) - 78:9C:85:01:CB:3D -> kitchenbleproxy: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, backing off: 0.55 (attempt: 6, last rssi: -75)
2022-11-04 20:40:09.090 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenbleproxy: BLE connection limits: used=1 free=2 limit=3
2022-11-04 20:40:09.641 DEBUG (MainThread) [bleak_retry_connector] FRONT DOOR (78:9C:85:01:CB:3D) - 78:9C:85:01:CB:3D -> kitchenbleproxy: Connecting (attempt: 7, last rssi: -75)
[... cut ...]
2022-11-04 20:40:17.018 DEBUG (MainThread) [bleak_retry_connector] FRONT DOOR (78:9C:85:01:CB:3D) - 78:9C:85:01:CB:3D -> kitchenbleproxy: Connecting (attempt: 8, last rssi: -75)
2022-11-04 20:40:17.019 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] kitchenbleproxy: Checking can connect, available=True, ble_connections_free=2 result=True
2022-11-04 20:40:17.122 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenbleproxy: BLE connection limits: used=2 free=1 limit=3
```

### Additional information

The above log was the latest one that @bdraco looked at and determined I should likely log an issue with ESPHome team.

bdraco, 2022-11-04:
The expectation is that `ESP_GATT_CONN_FAIL_ESTABLISH` might be resolved by an ESP-IDF upgrade

InstigatorX, 2022-11-04:
> The expectation is that `ESP_GATT_CONN_FAIL_ESTABLISH` might be resolved by an ESP-IDF upgrade

How is that accomplished, @bdraco ?

austwhite, 2022-11-04:
> > The expectation is that `ESP_GATT_CONN_FAIL_ESTABLISH` might be resolved by an ESP-IDF upgrade
> 
> How is that accomplished, @bdraco ?

@bdraco left that message for the ESPHome team to look at.  As I mentioned in the original issue, I had worked with @bdraco  to get this far .

sarabveer, 2022-11-16:
Updated to 2022.11 today, getting the error`Cannot notify GATT characteristic, not found.` when trying to use the bluetooth proxy with the Yale Access HomeAssistant integration.

bdraco, 2022-11-16:
Well thats better since it means it getting further.  Please post the whole error, trace, and console logs from the esp device.

austwhite, 2022-11-21:
Here is an extract of my log file.
I hope there is something interesting in this for you @bdraco 

2022-11-22 08:04:55.856 DEBUG (MainThread) [yalexs_ble.push] FRONT DOOR (78:9C:85:01:CB:3D): Starting update
2022-11-22 08:04:55.856 DEBUG (MainThread) [yalexs_ble.lock] FRONT DOOR (78:9C:85:01:CB:3D): Connecting to the lock
2022-11-22 08:04:55.856 DEBUG (MainThread) [bleak_retry_connector] FRONT DOOR (78:9C:85:01:CB:3D) - 78:9C:85:01:CB:3D -> kitchenbleproxy: Connecting (attempt: 1, last rssi: -73)
2022-11-22 08:04:55.857 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] kitchenbleproxy: Checking can connect, available=True, ble_connections_free=2 result=True
2022-11-22 08:04:55.896 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenbleproxy: BLE connection limits: used=1 free=2 limit=3
2022-11-22 08:05:00.913 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: C4:7C:8D:6C:89:DC AdvertisementData(local_name='Flower care', service_data={'0000fe95-0000-1000-8000-00805f9b34fb': b'q \x98\x00O\xdc\x89l\x8d|\xc4\r\x08\x10\x01\x00'}, service_uuids=['0000fe95-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-74) connectable: True match: set() rssi: -74
2022-11-22 08:05:08.600 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.LightState'>, 174803132): LightState(key=174803132, state=True, brightness=0.9019608, color_mode=3, color_brightness=1.0, red=1.0, green=1.0, blue=1.0, white=1.0, color_temperature=0.0, cold_white=1.0, warm_white=1.0, effect='')
2022-11-22 08:05:08.631 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.LightState'>, 2472414331): LightState(key=2472414331, state=True, brightness=0.9019608, color_mode=3, color_brightness=1.0, red=1.0, green=1.0, blue=1.0, white=1.0, color_temperature=0.0, cold_white=1.0, warm_white=1.0, effect='')
2022-11-22 08:05:08.699 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: C4:7C:8D:6C:89:E4 AdvertisementData(local_name='Flower care', service_data={'0000fe95-0000-1000-8000-00805f9b34fb': b'q \x98\x00\xef\xe4\x89l\x8d|\xc4\r\x07\x10\x03\x00\x00\x00'}, service_uuids=['0000fe95-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-72) connectable: True match: set() rssi: -72
2022-11-22 08:05:08.892 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: C4:7C:8D:6C:89:DC AdvertisementData(local_name='Flower care', service_data={'0000fe95-0000-1000-8000-00805f9b34fb': b'q \x98\x00P\xdc\x89l\x8d|\xc4\r\t\x10\x02\x00\x00'}, service_uuids=['0000fe95-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-74) connectable: True match: set() rssi: -74
2022-11-22 08:05:11.233 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: 55:EB:27:30:A1:5B AdvertisementData(manufacturer_data={76: b'\x10\x05p\x1c\x98]j'}, tx_power=-127, rssi=-98) connectable: True match: set() rssi: -98
2022-11-22 08:05:12.666 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: C4:7C:8D:6C:89:E4 AdvertisementData(local_name='Flower care', service_data={'0000fe95-0000-1000-8000-00805f9b34fb': b'q \x98\x00\xf0\xe4\x89l\x8d|\xc4\r\x08\x10\x01\x1d'}, service_uuids=['0000fe95-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-78) connectable: True match: set() rssi: -78
2022-11-22 08:05:15.860 DEBUG (MainThread) [bleak_retry_connector] FRONT DOOR (78:9C:85:01:CB:3D) - 78:9C:85:01:CB:3D -> kitchenbleproxy: Timed out trying to connect (attempt: 1, last rssi: -73)
2022-11-22 08:05:16.111 DEBUG (MainThread) [bleak_retry_connector] FRONT DOOR (78:9C:85:01:CB:3D) - 78:9C:85:01:CB:3D -> kitchenbleproxy: Connecting (attempt: 2, last rssi: -73)
2022-11-22 08:05:16.111 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth] kitchenbleproxy: Checking can connect, available=True, ble_connections_free=2 result=True
2022-11-22 08:05:16.162 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenbleproxy: BLE connection limits: used=1 free=2 limit=3
2022-11-22 08:05:16.557 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] kitchenbleproxy: 78:9C:85:01:CB:3D AdvertisementData(local_name='Aug', manufacturer_data={465: b'\x00', 22: b' |\xa5\xcc\xc9\xd5[\xc1D\x8d\x8aE<\x90\x14u\xd5k\x01\x03\x04\x00\x00'}, service_uuids=['bd4ac610-0b45-11e3-8ffd-0800200c9a66'], tx_power=-127, rssi=-70) connectable: True match: set() rssi: -70
2022-11-22 08:05:23.245 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth.client] kitchenbleproxy: Aug - 78:9C:85:01:CB:3D: Connection state changed to connected=True mtu=300 error=0
2022-11-22 08:05:23.245 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth.client] kitchenbleproxy: Aug - 78:9C:85:01:CB:3D: Connection state changed to connected=True mtu=300 error=0
2022-11-22 08:05:23.245 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth.client] kitchenbleproxy: Aug - 78:9C:85:01:CB:3D: Connection state changed to connected=True mtu=300 error=0
2022-11-22 08:05:23.246 DEBUG (MainThread) [homeassistant.components.esphome.bluetooth.client] kitchenbleproxy: Aug - 78:9C:85:01:CB:3D: Connection state changed to connected=True mtu=300 error=0
[... cut ...] ate=False)
2022-11-22 08:08:55.891 DEBUG (MainThread) [homeassistant.components.bluetooth] Rediscovered adapters: {}
2022-11-22 08:09:07.368 DEBUG (MainThread) [yalexs_ble.push] Waiting for advertisement callbacks for FRONT DOOR (78:9C:85:01:CB:3D)
2022-11-22 08:09:33.134 DEBUG (MainThread) [homeassistant.components.esphome.entry_data] kitchenledstrip: dispatching update with key (<class 'aioesphomeapi.model.SensorState'>, 1013851870): SensorState(key=1013851870, state=32.5, missing_state=False)

bdraco, 2022-11-21:
There are still a few issues to work through that I outlined here https://github.com/esphome/esphome/pull/4049

austwhite, 2022-11-21:
If you need anything else, as far as testing or logs, please let me know :)

bdraco, 2022-11-22:
These should work with the following changes

1. Make sure you have the `web_server` component turned off https://github.com/esphome/esphome-docs/pull/2469
2. Install this PR to improve reliability https://github.com/esphome/esphome/pull/4049
3. Install these PRs to reduce memory pressure/improve performance:
   https://github.com/esphome/esphome/pull/4068
   https://github.com/esphome/esphome/pull/4069
   https://github.com/esphome/esphome/pull/4063 

I've merge all those together into https://github.com/bdraco/esphome/tree/ble_debug_services if someone wants to try it without having to merge everything.

Jaharmi, 2022-11-24:
Not sure if I did this correctly, but the config YAML at least validated. :)

Note that I'm up-to-date with ESPHome v2022.11.2. I had previously updated ("Update All") my device configs before trying to take some time today to try the bleeding edge.

I added this to the config for one of my ESP32s:

```
external_components:
  - source: github://bdraco/esphome@ble_debug_services
    components: [ esp32_ble_client, esp32_ble_tracker, bluetooth_proxy ]
```

It validated — after multiple previous attempts to do the same thing didn't — so I then tried to install it wirelessly. I got the following:

```
src/esphome/components/esp32_ble_client/ble_characteristic.cpp: In member function 'esphome::esp32_ble_client::BLEDescriptor* esphome::esp32_ble_client::BLECharacteristic::get_descriptor(esphome::esp32_ble_tracker::ESPBTUUID)':
src/esphome/components/esp32_ble_client/ble_characteristic.cpp:53:6: error: expected '(' before 'unlikely'
   if unlikely (!this->parsed)
      ^
src/esphome/components/esp32_ble_client/ble_characteristic.cpp: In member function 'esphome::esp32_ble_client::BLEDescriptor* esphome::esp32_ble_client::BLECharacteristic::get_descriptor_by_handle(uint16_t)':
src/esphome/components/esp32_ble_client/ble_characteristic.cpp:65:6: error: expected '(' before 'unlikely'
   if unlikely (!this->parsed)
      ^
*** [/data/firebeetle-061cb8/.pioenvs/firebeetle-061cb8/src/esphome/components/esp32_ble_client/ble_characteristic.cpp.o] Error 1
src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp: In member function 'virtual void esphome::bluetooth_proxy::BluetoothProxy::loop()':
src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp:115:10: error: expected '(' before 'likely'
       if likely (!service->parsed)
          ^
src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp:122:12: error: expected '(' before 'likely'
         if likely (!characteristic->parsed)
            ^
*** [/data/firebeetle-061cb8/.pioenvs/firebeetle-061cb8/src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp.o] Error 1
```

bdraco, 2022-11-24:
You can't install it as a external component. You need to check out the branch and do a pip install --upgrade .

nprez83, 2022-11-25:
> You can't install it as a external component. You need to check out the branch and do a pip install --upgrade .

I'm sorry, I know this is a newbie question, but I've never installed a package using pip install. Would you please let me know how to do this? I'm running HassOS in a raspberry pi4. Thanks in advance.

sarabveer, 2022-11-25:
> I'm sorry, I know this is a newbie question, but I've never installed a package using pip install. Would you please let me know how to do this? I'm running HassOS in a raspberry pi4. Thanks in [...]

Yea I am in the same situation. I can't find any docs on how to use/build the esphome CLI tool when cloned from the git repo, otherwise I'd gladly help and help debug this.

bdraco, 2022-11-26:
I've updated the branch with this fix as well https://github.com/esphome/esphome/pull/4092

I've tested with 5 different yale / august lock models and everything works now

bdraco, 2022-11-26:
```
git clone --depth 1 https://github.com/bdraco/esphome -b ble_debug_services esphome_yale_locks
cd esphome_yale_locks
pip3 install --upgrade .
esphome compile PATH_TO_YOUR.yaml
esphome upload PATH_TO_YOUR.yaml
```

sarabveer, 2022-11-26:
Just installed the firmware built from your branch, it works (although with a delay).
```
[22:20:04][I][bluetooth_proxy:178]: [0] [<AUGUST_BT_MAC>] Using connect cache
[22:20:04][I][bluetooth_proxy:089]: [0] Used connection by [<AUGUST_BT_MAC>]
[22:20:04][I][bluetooth_proxy:086]: [1] Free connection
[22:20:04][I][bluetooth_proxy:086]: [2] Free connection
[22:20:04][D][esp32_ble_tracker:176]: Pausing scan to make connection...
[22:20:04][I][esp32_ble_client:059]: [0] [<AUGUST_BT_MAC>] 0x00 Attempting BLE connection
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 0x1800
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x1  end_handle: 0x5
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 0x1801
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x6  end_handle: 0x6
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 0x180A
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x7  end_handle: 0xf
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 0xFE24
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x10  end_handle: 0x20
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 0000003E-0000-1000-8000-0026BB765291
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x21  end_handle: 0x39
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 000000A2-0000-1000-8000-0026BB765291
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x3a  end_handle: 0x3f
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 00000055-0000-1000-8000-0026BB765291
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x40  end_handle: 0x4e
[22:20:10][I][esp32_ble_client:164]: [0] [<AUGUST_BT_MAC>] Service UUID: 00000045-0000-1000-8000-0026BB765291
[22:20:10][I][esp32_ble_client:166]: [0] [<AUGUST_BT_MAC>]  start_handle: 0x4f  end_handle: 0x5f
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic 0x2A00, handle 0x3, properties 0x2
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic 0x2A01, handle 0x5, properties 0x2
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic 0x2A29, handle 0x9, properties 0x2
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic 0x2A24, handle 0xb, properties 0x2
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic 0x2A25, handle 0xd, properties 0x2
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic 0x2A26, handle 0xf, properties 0x2
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic BD4AC611-0B45-11E3-8FFD-0800200C9A66, handle 0x12, properties 0xc
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic BD4AC612-0B45-11E3-8FFD-0800200C9A66, handle 0x14, properties 0x20
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic BD4AC613-0B45-11E3-8FFD-0800200C9A66, handle 0x17, properties 0xc
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic BD4AC614-0B45-11E3-8FFD-0800200C9A66, handle 0x19, properties 0x20
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic BD4AC615-0B45-11E3-8FFD-0800200C9A66, handle 0x1c, properties 0x14
[22:20:10][I][esp32_ble_client:062]: [0] [<AUGUST_BT_MAC>]  characteristic BD4AC616-0B45-11E3-8FFD-0800200C9A66, handle 0x1f, properties 0x14
[22:20:10][I][esp32_ble_client:047]: [0] [<AUGUST_BT_MAC>]    descriptor 0x2902, handle 0x1a
[22:20:14][I][esp32_ble_client:047]: [0] [<AUGUST_BT_MAC>]    descriptor 0x2902, handle 0x15
[22:20:15][W][bluetooth_proxy.connection:075]: [0] [<AUGUST_BT_MAC>] Error writing char/descriptor at handle 0x17, status=133
[22:20:15][I][bluetooth_proxy.connection:024]: [0] [<AUGUST_BT_MAC>] Disconnected, freeing slot.
[22:20:15][I][bluetooth_proxy:086]: [0] Free connection
[22:20:15][I][bluetooth_proxy:086]: [1] Free connection
[22:20:15][I][bluetooth_proxy:086]: [2] Free connection
[22:20:15][I][bluetooth_proxy:086]: [0] Free connection
[22:20:15][I][bluetooth_proxy:086]: [1] Free connection
[22:20:15][I][bluetooth_proxy:086]: [2] Free connection
```

I am not sure what this error means, but thats what I can see so far. The August Lock/Unlock works. Haven't checked the door sensor yet.

I assume I need to get an ESP32 with ethernet to speed up the time needed to lock/unlock? Somehow my Apple TV which is no where near the lock can operate the lock via HomeKit with little delay (even when im away).

bdraco, 2022-11-26:
The Apple TV will always be faster. It measures ram in megabytes.  The esp is a 2$ chip that measures ram in bytes.

nprez83, 2022-11-26:
I was able to check out the check out the branch, and to the upgrade, but I'm having trouble compiling my yaml file. Here's the error I get. 

```  
~ esphome compile config/esphome/lilygo-tdongle-s3-2.yaml
INFO Reading configuration config/esphome/lilygo-tdongle-s3-2.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
DEPRECATED! A legacy library storage `/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.piolibdeps` has been found in a project. 
Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.piolibdeps` folder.
More details -> https://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps
Processing lilygo-tdongle-s3-2 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32 @ 5.0.0)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.0.0
INFO Installing platformio/espressif32 @ 5.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Platform Manager: espressif32@5.0.0 has been installed!
INFO espressif32@5.0.0 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20003.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20003.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20003.220626 has been installed!
INFO framework-arduinoespressif32@3.20003.220626 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.30300.0
INFO Installing platformio/tool-esptoolpy @ ~1.30300.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.30300.0 has been installed!
INFO tool-esptoolpy@1.30300.0 has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Improv @ 1.2.3
Compiling /root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_connection.cpp.o
sh: xtensa-esp32s3-elf-g++: not found
Compiling /root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_frame_helper.cpp.o
sh: xtensa-esp32s3-elf-g++: not found
Compiling /root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_pb2.cpp.o
*** [/root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_pb2_service.cpp.o
sh: xtensa-esp32s3-elf-g++: not found
*** [/root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: xtensa-esp32s3-elf-g++: not found
*** [/root/config/esphome/.esphome/build/lilygo-tdongle-s3-2/.pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
===================================================================================================== [FAILED] Took 132.97 seconds ======================
```

Here's my config file. I'm using the lilygo Tdongle S3.

```
esphome:
  name: lilygo-tdongle-s3-2

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0

logger:
api:
ota:
  password: "fe78b1df0729ed126b87d7f9a1f36619"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lilygo-Tdongle-S3-2"
    password: "EZbXX7KWQQqW"

captive_portal:
improv_serial:

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

sarabveer, 2022-11-26:
> The Apple TV will always be faster. It measures ram in megabytes.  The esp is a 2$ chip that measures ram in bytes.
> 
> 

Yea I get that :). It was just my observation.

Have you tested Ethernet vs WiFi and seen a difference with the delay?

bdraco, 2022-11-26:
> > The Apple TV will always be faster. It measures ram in megabytes.  The esp is a 2$ chip that measures ram in bytes.
> > 
> > 
> 
> Yea I get that :). It was just my observation.
> 
> Have you tested Ethernet vs WiFi and seen a difference with the delay?

Ethernet is definitely less latent. The difference is noticeable if you have a good Rssi value

bdraco, 2022-11-26:
If you repull that branch is should speed up connect times a bit as well with the last push

nprez83, 2022-11-26:
Any idea why I'm getting the error when trying to compile the files? Seems related to the api component. I repulled the branch with no luck. I've looked into the .pioenvs/lilygo-tdongle-s3-2/src/esphome/components/api/ folder, and it's empty, so no surprises it can't find those files. Could it be related to my device (lilygo t-dongle s3) or my yaml file?

nprez83, 2022-11-26:
For what is worth, I tried uninstalling the esphome (dev) add on, deleting the esphome folder, and reinstalling addon fresh, replacing my yaml files, and still get the same error as above.

beikeland, 2022-11-27:
Still having issues on my end with the latest pull, is there a preferred setup to test with? I see both esp-idf and arduino used, some have scan_parameters set for esp32_ble_tracker etc.

bdraco, 2022-11-27:
> > The Apple TV will always be faster. It measures ram in megabytes.  The esp is a 2$ chip that measures ram in bytes.
> > 
> > 
> 
> Yea I get that :). It was just my observation.
> 
> Have you tested Ethernet vs WiFi and seen a difference with the delay?

Ethernet is definitely less latent.

bdraco, 2022-11-29:
The changes are merged to dev so no need to pull the branch anymore.  You can flash directly from the default branch.

Home assistant must be updated to nightly build to use the new code path

spartandrew18, 2022-12-01:
I just bought an esp32 to try ESPhome proxy to see if it would allow me to join my lock and I am still stuck with failures.

INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-c9a0b4.yaml...
INFO Starting log output from esp32-bluetooth-proxy-c9a0b4.local using esphome API
INFO Successfully connected to esp32-bluetooth-proxy-c9a0b4.local
[19:25:21][I][app:102]: ESPHome version 2022.11.3 compiled on Nov 30 2022, 18:24:57
[19:25:21][I][app:104]: Project esphome.bluetooth-proxy version 1.0
[19:25:21][C][wifi:504]: WiFi:
[19:25:21][C][wifi:362]:   Local MAC: C0:49:EF:C9:A0:B4
[19:25:21][C][wifi:363]:   SSID: 'Andrew-204'[redacted]
[19:25:22][C][wifi:364]:   IP Address: 192.168.68.219
[19:25:24][C][wifi:366]:   BSSID: 32:87:BA:1C:8D:F4[redacted]
[19:25:24][C][wifi:367]:   Hostname: 'esp32-bluetooth-proxy-c9a0b4'
[19:25:24][C][wifi:369]:   Signal strength: -39 dB ▂▄▆█
[19:25:24][C][wifi:373]:   Channel: 3
[19:25:24][C][wifi:374]:   Subnet: 255.255.255.0
[19:25:24][C][wifi:375]:   Gateway: 192.168.68.1
[19:25:24][C][logger:293]: Logger:
[19:25:24][C][logger:294]:   Level: DEBUG
INFO esp32-bluetooth-proxy-c9a0b4.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO esp32-bluetooth-proxy-c9a0b4.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-bluetooth-proxy-c9a0b4.local
WARNING Disconnected from API
INFO Successfully connected to esp32-bluetooth-proxy-c9a0b4.local
[19:27:21][I][esp32_ble_client:066]: [0] [78:9c:85:21:2b:cc] Disconnecting.
[19:27:23][D][esp32_ble_client:039]: [0] [78:9c:85:21:2b:cc] Found device
[19:27:23][I][esp32_ble_client:054]: [0] [78:9c:85:21:2b:cc] Attempting BLE connection
[19:27:30][W][esp32_ble_client:106]: [0] [] Connection failed, status=133
[19:27:30][D][esp32_ble_tracker:264]: Starting scan...
[19:27:51][I][esp32_ble_client:066]: [0] [78:9c:85:21:2b:cc] Disconnecting.
[19:28:22][D][esp32_ble_client:039]: [0] [78:9c:85:21:2b:cc] Found device
[19:28:22][I][esp32_ble_client:054]: [0] [78:9c:85:21:2b:cc] Attempting BLE connection
[19:28:26][W][esp32_ble_client:106]: [0] [] Connection failed, status=133
[19:28:26][D][esp32_ble_tracker:264]: Starting scan...
[19:28:30][D][esp32_ble_client:039]: [0] [78:9c:85:21:2b:cc] Found device
[19:28:30][I][esp32_ble_client:054]: [0] [78:9c:85:21:2b:cc] Attempting BLE connection
[19:28:45][I][esp32_ble_client:154]: [0] [78:9c:85:21:2b:cc] Service UUID: 0x1800
[19:28:45][I][esp32_ble_client:156]: [0] [78:9c:85:21:2b:cc]  start_handle: 0x1  end_handle: 0x7
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A00, handle 0x3, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A01, handle 0x5, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A04, handle 0x7, properties 0x2
[19:28:45][I][esp32_ble_client:154]: [0] [78:9c:85:21:2b:cc] Service UUID: 0x1801
[19:28:45][I][esp32_ble_client:156]: [0] [78:9c:85:21:2b:cc]  start_handle: 0x8  end_handle: 0xa
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A05, handle 0xa, properties 0x20
[19:28:45][I][esp32_ble_client:154]: [0] [78:9c:85:21:2b:cc] Service UUID: 0x180A
[19:28:45][I][esp32_ble_client:156]: [0] [78:9c:85:21:2b:cc]  start_handle: 0xb  end_handle: 0x17
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A29, handle 0xd, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A24, handle 0xf, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A25, handle 0x11, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A27, handle 0x13, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A26, handle 0x15, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 0x2A28, handle 0x17, properties 0x2
[19:28:45][I][esp32_ble_client:156]: [0] [78:9c:85:21:2b:cc]  start_handle: 0x18  end_handle: 0x28
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic BD4AC612-0B45-11E3-8FFD-0800200C9A66, handle 0x1c, properties 0x20
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic BD4AC613-0B45-11E3-8FFD-0800200C9A66, handle 0x1f, properties 0xc
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic BD4AC614-0B45-11E3-8FFD-0800200C9A66, handle 0x21, properties 0x20
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic BD4AC615-0B45-11E3-8FFD-0800200C9A66, handle 0x24, properties 0x14
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic BD4AC616-0B45-11E3-8FFD-0800200C9A66, handle 0x27, properties 0x14
[19:28:45][I][esp32_ble_client:154]: [0] [78:9c:85:21:2b:cc] Service UUID: 0000003E-0000-1000-8000-0026BB765291
[19:28:45][I][esp32_ble_client:156]: [0] [78:9c:85:21:2b:cc]  start_handle: 0x29  end_handle: 0x47
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x2b, properties 0x2
[19:28:45][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 00000014-0000-1000-8000-0026BB765291, handle 0x2d, properties 0xa
[... cut ...]
[19:31:28][I][esp32_ble_client:156]: [0] [78:9c:85:21:2b:cc]  start_handle: 0x71  end_handle: 0xffff
[19:31:28][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x73, properties 0x2
[19:31:28][I][esp32_ble_client:059]: [0] [78:9c:85:21:2b:cc]  characteristic 000000A5-0000-1000-8000-0026BB765291, handle 0x75, properties 0xa
[19:31:28][W][bluetooth_proxy.connection:074]: [0] [78:9c:85:21:2b:cc] Error writing char/descriptor at handle 0x22, status=128

bdraco, 2022-12-01:
You'll need esphome 2022.12.x and Home Assistant 2022.12.x for it to work but it looks like you found the issue on the esphome side:

https://github.com/esphome/issues/issues/3761

spartandrew18, 2022-12-01:
How do I get HA 2022.12.x? I am on 2022.11.5

bdraco, 2022-12-01:
Its not released until Dec 7th

robloh, 2022-12-01:
The beta just came out, so you could switch to that once it’s available for download

spartandrew18, 2022-12-01:
I just switched to beta channel but its not showing up in the updates section?

robloh, 2022-12-01:
It takes a bit of time to finish building & uploading and becoming available. Give it a few hours. I think you can also upgrade thru the cli, but I’ve never tried that myself

spartandrew18, 2022-12-01:
Ok I will let you know if I get it working.

bdraco, 2022-12-01:
> Ok I finally got it to connect. However the response time is very slow? Is this normal? Like 10-15 seconds or longer to unlock or lock.

Mine take 4-5s to do an op

Did you reflash the esphome device with `dev`?

spartandrew18, 2022-12-01:
[image]
[image]

INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-c9a0b4.yaml...
INFO Configuration is valid!
substitutions:
  name: esp32-bluetooth-proxy-c9a0b4
  wifi_ap_password: ''
esphome:
  name: esp32-bluetooth-proxy-c9a0b4
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: '1.0'
  build_path: .esphome/build/esp32-bluetooth-proxy-c9a0b4
  platformio_options: {}
  includes: []
  libraries: []
  min_version: 2022.12.0
esp32:
  board: esp32dev
  framework:
    version: 2.0.5
    source: ~3.20005.0
    platform_version: platformio/espressif32 @ 5.2.0
    type: arduino
  variant: ESP32
wifi:
  ap:
    password: ''
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  networks:
  - ssid: Andrew-204
    password: Drdrew1805
    priority: 0.0
  use_address: esp32-bluetooth-proxy-c9a0b4.local
api:
  port: 6053
  password: ''
  reboot_timeout: 15min
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
ota:
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10
improv_serial: {}
captive_portal: {}
dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
    duration: 5min
    continuous: true
bluetooth_proxy:
  active: true
  connections:
  - {}
  - {}
  - {}
button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic
  disabled_by_default: false
  icon: mdi:restart-alert
  device_class: restart

spartandrew18, 2022-12-01:
It seems like it has to reconnect to the device every time I click the entity.

bdraco, 2022-12-01:
> It seems like it has to reconnect to the device every time I click the entity.

Yes, thats normal for battery powered devices since otherwise an active connection would chew the battery

Sometimes if I'm lucky mine works in about 3s

[image]

spartandrew18, 2022-12-01:
Darn, then the August Integration uses WiFi and that seems to be alot faster and more consistent. I thought since this was local push it would be faster than the wifi.

spartandrew18, 2022-12-01:
I guess I can try bringing it closer and see what happens.

bdraco, 2022-12-01:
Also if you are using a Wi-Fi connected proxy the air time is shared between ble and Wi-Fi.  All of mine are ethernet

aeozyalcin, 2022-12-03:
For those who think the latency/delay is too long, I have created an August BLE to MQTT bridge that runs on a Raspberry Pi Zero W, and is extremely quick to respond (<2 seconds). It's been running great for the past month on my system.

[aeozyalcin/August2MQTT](https://github.com/aeozyalcin/August2MQTT)

bdraco, 2022-12-03:
> For those who think the latency/delay is too long, I have created an August BLE to MQTT bridge that runs on a Raspberry Pi Zero W, and is extremely quick to respond (<2 seconds). It's been running [...]

Assuming you aren't concerned with the battery trade off and using up a ble connection slot (esphome only supports 3 per device)....If you wanted to add an always connected option to the core integration, I'm sure we could get a PR reviewed.

aeozyalcin, 2022-12-03:
> Assuming you aren't concerned with the battery trade off and using up a ble connection slot (esphome only supports 3 per device)....If you wanted to add an always connected option to the core [...]

I am actually sending raw HCI commands to the Bluetooth chipset to increase the connection latency to 1 second, with a timeout of 3 seconds. Default is something like 50ms, which would be horrible for battery life. I have it detailed in the readme of the repo.

Regarding getting a PR for HA, are there any BLE integrations today that make use of full-time active connections? My understanding was that this was not supported.

bdraco, 2022-12-03:
> I am actually sending raw HCI commands to the Bluetooth chipset to increase the connection latency to 1 second, with a timeout of 3 seconds. Default is something like 50ms, which would be horrible [...]

Led-ble holds the connection indefinitely and auto reconnects.

> My understanding was that this was not supported. 

It's definitely supported but probably not a good on by default option since connection slots are limited

bdraco, 2022-12-03:
Also SwitchBot holds the connection for a rolling 8.5s after the last command to wait for more commands before disconnecting

bdraco, 2022-12-03:
Also you might want to check dev/2022.12.x esphome as well since the experience is much better.

My unlock times are 3-5s with good signal and having to reconnect every time. 

It's never going to be as fast as 2s though since the connection establish takes 800-1200ms and the operation itself takes about 1600ms

aeozyalcin, 2022-12-03:
> Also SwitchBot holds the connection for a rolling 8.5s after the last command to wait for more commands before disconnecting

How are the ESP32's holding a BLE connection when they need to switch over to WiFi to talk with HA API? Or is full-time connection only supported on ESP32's with ethernet connectors?

bdraco, 2022-12-03:
> How are the ESP32's holding a BLE connection when they need to switch over to WiFi to talk with HA API? Or is full-time connection only supported on ESP32's with ethernet connectors?

It's interleaved

spartandrew18, 2022-12-03:
> For those who think the latency/delay is too long, I have created an August BLE to MQTT bridge that runs on a Raspberry Pi Zero W, and is extremely quick to respond (<2 seconds). It's been running [...]

Ill give it a try but I am still very confused on how to do your instructions? I create a config.json file in file editor on home assistant and edit the values in a note editor for the august_bridge.py? How exactly do I run this thing. I cannot access root in home assistant to download those packages for setup.sh.

aeozyalcin, 2022-12-03:
> Ill give it a try but I am still very confused on how to do your instructions? I create a config.json file in file editor on home assistant and edit the values in a note editor for the [...]

Awesome! Are you planning on running this on the same machine that also runs home assistant, or you have a separate machine? Just clone the repo, and create a config.json file in the same directory as `mqtt_august_bridge.py`, and follow the linked instructions for populating that file. 

I don't want to hijack this thread, do you want to create a quick issue on the repo, and I can walk you through it there?

ampersandru, 2022-12-08:
has the [web flasher](https://esphome.github.io/bluetooth-proxies) been updated to the version with the fix? So once its flashed with the latest version and you update Home Assistant to 2022.12, it should work better now?

bdraco, 2022-12-08:
It was updated this morning to the latest beta, but there is another fix coming so I'd wait until tomorrow

liquidbear99, 2022-12-09:
Looks like there still is an issue but the integration does find the lock. This is the error I get. Let me know if the full log would be helpful:

INFO esp32-bluetooth-proxy-0d340c.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO esp32-bluetooth-proxy-0d340c.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-bluetooth-proxy-0d340c.local
WARNING Disconnected from API

bdraco, 2022-12-09:
That's the connection dropping between hass and the esphome device not the Bluetooth.

Probably will need full logs on both sides to figure out why

gsuh, 2022-12-11:
Firmware 2022.12.0b2 fixed this issue for me

ampersandru, 2022-12-13:
just flashed to 2022.12.0b5 today and updated HA to 2022.12.4 and conncted to my august pro 3rd gen locks without issue. Delay doesn't seem that bad (around 5ish seconds). I have the August locks already connected via Zwave and Wifi, is having it connected via BLE going to cause significant battery drain? Zwave seems pretty damn spotty on it, so wanted another local way of controlling

samwiseg0, 2023-03-31:
I seem to be having an issue with esphome bluetooth proxy where the connection errors out after being connected successfully. As far as I can tell all the fixes should have been merged in? Is there something I am missing?

Logs from esphome:

```
[15:01:24][V][bluetooth_proxy.connection:206]: [0] [78:9C:85:33:55:5A] Writing GATT characteristic handle 31
[15:01:24][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 4
[15:01:24][V][esp32_ble_client:114]: [0] [78:9C:85:33:55:5A] gattc_event_handler: event=4 gattc_if=4
[15:01:24][W][bluetooth_proxy.connection:101]: [0] [78:9C:85:33:55:5A] Error writing char/descriptor at handle 0x1F, status=133
[15:01:24][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 41
[15:01:24][V][esp32_ble_client:114]: [0] [78:9C:85:33:55:5A] gattc_event_handler: event=41 gattc_if=4
[15:01:24][V][esp32_ble_client:168]: [0] [78:9C:85:33:55:5A] ESP_GATTC_DISCONNECT_EVT, reason 19
[15:01:24][V][bluetooth_proxy:074]: [0] Free connection
[15:01:24][V][bluetooth_proxy:074]: [1] Free connection
[15:01:24][V][bluetooth_proxy:074]: [2] Free connection
```

itpeters, 2023-05-06:
@samwiseg0 did you ever figure out anything more? I'm seeing similar behavior from my bluetooth proxy talking to my yale lock, with similar errors:

```
[23:56:00][I][bluetooth_proxy:250]: [0] [78:9C:85:1A:74:F3] Connecting v3 with cache
[23:56:00][D][esp32_ble_tracker:214]: Pausing scan to make connection...
[23:56:00][I][esp32_ble_client:064]: [0] [78:9C:85:1A:74:F3] 0x00 Attempting BLE connection
[23:56:01][D][esp-idf:000]: W (91506) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[23:56:02][I][esp32_ble_client:144]: [0] [78:9C:85:1A:74:F3] Connected
[23:56:02][D][esp32_ble_tracker:246]: Starting scan...
[23:56:25][D][esp-idf:000]: W (116042) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13

[23:56:25][D][esp-idf:000]: W (116049) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x13

[23:56:25][D][esp-idf:000]: W (116051) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13

[23:56:25][D][esp-idf:000]: W (116054) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x13

[23:56:25][W][bluetooth_proxy.connection:101]: [0] [78:9C:85:1A:74:F3] Error writing char/descriptor at handle 0x17, status=133
```

lloydhazlett, 2023-05-07:
I'm seeing exactly the same errors as @itpeters with ESPHome version 2023.4.2 and HA 2023.4.6. Have tried making scan window 1000ms and interval 2000ms but made no difference compared to the defaults.

[8 more replies]

closed 2023-11-13 by bdraco