esphome/issues #3762: SI7021 sensor not working for Sonoff TH Elite

kristjanbjarni, 2022-11-04:
### The problem

I am unable to get the THS01 (SI7021) temparature/humidity sensor working on the Sonoff TH Elite (THR320D) on the ESP32 platform.

### Which version of ESPHome has the issue?

 2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.1

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

dht

### Example YAML snippet

```yaml
substitutions:
  devicename: thr3_1
  
esphome:
  name: ${devicename}
  comment: Greenhouse temperature

esp32:
  board: nodemcu-32s

wifi:
  reboot_timeout: 0s
  networks:
  - ssid: "Skynet"
    password: !secret wifi
  - ssid: "Skynet_3H"
    password: !secret wifi
  ap:
    ssid: "${devicename}_fallback"
    password: !secret wifi

captive_portal:

logger:
  baud_rate: 0

api:
  reboot_timeout: 0s

ota:

sensor:
  - platform: dht
    model: SI7021
    pin: GPIO25
    temperature:
      name: "${devicename} Temperature"
      id: temperature
    humidity:
      name: "${devicename} Humidity"
      id: humidity

binary_sensor:
  - platform: gpio
    id: push_button
    pin:
      number: GPIO00
      inverted: True
    internal: true
    on_press:
      - switch.toggle: button

switch:
  - platform: gpio
    pin: GPIO19
    id: relay_reset

  - platform: gpio
    pin: GPIO22
    id: relay_set

  - platform: gpio
    id: power_led
    name: "${devicename} Power Led"
    pin:
      number: GPIO16
      inverted: True

  - platform: gpio
    id: connected_led
    name: "${devicename} Connected Led"
    pin:
      number: GPIO15
      inverted: True

  - platform: gpio
    id: auto_led
    name: "${devicename} Auto Led"
    pin:
      number: GPIO13
      inverted: True

  - platform: template
    name: "${devicename} relay"
    id: relay
    lambda: |-
      return id(power_led).state;
    turn_on_action:
      - switch.turn_on: relay_set
      - switch.turn_on: power_led
      - delay: 0.1s
      - switch.turn_off: relay_set
    turn_off_action:
      - switch.turn_on: relay_reset
      - switch.turn_off: power_led
      - delay: 0.1s
      - switch.turn_off: relay_reset

  - platform: template
    name: "${devicename} Automation"
    id: automation
    lambda: |-
      return id(power_led).state and id(auto_led).state;
    turn_on_action:
      - switch.turn_on: relay
      - switch.turn_on: auto_led
    turn_off_action:
      - switch.turn_off: auto_led
      - if:
          condition:
            - switch.is_off: connected_led
          then:
          - switch.turn_off: relay

  - platform: template
    name: "${devicename} Button"
    id: button
    lambda: |-
      return id(power_led).state;
    turn_on_action:
      - switch.turn_on: relay
      - switch.turn_on: connected_led
      - switch.turn_off: auto_led
    turn_off_action:
      - switch.turn_off: relay
      - switch.turn_off: connected_led
      - switch.turn_off: auto_led

display:
  platform: tm1621
  id: tm1621_display
  update_interval: 1s
  cs_pin: GPIO17
  data_pin: GPIO05
  read_pin: GPIO23
  write_pin: GPIO18
  lambda: |-
    it.display_celsius(true);
    it.display_humidity(true);
    it.printf(0, "%.1f", id(temperature).state);
    it.printf(1, "%.1f", id(humidity).state);
```

### Anything in the logs that might be useful for us?

```txt
[13:32:44][W][dht:163]: Waiting for DHT communication to clear failed!
[13:32:44][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number). [13:32:44][D][sensor:127]: 'thr3_1 Temperature': Sending state nan °C with 1 decimals of accuracy [13:32:44][D][sensor:127]: 'thr3_1 Humidity': Sending state nan % with 0 decimals of accuracy
```

kristjanbjarni, 2022-11-04:
Never mind, found out that you need to explicitly set GPIO27 as 3v out:

output:
  - platform: gpio
    pin: GPIO27
    id: sensor_3v_output
    inverted: true

Then everything works.

closed 2022-11-04 by kristjanbjarni

erdembey, 2023-02-24:
> Never mind, found out that you need to explicitly set GPIO27 as 3v out:
> 
> ```yaml
> output:
>   - platform: gpio
>     pin: GPIO27
>     id: sensor_3v_output
>     inverted: true
> ```
> 
> Then everything works.

Can you provide an updated yaml file ?

kristjanbjarni, 2023-02-25:
> Can you provide an updated yaml file ?

Sure. This is what I am currently using for my newly built greenhouse. Don't have the heater yet so no temperature automation is currently specified, but the manual button should work.

``` yaml
substitutions:
  devicename: thr3_1
  
esphome:
  name: ${devicename}
  comment: Greenhouse temperature

esp32:
  board: nodemcu-32s

wifi:
  reboot_timeout: 0s
  networks:
  - ssid: "Skynet"
    password: !secret wifi
  - ssid: "Skynet_3H"
    password: !secret wifi
  ap:
    ssid: "${devicename}_fallback"
    password: !secret wifi

captive_portal:

logger:
  baud_rate: 0

api:
  reboot_timeout: 0s

ota:

output:
  - platform: gpio
    pin: GPIO27
    id: sensor_3v_output
    inverted: true

sensor:
  - platform: dht
    model: SI7021
    pin: GPIO25
    temperature:
      name: "${devicename} Temperature"
      id: temperature
    humidity:
      name: "${devicename} Humidity"
      id: humidity

binary_sensor:
  - platform: gpio
    id: push_button
    pin:
      number: GPIO00
      inverted: True
    internal: true
    on_press:
      - switch.toggle: button

switch:
  - platform: gpio
    pin: GPIO19
    id: relay_reset

  - platform: gpio
    pin: GPIO22
    id: relay_set

  - platform: gpio
    id: power_led
    name: "${devicename} Power Led"
    pin:
      number: GPIO16
      inverted: True

  - platform: gpio
    id: connected_led
    name: "${devicename} Connected Led"
    pin:
      number: GPIO15
      inverted: True

  - platform: gpio
    id: auto_led
    name: "${devicename} Auto Led"
    pin:
      number: GPIO13
      inverted: True

  - platform: template
    name: "${devicename} relay"
    id: relay
    lambda: |-
      return id(power_led).state;
    turn_on_action:
      - switch.turn_on: relay_set
      - switch.turn_on: power_led
      - delay: 0.1s
      - switch.turn_off: relay_set
    turn_off_action:
      - switch.turn_on: relay_reset
      - switch.turn_off: power_led
      - delay: 0.1s
      - switch.turn_off: relay_reset

  - platform: template
    name: "${devicename} Automation"
    id: automation
    lambda: |-
      return id(power_led).state and id(auto_led).state;
    turn_on_action:
      - switch.turn_on: relay
      - switch.turn_on: auto_led
    turn_off_action:
      - switch.turn_off: auto_led
      - if:
          condition:
            - switch.is_off: connected_led
          then:
          - switch.turn_off: relay

  - platform: template
    name: "${devicename} Button"
    id: button
    lambda: |-
      return id(power_led).state;
    turn_on_action:
      - switch.turn_on: relay
      - switch.turn_on: connected_led
      - switch.turn_off: auto_led
    turn_off_action:
      - switch.turn_off: relay
      - switch.turn_off: connected_led
      - switch.turn_off: auto_led

display:
  platform: tm1621
  id: tm1621_display
  update_interval: 1s
  cs_pin: GPIO17
  data_pin: GPIO05
  read_pin: GPIO23
  write_pin: GPIO18
  lambda: |-
    it.display_celsius(true);
    it.display_humidity(true);      
    it.printf(0, "%.1f", id(temperature).state);
    it.printf(1, "%.1f", id(humidity).state);

```

erdembey, 2023-02-25:
Thank you very much. I've reinstalled with updated yaml below, powercycle but still recieving ;
Requesting data from DHT failed!
Invalid readings! Please check your wiring (pull-up resistor, pin number).

```
esp32:
  board: nodemcu-32s

logger:

output:
  - platform: gpio
    pin: GPIO27
    id: sensor_3v_output
    inverted: true

sensor:
  - platform: dht
    pin: GPIO25
    model: SI7021
    temperature:
      name: "${friendly_name} Temp"
      id: "${name}_temp"
    humidity:
      name: "${friendly_name} Humi"
      id: "${name}_humi"
    update_interval: 60s
```

kristjanbjarni, 2023-02-25:
> Thank you very much @kristjanbjarni . I've reinstalled with updated yaml below, powercycle but still recieving ; Requesting data from DHT failed! Invalid readings! Please check your wiring (pull-up resistor, pin number).

@erdembey Not sure why it doesn't work. This config currently works for me. Are you sure you are using the THS01 sensor? You could try changing:

``` yaml
model: AUTO_DETECT
```
if you are using some other type of sensor. Although I think using `model: SI7021` is required for the THS01 sensor to work.

However if you have the waterproof sensor DS18B20 with only temperature you would need instead a config perhaps something like this:

``` yaml
dallas:
  - pin: GPIO25

sensor:
  - platform: dallas
    index: 0
    id: temperature
    name: "${devicename} Temperature"
    filters:
      - filter_out: nan
```

I used this very helpful page to get all the correct pin ids:
[Sonoff TH Elite 20A Temperature and Humidity Monitoring Switch Module (THR320D)](https://templates.blakadder.com/sonoff_THR320D.html)

erdembey, 2023-02-25:
@kristjanbjarni i've added auto_detect and it shows;
Auto-detected model: DHT11
Strange i've 3 THS01 only one of them seems to be working. I will be recompile with DHT11 and retry.
Thanks

kristjanbjarni, 2023-03-28:
@erdembey 

I believe there is an error in my config for the relay, the io pins where reversed but just happened to work because of inversion, but it's not a good setup. I believe it should be like this instead:

``` yaml
switch:
  - platform: gpio
    id: relay_set
    pin:
      number: GPIO19
      inverted: true

  - platform: gpio
    id: relay_reset
    pin:
      number: GPIO22
      inverted: true
```

Here is a link to a better config from someone else for this device:
https://www.esphome-devices.com/devices/Sonoff-THR320D