esphome/issues #3763: Extreme puls_meter readings and swings

emeyedeejay, 2022-11-04:
### The problem

An existing pulse_meter setup has been running well for months but recently started behaving very weirdly. The readins will swing from e.g. 1000W to like 36 000 000W and back again.

I cannot recall making any changes to the setup but is it possible a recent update broke something?

My config:

```
sensor:    # Pulse meter
  - platform: pulse_meter
    name: 'Current Power Consumption'
    id: sensor_current_power_consumption
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: D2
    internal_filter_mode: pulse
    filters:
  # multiply value = (60 / imp value) * 1000
  # - multiply: 60
    - lambda: return x * ((60.0 / 1000) * 1000.0);
```

[image]

Looks like it went nuts on 24 October - doe that match any release date maybe?

[image]

Some log output:

```
INFO Reading configuration /config/esphome/electricity-usage-monitor.yaml...
INFO Detected timezone 'Africa/Johannesburg'
INFO Starting log output from electricity-usage-monitor.local using esphome API
INFO Successfully connected to electricity-usage-monitor.local
[22:42:52][I][app:102]: ESPHome version 2022.10.1 compiled on Oct 26 2022, 12:16:53
[22:42:52][C][wifi:502]: WiFi:
[22:42:52][C][wifi:360]:   Local MAC: B4:E6:2D:13:F0:A9
[22:42:52][C][wifi:361]:   SSID: 'JNET'[redacted]
[22:42:52][C][wifi:362]:   IP Address: 192.168.0.120
[22:42:52][C][wifi:363]:   BSSID: 78:8A:20:F7:3B:46[redacted]
[22:42:52][C][wifi:365]:   Hostname: 'electricity-usage-monitor'
[22:42:52][C][wifi:367]:   Signal strength: -71 dB ▂▄▆█
[22:42:52][C][wifi:371]:   Channel: 6
[22:42:52][C][wifi:372]:   Subnet: 255.255.255.0
[22:42:52][C][wifi:373]:   Gateway: 192.168.0.1
[22:42:52][C][wifi:374]:   DNS1: 192.168.0.1
[22:42:52][C][wifi:375]:   DNS2: 0.0.0.0
[22:42:52][C][logger:275]: Logger:
[22:42:52][C][logger:276]:   Level: DEBUG
[22:42:52][C][logger:277]:   Log Baud Rate: 115200
[22:42:52][C][logger:278]:   Hardware UART: UART0
[22:42:52][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Home power monitor'
[22:42:52][C][gpio.binary_sensor:016]:   Pin: GPIO4
[22:42:52][C][pulse_meter:069]: Pulse Meter 'Current Power Consumption'
[22:42:52][C][pulse_meter:069]:   Device Class: 'power'
[22:42:52][C][pulse_meter:069]:   State Class: 'measurement'
[22:42:52][C][pulse_meter:069]:   Unit of Measurement: 'W'
[22:42:52][C][pulse_meter:069]:   Accuracy Decimals: 0
[22:42:52][C][pulse_meter:069]:   Icon: 'mdi:flash-outline'
[22:42:52][C][pulse_meter:070]:   Pin: GPIO4
[22:42:52][C][pulse_meter:074]:   Filtering pulses shorter than 13 µs
[22:42:52][C][pulse_meter:076]:   Assuming 0 pulses/min after not receiving a pulse for 300s
[22:42:52][C][total_daily_energy:023]: Total Daily Energy 'Home daily power'
[22:42:52][C][total_daily_energy:023]:   Device Class: 'energy'
[22:42:52][C][total_daily_energy:023]:   State Class: 'total_increasing'
[... 181 lines cut ...]
[22:43:12][D][sensor:126]: 'Home total power': Sending state 128.34399 kWh with 3 decimals of accuracy
[22:43:12][D][binary_sensor:036]: 'Home power monitor': Sending state ON
[22:43:14][D][sensor:126]: 'Home total power': Sending state 128.34599 kWh with 3 decimals of accuracy
[22:43:14][D][binary_sensor:036]: 'Home power monitor': Sending state OFF
[22:43:17][D][sensor:126]: 'Current Power Consumption': Sending state 1401.86914 W with 0 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home daily power': Sending state 1732.92761 kWh with 3 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home total power': Sending state 128.34801 kWh with 3 decimals of accuracy
[22:43:17][D][sensor:126]: 'Current Power Consumption': Sending state 240000.00000 W with 0 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home daily power': Sending state 1732.92993 kWh with 3 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home total power': Sending state 128.34900 kWh with 3 decimals of accuracy
[22:43:17][D][binary_sensor:036]: 'Home power monitor': Sending state ON
[22:43:19][D][binary_sensor:036]: 'Home power monitor': Sending state OFF
[22:43:22][D][sensor:126]: 'Current Power Consumption': Sending state 257142.85938 W with 0 decimals of accuracy
[22:43:22][D][sensor:126]: 'Home daily power': Sending state 1733.29260 kWh with 3 decimals of accuracy
[22:43:22][D][sensor:126]: 'Home total power': Sending state 128.35500 kWh with 3 decimals of accuracy
[22:43:22][D][sensor:126]: 'Current Power Consumption': Sending state 3600000.00000 W with 0 decimals of accuracy
```

### Additional information

[image]

Looks like it went nuts on 24 October - doe that match any release date maybe?

[image]

ssieb, 2022-11-06:
Duplicate of https://github.com/esphome/issues/issues/3730

closed 2022-11-06 by ssieb

emeyedeejay, 2022-11-07:
> Duplicate of #3730

Thank you and my bad ... I should have searched. My apologies.