esphome/issues #3764: Compiling Error

Sjeff, 2022-11-04:
### The problem

I'm trying to get the shelly 1 plus to act as my garage door opener but something in the Cover template isn't right.

can someone point me in the right direction?

when i remove te cover completely i dont have any compiling errors

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP32

### Board

Shelly 1 plus

### Component causing the issue

cover template

### Example YAML snippet

```yaml
substitutions:
  device_name: "Shelly Plus 1 Garage"
  # Higher value gives lower watt readout
  current_res: "0.001"
  # Lower value gives lower voltage readout
  voltage_div: "1925"

esphome:
  name: shelly-plus-1-garage
  platformio_options:
    board_build.f_cpu: 160000000L

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"

wifi:
  ssid: !secret wifi_http_ssid
  password: !secret wifi_http_password
  fast_connect: true
  use_address: 192.168.5.186

logger:

api:
  password: !secret esphome_api_password
  encryption:
    key: !secret esphome_api_encryption_key

ota:
[... 135 lines cut ...]
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o:(.literal._Z5setupv+0x1ec): undefined reference to `esphome::template_::TemplateCover::get_close_trigger() const'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o:(.literal._Z5setupv+0x1f0): undefined reference to `esphome::template_::TemplateCover::set_has_position(bool)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o:(.literal._Z5setupv+0x1f4): undefined reference to `esphome::template_::TemplateCover::set_optimistic(bool)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o:(.literal._Z5setupv+0x1f8): undefined reference to `esphome::template_::TemplateCover::set_assumed_state(bool)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o: in function `esphome::binary_sensor::Filter::Filter()':
/root/config/.esphome/build/shelly-plus-1-garage/src/esphome/components/binary_sensor/filter.h:12: undefined reference to `esphome::template_::TemplateCover::TemplateCover()'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o: in function `__gnu_cxx::new_allocator<esphome::binary_sensor::Filter*>::allocate(unsigned int, void const*)':
/root/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/ext/new_allocator.h:111: undefined reference to `esphome::cover::Cover::set_device_class(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o: in function `esphome::binary_sensor::Filter** std::__copy_move<false, true, std::random_access_iterator_tag>::__copy_m<esphome::binary_sensor::Filter*>(esphome::binary_sensor::Filter* const*, esphome::binary_sensor::Filter* const*, esphome::binary_sensor::Filter**)':
/root/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_algobase.h:368: undefined reference to `esphome::template_::TemplateCover::set_state_lambda(std::function<esphome::optional<float> ()>&&)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o: in function `setup()':
/root/config/.esphome/build/shelly-plus-1-garage/src/main.cpp:256: undefined reference to `esphome::template_::TemplateCover::get_open_trigger() const'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /root/config/.esphome/build/shelly-plus-1-garage/src/main.cpp:311: undefined reference to `esphome::template_::TemplateCover::get_close_trigger() const'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/main.o: in function `setup()':
/root/config/shelly-plus-1-garage.yaml:78: undefined reference to `esphome::template_::TemplateCover::set_has_position(bool)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /root/config/shelly-plus-1-garage.yaml:78: undefined reference to `esphome::template_::TemplateCover::set_optimistic(bool)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /root/config/shelly-plus-1-garage.yaml:78: undefined reference to `esphome::template_::TemplateCover::set_assumed_state(bool)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /root/config/shelly-plus-1-garage.yaml:78: undefined reference to `esphome::template_::TemplateCover::set_has_position(bool)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/esphome/core/controller.o:(.literal._ZN7esphome10Controller16setup_controllerEb+0x20): undefined reference to `esphome::cover::Cover::add_on_state_callback(std::function<void ()>&&)'
/root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/shelly-plus-1-garage/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/root/config/.esphome/build/shelly-plus-1-garage/src/esphome/core/controller.cpp:41: undefined reference to `esphome::cover::Cover::add_on_state_callback(std::function<void ()>&&)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/shelly-plus-1-garage/firmware.elf] Error 1
========================= [FAILED] Took 22.65 seconds =========================
```

ssieb, 2022-11-04:
You're using idf.  Try doing a clean build.

Sjeff, 2022-11-05:
thanks! got it working after a clean build!

closed 2022-11-05 by Sjeff