esphome/issues #3767: SM300D2 Sensor - Invalid Preamble - Reading Data Time out

EdDickens, 2022-11-05:
### The problem

I have a new SM300D2 sensor connected to an ESP01. I’ve got the ESP flashed with a basic config to get things running. All the Entities show up in HA with no problems. The problem is actually reading from the sensor. Seem to be having the same issues [reported here](https://community.home-assistant.io/t/esphome-sdm300d2-issue/324891). From that thread it seems the issue was fixed at one point but I cannot get anything from the sensor.

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

SM300D2

### Example YAML snippet

```yaml
esphome:
  name: air-quality

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0
  
# Enable Home Assistant API
api:
#  password: !secret haapi_password

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi1_ssid
  password: !secret wifi1_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Air-Quality Fallback Hotspot"
    password: !secret fb_password

captive_portal:

switch:
  - platform: restart
    id: AirQual_restart
    name: "Air Quality Restart"
    
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

sensor:
  - platform: sm300d2
    co2:
      name: "CO2"
      id: "AirQual_CO2"
    formaldehyde:
      name: "Formaldehyde"
      id: "AirQual_Formaldehyde"
    tvoc:
      name: "TVOC"
      id: "AirQual_TVOC"
    pm_2_5:
      name: "PM2.5"
      id: "AirQual_PM25"
    pm_10_0:
      name: "PM10"
      id: "AirQual_PM10"
    temperature:
      name: "Temperature"
      id: "AirQual_Temperature"
    humidity:
      name: "Humidity"
      id: "AirQual_Humidity"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[13:43:49][W][sm300d2:026]: Invalid preamble for SM300D2 response!
[13:44:50][E][uart:015]: Reading from UART timed out at byte 0!
[13:44:50][W][sm300d2:020]: Reading data from SM300D2 failed!
[13:45:50][E][uart:015]: Reading from UART timed out at byte 0!
[13:45:50][W][sm300d2:020]: Reading data from SM300D2 failed!
[13:46:50][E][uart:015]: Reading from UART timed out at byte 0!
[13:46:50][W][sm300d2:020]: Reading data from SM300D2 failed!
[13:47:50][W][sm300d2:026]: Invalid preamble for SM300D2 response!
[13:48:50][E][uart:015]: Reading from UART timed out at byte 0!
[13:48:50][W][sm300d2:020]: Reading data from SM300D2 failed!
```

Duker4444, 2022-12-29:
I'm seeing the same thing. using a D1 mini on ESPHOME 2022.3.1

Serial Data is incomming however it's not being processed correctly.. 
None of the datasamples are picked up.. 

Sample of that debug output from the UART. 

[19:58:39][D][uart_debug:158]: <<< "\x01\x03\x0E\x01\xD9\x00\f\x00/\x00\x11\x00\x1F\b\xDC\n"
[19:58:39][D][uart_debug:158]: <<< "\x1A\xFA\xDF\x01\x03\x0E\x01\xD9\x00\f\x00/\x00\x10\x00\x1D\b\xD6\n"
[19:58:39][D][uart_debug:158]: <<< "\x1D\xF2\x1F\x01\x03\x0E\x01\xD9\x00\f\x00/\x00\x10\x00\x1D\b\xD9\n"
[19:58:39][D][uart_debug:158]: <<< "\x1E\x82\x1D\x01\x03\x0E\x01\xD9\x00\f\x00/\x00\x0F\x00\e\b\xD9\n"
[19:58:39][D][uart_debug:158]: <<< "\x1D\xA4\xDD\x01\x03\x0E\x01\xD9\x00\f\x00/\x00\x0F\x00\e\b\xD8\n"
[19:58:39][D][uart_debug:158]: <<< "\x1E\xB5\x1C\x01\x03\x0E\x01\xDB\x00\r\x000\x00\x0E\x00\x19\b\xD8\n"
[19:58:39][D][uart_debug:158]: <<< " \xD2B\x01\x03\x0E\x01\xDB\x00\r\x000\x00\x0E\x00\x19\b\xD9\n"
[19:58:39][D][uart_debug:158]: <<< "\x1E\x02R\x01\x03\x0E\x01\xDB\x00\r\x000\x00\r\x00\x18\b\xD8\n"
[19:58:39][D][uart_debug:158]: <<< "\x1D\x1DS\x01\x03\x0E\x01\xDB\x00\r\x000\x00\x0E\x00\x19\b\xD8\n"
[19:58:39][D][uart_debug:158]: <<< "\x1ES\x92\x01\x03\x0E\x01\xD9\x00\f\x00/\x00\r\x00\x18\b\xD9\n"
[19:58:39][D][uart_debug:158]: <<< "\x1D\xC3\x1D\x01\x03\x0E\x01\xDB\x00\r\x000\x00\r\x00\x18\b\xD8\n"
[19:58:39][D][uart_debug:158]: <<< "!\x1DB\x01\x03\x0E\x01\xDB\x00\r\x000\x00\x0E\x00\x19\b\xD6\n"
[19:58:39][D][uart_debug:158]: <<< "!rA\x01\x03\x0E\x01\xDB\x00\r\x000\x00\x0E\x00\x19\b\xD8\n"
[19:58:39][D][uart_debug:158]: <<< "\"S\x83\x01\x03\x0E\x01\xDB\x00\r\x000\x00\x0F\x00\e\b\xD8\n"
[19:58:39][D][uart_debug:158]: <<< "$\xBA\x81\x01\x03\x0E\x01\xDB\x00\r\x000\x00\x0F\x00\e\b\xD6\n"
[19:58:39][E][uart:015]: Reading from UART timed out at byte 0!
[19:58:39][W][sm300d2:020]: Reading data from SM300D2 failed!
[19:58:39][D][uart_debug:158]: <<< "%\x1A\x82"

Duker4444, 2023-01-03:
I can provide much more logs if that helps. . I've been looking into the data stream and it appears that the strings are not deteched.. just a constant stream of data now. 

I've got two of these units and they both output the same information. 
I'll be checking this post on a regular basis.   Forward thanks to anyone working on this..

redyuan43, 2023-05-03:
same issue.

Duker4444, 2023-05-21:
upgraded to 2023.5.0 and data is still the same. .

Most recent Dump
01:03:0E:02:6B:00:26:00:8A:00:08:00:0F:09:B5:0E:65:D6:22
01:03:0E:02:6C:00:26:00:8B:00:08:00:0F:09:B5:0E:65:D0:F5
01:03:0E:02:6E:00:26:00:8C:00:07:00:0D:09:B8:0E:67:67:45
01:03:0E:02:6F:00:26:00:8D:00:07:00:0D:09:B8:0E:68:28:50
01:03:0E:02:75:00:28:00:90:00:07:00:0D:09:B6:0E:64:CE:B4
01:03:0E:02:7B:00:29:00:94:00:07:00:0D:09:B9:0E:6C:DF:C3
01:03:0E:02:7E:00:29:00:96:00:06:00:0B:09:B6:0E:67:23:A2
Hoping someone may figure it out. 
Tried on both ESP32 and the D1 mini same outputs

Duker4444, 2023-05-22:
Alright my friend, I finally figured out what's going on. 

There are a few code changes that has to happen. 
you can see the modifications on my fork at https://github.com/Duker4444
in file sm300d2.cpp 

But this will only work for my sensors.. 
my example Serial output: 01:03:0E:02:7E:00:29:00:96:00:06:00:0B:09:B6:0E:67:23:A2
in mine the code looks for 01 and 03 to know where the beginning of the string to read from. 

[EdDickens] - You would have to modify yours to reference off of 7f and 7E  based on the strings you provided. 

 if (response[0] != 0x7ff || response[1] != 0x7e) {
    ESP_LOGW(TAG, "Invalid preamble for SM300D2 response!");
    this->status_set_warning();
    return;
}
in the sm300d2.cpp file of your own fork.

I also had to remove the checksum function since it's not really necessary.  It seems like they changed their method of validation since now, Mine have 19 placements in the string vs the normal 17. and the last few don't make sense. 

Unfortunately it appears that there are multiple string headers and no longer standardization. 
And we're missing documentation from the vendor as to what and why they changed. 

I'm still validating the changes I've made, But i'm now getting data that appears to be alot closer to the correct values. 
I only have two sensors on hand to compare against.

Duker4444, 2023-05-22:
I think I may have stumbled on the new sequence. 

Byte    name            Description

B1    Slave address   Address value

B2   function code  03h

B3  Return the number of bytes   0x0E

B4   data     eCO2 High byte

B5   data     eCO2 Low byte

B6   data   eCH2O High byte

B7   data  eCH2O Low byte

B8   data   TVOC High byte

B9  data   TVOC Low byte

B10 data  PM2.5 High byte

B11 data PM2.5 Low byte

B12  data PM10 High byte

B13 data  PM10 Low byte

B14 data  Temperature "High" byte

B15  data Temperature "Low" byte

B16 data Humidity "High" byte

B17 data Humidity "Low" byte

B18 CRC 16 low

B19 CRC16 high

Working to validate these since my readings seem a little high.

Duker4444, 2023-06-20:
Ive got the values worked out. Should see a new component her soon called sm300d2v2 since this is a v2 hardware variant.

rotilho, 2023-06-30:
Hey Duker! I see that you closed your PR, did anything change?

Duker4444, 2023-08-04:
I've got it working. however i'm stalled on getting it approved for commit. 
The ESPhome owner wanted to see if I could somehow merge the two devices instead of creating a 2nd hardware type. 
Unfortunately, My skills aren't quite that high. I've made a few attempts, But due to how the Serial is parsed. I haven't been able to make it universal and merge with the original code.