esphome/issues #3769: Wrong timezone ESP32

Salvora, 2022-11-06:
### The problem

I am not sure what changed but suddenly all my ESP32 devices report time wrong with -6 hours offset.
On the other hand my ESP8266 device reports the correct time.

I tested this using homeassistant time and sntp time.
Below are the configurations I have tested.

It reports the time as **UTC/GMT -3** hours instead it should be **UTC/GMT +3** hours.

```
time:
  - platform: homeassistant
    id: homeassistant_time
```

```
time:
  - platform: sntp
    servers: 192.168.1.1
    id: ntp_time
```

```
time:
  - platform: sntp
    id: ntp_time
```
```
time:
  - platform: sntp
    id: ntp_time
    timezone: Europe/Istanbul
```

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP32

### Board

devkit-v1

### Component causing the issue

time

### Example YAML snippet

```yaml
time:
  - platform: sntp
    servers: 192.168.1.1
    id: ntp_time
```

ssieb, 2022-11-06:
Does the last one with the timezone work?  What do you see in the logs?

Salvora, 2022-11-06:
I believe 2022.10.1 was working. Log is attached but in short, it correctly sees the time:
```

INFO Detected timezone 'Europe/Istanbul'
INFO Generating C++ source...
INFO Compiling app...
```

and sntp:

```
[17:58:17][C][sntp:050]: SNTP Time:
[17:58:17][C][sntp:051]:   Server 1: '192.168.1.1'
[17:58:17][C][sntp:052]:   Server 2: ''
[17:58:17][C][sntp:053]:   Server 3: ''
[17:58:17][C][sntp:054]:   Timezone: '<+03>-3'
```
[logs_emre-oda-esp32s-devkitv1_upload.txt](https://github.com/esphome/issues/files/9946824/logs_emre-oda-esp32s-devkitv1_upload.txt)

ssieb, 2022-11-06:
> I believe 2022.10.1 was working.

Are you saying it changed between `2022.10.1` and `2022.10.2`?
Where do you see the time being wrong?  I don't see the time used anywhere in the logs.

Salvora, 2022-11-06:
I see the time on the tm1637

Full yaml below:

```yaml
substitutions:
  devicename: "Emre Oda ESP32S"
  roomname: "Emre Oda"

esphome:
  name: emre-oda-esp32s-devkitv1

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino
    version: recommended

wifi:
  ssid: !secret esp_wifi_ssid
  password: !secret esp_wifi_pass
  id: emre_oda_esp32s_wi_fi

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$devicename Devkitv1"
    password: !secret esp_ap_password
    ap_timeout: 5min

captive_portal:

# Enable logging
logger:
# Enable debugging
debug:

# Enable Home Assistant API
api:
  password: !secret esp_api_password
  encryption:
    key: !secret emre-oda-esp32s-devkitv1_enc_key
  id: api_config

ota:
  password: !secret esp_ota_password

time:
  - platform: sntp
    servers: 192.168.1.1
    id: ntp_time

esp32_ble_tracker:
  scan_parameters:
    interval: 60s
    active: True
  id: esp_ble_tracker

bluetooth_proxy:
  active: true

globals:
  - id: display_status
    type: bool
    restore_value: true
    initial_value: "true"

  - id: display_mode_global
    type: int
    restore_value: true
    initial_value: "1"
[... 379 lines cut ...]
          data: 0x20DF10EF
          nbits: 32
      - lambda: |-
          id(emre_tv_power).publish_state(false);

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "$devicename IP Address"
      icon: mdi:ip
    ssid:
      name: "$devicename Connected SSID"
      icon: mdi:router-wireless
    bssid:
      name: "$devicename Connected BSSID"
      icon: mdi:router-wireless
    mac_address:
      name: "$devicename Wi-Fi MAC Address"
      icon: mdi:wifi-marker
      
  - platform: version
    name: "$devicename Version"
    icon: mdi:package-up

```

ssieb, 2022-11-06:
I don't know what's happening with the timezone conversion in your case.  That's a strange end result for the timezone value and I can get the same type of result with the `Etc/*` names as well.  Use `timezone: TRT-3` instead.

ssieb, 2022-11-06:
It's very strange that you say it's working properly with an 8266.  Can you paste logs from that?

Salvora, 2022-11-06:
8266 logs attached.
[logs_gozde-oda-esp8266-nodemcuv2_logs.txt](https://github.com/esphome/issues/files/9947038/logs_gozde-oda-esp8266-nodemcuv2_logs.txt)

Below is the config:
```yaml
substitutions:
  devicename: "Gözde Oda ESP8266"
  roomname: "Gözde Oda"

esphome:
  name: gozde-oda-esp8266-nodemcuv2

esp8266:
  board: nodemcuv2
  framework:
    version: recommended

wifi:
  ssid: !secret esp_wifi_ssid
  password: !secret esp_wifi_pass
  id: gozde_oda_esp8266_wi_fi

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$devicename Nodemcuv2"
    password: !secret esp_ap_password
    ap_timeout: 5min

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esp_api_password
  encryption:
    key: !secret gozde-oda-esp8266-nodemcuv2_enc_key
  id: api_config

ota:
  password: !secret esp_ota_password
  
time:
  - platform: sntp
    servers: 192.168.1.1
    id: ntp_server
    
globals:
  - id: display_status
    type: bool
    restore_value: true
    initial_value: 'true'

  - id: display_mode_global
    type: int
    restore_value: true
    initial_value: '1'

number:
  - platform: template
    id: display_intensity 
    name: "$devicename Ekran Parlaklığı (0-7)"
    icon: mdi:brightness-5
    optimistic: true
    mode: slider
    initial_value: 0
    min_value: 0
    max_value: 7
    step: 1
[... 161 lines cut ...]
    turn_off_action:
      - globals.set:
          id: display_status
          value: 'false' 
      - component.update: tm1637_display
             
text_sensor:
  - platform: wifi_info
    ip_address:
      name: "$devicename IP Address"
      icon: mdi:ip
    ssid:
      name: "$devicename Connected SSID"
      icon: mdi:router-wireless
    bssid:
      name: "$devicename Connected BSSID"
      icon: mdi:router-wireless
    mac_address:
      name: "$devicename Wi-Fi MAC Address"
      icon: mdi:wifi-marker
      
  - platform: version
    name: "$devicename Version"
    icon: mdi:package-up
```

Salvora, 2022-11-06:
`timezone: TRT-3` gives the correct time.

Blacksli, 2022-11-07:
confirm the problem.

If UTC+5 then the output is -5 (10 hours difference)

Tried

esp32:
   board: esp32dev
   framework:
     type: esp-idf

and tried arduino framework

If you use timezones such as:

#define TZ_Asia_Yekaterinburg **PSTR("<+05>-5")**
#define TZ_Asia_Thimphu **PSTR("<+06>-6")**

it displays wrong

[image]
[image]

if you specify
**PSTR("<-05>-5")**
then everything is correct
[image]

if you specify
#define TZ_Europe_Moscow **PSTR("MSK-3")**
then everything is correct

Blacksli, 2022-11-07:
```
time:
  - platform: sntp
    id: sntp_time
    timezone: Asia/Karachi
    on_time:
      - seconds: /5
        then:
          - lambda: |-
             ESP_LOGE("This", "Now: %u:%02u:%02u", id(sntp_time).now().hour, id(sntp_time).now().minute, id(sntp_time).now().second);

```

[image]

ssieb, 2022-11-07:
Yes, that's the problem.  But I don't know where exactly the cause is.

Salvora, 2022-11-13:
Esphome 2022.10.0 was fine. Whatever changed broke something.

Kaan88, 2022-12-18:
Facing the same issue on the latest version still.

Salvora, 2022-12-30:
still issue

Ssobarzo, 2023-01-27:
It persist on 2022.12.8.

locolc, 2023-02-01:
found the same issue for America/Buenos_aires config as well. ESPHOME version 2022.12.8

locolc, 2023-02-02:
i found out (thanks to Joaquin88) that using "timezone: WART4WARST,J1/0,J365/25" instead of "timezone: America/Argentina/Buenos_aires" the sync works correctly... at least for my timezone

daniellhotka, 2023-02-16:
I can confirm locolc's findings and solution, also version 2022.12.8

I tried other timezone codes corresponding to other places on the globe, and they resolve fine. The problem is only when I set a timezone that matches the one in my HomeAssistant installation. Then it inverts -3 to +3

nkgiovannivl, 2023-02-24:
Hi, same issue for me, so based om @locolc comment, my GMT -5 () does not works with timezone: America/Bogota, if i use timezone: EST+5 works!, i'm not sure if GMT -5 (America/Bogota) = EST+5 all the year.... can you confirm please? Thanks!

fakuivan, 2023-11-18:
Same here for GMT-3. The offset seems to be interpreted backwards, GMT+3 subtracts three hours from GMT while GMT-3 adds three hours.

edwardtfn, 2024-02-06:
I've tried different scenarios here and looks like the issue is related on how ESP interprets the Posix Timezone info provided by TZ Database (via Python tzdata when compiling). This is the format as per docs: https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html
Tree different formats are supported and looks like it works fine when the format is like `<+06>-6` or `GMT-6`, which is the first format described there:
```txt
The first format is used when there is no Daylight Saving Time (or summer time) in the local time zone:

std offset
The std string specifies the time zone abbreviation. It must be three or more characters long and must not contain a leading colon, embedded digits, commas, nor plus and minus signs. There is no space character separating the time zone abbreviation from the offset, so these restrictions are necessary to parse the specification correctly.

The offset specifies the time value you must add to the local time to get a Coordinated Universal Time value. It has syntax like [+|-]hh[:mm[:ss]]. This is positive if the local time zone is west of the Prime Meridian and negative if it is east. The hour must be between 0 and 24, and the minute and seconds between 0 and 59.
```
Note that the specification says the *offset* is the time to be added to the **local time zone**, and not to the **UTC**.

The problem is that in both `Arduino` or `esp-idf v4`, the local time provided to ESPHome is incorrectly calculated. On `esp-idf v5` this was fixed and the time is displayed correctly.

I don't know how are the plans to update the frameworks used by ESPHome. If there's no plans to replace it in the near future, we probably should have this handled in some way from ESPHome side. As this was already fixed on ESP-IDF v5, I think is unlikely to have a fix to an earlier version coming from Espressif.

closed 2025-09-02 by jesserockz