esphome/issues #3770: Disconnected from ESPHome API

bedson87, 2022-11-06:
### The problem

Several times per hour, the ESP32 device (in logs 192.168.10.26) loses connection with the API and then reconnects. Unfortunately this affects messages over the loudspeaker (dfplayer over uart). I tried to set a static IP but it didn't change anything.

If there's anything else I can provide to help debug this issue, please let me know.
Thank you in advance!

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.1

### What platform are you using?

ESP32

### Board

ESP32-wroom-32U

### Example YAML snippet

```yaml
esphome:
  name: esp32-ble

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_password
  #obsługa DF
  services:
  - service: dfplayer_next
    then:
      - dfplayer.play_next:
  - service: dfplayer_previous
    then:
      - dfplayer.play_previous:
  - service: dfplayer_play
    variables:
      file: int
    then:
      - dfplayer.play: !lambda 'return file;'
  - service: dfplayer_play_loop
    variables:
      file: int
      loop_: bool
    then:
      - dfplayer.play:
          file: !lambda 'return file;'
          loop: !lambda 'return loop_;'
  - service: dfplayer_play_folder
    variables:
      folder: int
      file: int
    then:
      - dfplayer.play_folder:
          folder: !lambda 'return folder;'
          file: !lambda 'return file;'

  - service: dfplayer_play_loop_folder
    variables:
      folder: int
    then:
      - dfplayer.play_folder:
          folder: !lambda 'return folder;'
          loop: true

  - service: dfplayer_set_device_tf
    then:
      - dfplayer.set_device: TF_CARD

  - service: dfplayer_set_device_usb
    then:
      - dfplayer.set_device: USB

  - service: dfplayer_set_volume
    variables:
      volume: int
    then:
      - dfplayer.set_volume: !lambda 'return volume;'
  - service: dfplayer_set_eq
    variables:
      preset: int
    then:
      - dfplayer.set_eq: !lambda 'return static_cast<dfplayer::EqPreset>(preset);'

  - service: dfplayer_sleep
    then:
      - dfplayer.sleep

  - service: dfplayer_reset
    then:
      - dfplayer.reset

  - service: dfplayer_start
    then:
      - dfplayer.start

  - service: dfplayer_pause
    then:
      - dfplayer.pause

  - service: dfplayer_stop
    then:
      - dfplayer.stop

  - service: dfplayer_random
    then:
      - dfplayer.random

  - service: dfplayer_volume_up
    then:
      - dfplayer.volume_up

  - service: dfplayer_volume_down
    then:
      - dfplayer.volume_down

ota:
  password: "XXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.10.26
  fast_connect: on
  manual_ip:
    static_ip: 192.168.10.26
    gateway: 192.168.10.1
    subnet: 255.255.255.0

time:
  - platform: homeassistant
    id: homeassistant_time

captive_portal:

# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:
  scan_parameters:
    active: false

#podłączenia
dallas:
  - pin: GPIO21
    
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600

dfplayer:
  on_finished_playback:
    then:
      logger.log: 'Playback finished event'

#sensory dla ble
sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: C4:7C:8D:6D:5B:F9
    temperature:
      name: "Kwiatek Leon Temperature"
    moisture:
      name: "Kwiatek Leon Moisture"
    illuminance:
      name: "Kwiatek Leon Illuminance"
    conductivity:
      name: "Kwiatek Leon Soil Conductivity"

  - platform: xiaomi_hhccjcy01
    mac_address: C4:7C:8D:6C:B9:40
    temperature:
      name: "Kwiatek inny Temperature"
    moisture:
      name: "Kwiatek inny Moisture"
    illuminance:
      name: "Kwiatek inny Illuminance"
    conductivity:
      name: "Kwiatek inny Soil Conductivity"
#    battery_level:
#      name: "Kwiatek Leon Battery Level"
  
# pomiar temp dallas
  - platform: dallas
    address: 0x552b1ec70664ff28
    name: "ESP32BLE czujnik temperatury"

#dodatkowe
  - platform: uptime
    name: "ESP32BLE Last Restart"
    update_interval: 300s
    device_class: "timestamp"
    unit_of_measurement: ""
    filters:
      - lambda: |-
          auto time = id(homeassistant_time).now();
          if (time.is_valid()) {
            return int(time.timestamp) - int(x);
          } else {
            return {};
          }
  - platform: wifi_signal
    name: "ESP32BLE WiFi Signal"
    update_interval: 300s
    device_class: "signal_strength"
```

### Additional information

[esphome_filter.log](https://github.com/esphome/issues/files/9946975/esphome_filter.log)
[logs_esp32-ble_logs.txt](https://github.com/esphome/issues/files/9946976/logs_esp32-ble_logs.txt)

ssieb, 2022-11-06:
Is the device rebooting?  Can you get serial logs to see what's happening?

bedson87, 2022-11-08:
> Is the device rebooting? Can you get serial logs to see what's happening?

Thanks for the answer.  I can't see the device resetting.  I checked yesterday quickly that via uart output was identical to the one via wireless.  I will try to check and past logs for days.  I also changed the power supply, but it didn't change anything. I guess it was ok before the last updates but maybe that's unrelated.  I am still looking for the cause of the problem..

closed 2022-11-08 by bedson87