esphome/issues #3773: PID Climate does not support presets

wildekek, 2022-11-10:
### The problem

The documentation of `PID Climate` mentions it supports all features of the `Climate` component. It does not seems to support any of the Climate preset functions though, see yaml/error logs for details.

Main reason I'd like this is the convenience of triggering a preset via Home Assistant.  But it would additionally allow me to add power saving features to my heating system. In my particular case I want to watch the `on_state` events for preset changes and when I'm away, I want the heater to stop pre-heating water. (using OpenTherm). I could imagine other energy savings to be possible when this is unlocked.

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Component causing the issue

PID Climate

### Example YAML snippet

```yaml
climate:
  - platform: pid
    id: central_heating
    name: "Central heating"
    visual:
      min_temperature: 16
      max_temperature: 25
      temperature_step: 0.5
    heat_output: t_set
    default_target_temperature: 16
    sensor: ch_room_temperature
    control_parameters: 
      kp: 0.4
      ki: 0.004
    preset: AWAY
    custom_preset: 
      - name: Vacation
        default_target_temperature_low: 20 °C`
```

### Anything in the logs that might be useful for us?

```txt
[preset] is an invalid option for [climate.pid]. Please check the indentation.
[custom_preset] is an invalid option for [climate.pid]. Please check the indentation.
```

### Additional information

Example of Home Assistant service call:

> service: climate.set_preset_mode
> data:
>   preset_mode: home
> target:
>   entity_id: climate.central_heating
> 
Log:
> websocket_api script: Error executing script. Error for call_service at pos 1: Entity climate.central_heating does not support this service.

randybb, 2022-11-10:
These variables are not for the climate component, but for its action https://esphome.io/components/climate/index.html#climate-control-action

wildekek, 2023-10-21:
To make presets work, the preset configuration and actions from the thermostat component should be ported into PID. I'll have a look in the next few days to see if I can get that to work. I wonder though if it wouldn't make sense to add them to the climate base component, to remove duplication and have it available for other thermostats (eg Bang Bang) for free. What's your take on this @OttoWinter?

erapade, 2023-12-25:
> To make presets work, the preset configuration and actions from the thermostat component should be ported into PID. I'll have a look in the next few days to see if I can get that to work. I wonder [...]

Did you or @OttoWinter sort this out?

wildekek, 2024-01-03:
The proper way to implement this imo, is to add presets to the base climate component. I'll open a new issue for that, as I now have better understanding of the solution I want from this issue.

closed 2024-01-03 by wildekek

ChopperRob, 2024-03-10:
Hi @wildekek , did you open a new issue for this? would love to see the presets being available in the pid climate controller.