esphome/issues #3775: esp32_camera: on_stream_start runs every frame

krisnoble, 2022-11-10:
### The problem

I've got an esp32cam board with an LED that I want to turn on automatically when viewing via Home Assistant and turn off when not needed. I have it set up as a GPIO switch with the intention of being able to turn the LED on/off manually also. Using
```
on_stream_start:
  - then:
    - switch.turn_on: led
```
and the corresponding `on_stream_stop` works, but it seems from the logs that the `on_stream_start` gets fired every time the camera sends a frame to HA, which means if I turn it off manually during a stream, it almost immediately turns back on again.

Is this the intended behaviour? I was expecting the `on_stream_start` to only fire once.

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32cam

### Component causing the issue

esp32_camera

### Anything in the logs that might be useful for us?

```txt
[17:10:24][D][esp32_camera:168]: Got Image: len=27189
[17:10:25][D][switch:013]: 'Sunlu S8 camera LED' Turning ON.
[17:10:25][D][esp32_camera:168]: Got Image: len=27092
[17:10:26][D][switch:013]: 'Sunlu S8 camera LED' Turning ON.
[17:10:26][D][esp32_camera:168]: Got Image: len=27152
[17:10:26][D][switch:013]: 'Sunlu S8 camera LED' Turning ON.
```
```

junalmeida, 2023-01-04:
Interesting. I totally ignored this trigger just because I do expect to have a trigger that fires right before each snapshot, which seems to be the case based on your logs.

junalmeida, 2023-01-04:
Hm, I gave it a try, and it does not seem to be running on every frame according to my logs: 

<html>
<body>

10:08:50 | [D] | [light:035] | 'Water Meter Device Flash' Setting:
-- | -- | -- | --
10:08:50 | [D] | [light:046] | State: ON
10:08:50 | [D] | [esp32_camera:172] | Got Image: len=31237
10:08:50 | [D] | [esp32_camera_web_server:198] | MJPG: 31237B 100ms (10.0fps)
10:08:50 | [D] | [esp32_camera:172] | Got Image: len=60070
10:08:50 | [D] | [esp32_camera_web_server:198] | MJPG: 60070B 245ms (4.1fps)
10:08:50 | [D] | [esp32_camera:172] | Got Image: len=58959
10:08:50 | [D] | [esp32_camera_web_server:198] | MJPG: 58959B 223ms (4.5fps)
10:08:50 | [D] | [esp32_camera:172] | Got Image: len=56955
10:08:50 | [D] | [esp32_camera_web_server:198] | MJPG: 56955B 223ms (4.5fps)
10:08:51 | [D] | [esp32_camera:172] | Got Image: len=55582
10:08:51 | [D] | [esp32_camera_web_server:198] | MJPG: 55582B 253ms (4.0fps)
10:08:51 | [D] | [esp32_camera:172] | Got Image: len=53096
10:08:51 | [D] | [light:035] | 'Water Meter Device Flash' Setting:
10:08:51 | [D] | [light:046] | State: OFF
10:08:51 | [I] | [esp32_camera_web_server:208] | STREAM: closed. Frames: 5

</body>
</html>

My `esp32_camera` config:
```
  id: camera
  on_stream_start:
    then:
      - light.turn_on: light_flash
  on_stream_stop:
    then:
      - light.turn_off: light_flash
```

Maybe it is fixed already, however I miss an event to run before and after each image snapshot.

krisnoble, 2023-01-04:
Weird, I'm still experiencing the same issue on the latest release. I did wonder if it might be a bug related to my use of a switch versus your use of a light but I've just tested changing the switch to a binary light and had the same issue. I can see from the logs that you're running the web server component whereas I'm not but I don't see why that should cause such a problem.

I agree it would be nice to have a `before_stream_start` trigger option too. And perhaps a conceptual separation between a stream vs a snapshot since we might want to do different things in each case.

junalmeida, 2023-01-04:
https://github.com/esphome/feature-requests/issues/1673 This is closed as completed, however only automations for stream are available. I don't see any automation on idle snapshots.